nextcloud/nextcloudpi #381: better "docker run" instructions

testbird, 2018-01-29:
Found the following that may improve the docker instructions you give in your article.

Automatically determine IP:
```
IP=`hostname -I | { read first anddropthis ; echo $first ; }`
```

Also add these to the docker run command:
`--restart=always` to have it running after reboot etc.
`--dns=192.168.0.1` (IP adjusted or auto-determined) to prevent docker from leaking usage info to google dns servers

testbird, 2018-01-30:
No, wait! Finally found a working way to attach the docker container to the lan with an own IP.
No more messing with port forwarding, collisions, host IPs, or domains!

1. Create a new docker network "lan" that is directly attached to your local network on the ethernet level (layer 2). Re-use the same configuration from your lan interface (in this example it's eth0 on the 192.168.1.x subnet):
  `docker network create --driver macvlan --subnet 192.168.1.0/24 --gateway 192.168.1.1 -o parent=eth0 lan`

1. Run the NCP container (x86 in this example) attached to the new docker "lan" network:
  `docker run --restart=always --net=lan --ip=192.168.1.22 --dns=192.168.1.1 -d -v ncdata:/data --name nextcloudpi ownyourbits/nextcloudpi-x86`

testbird, 2018-01-31:
It may even be preferable to include the docker network create stuff into the dockerfile, and to do automatic (shell) determination of the network interface.  That is the one that mentioned in the default gateway route.  And to use its subnet configuration.

Then only promote the `docker-compose` way to regular users. It looks like it could provide the easiest* way to install and update NCP dockers, i.e. instructions (or a ncp-docker script?) to download the compose file and using the same `docker-compose pull ; docker-compose [nc|ncp|ncp-armhf] up -d` to install as well as to update (https://stackoverflow.com/a/43515922)

*maybe publish the dockerfiles separately (not requiring to git clone the whole source tree) and **renamed accrding to what the install.

Aeon512, 2018-02-17:
I would stay with the docker-compose file but would add the label to _officially_ support watchtower (https://github.com/v2tec/watchtower). That way, whenever a new docker image is released it should be replaced within some minutes.

So far, there have been problems when using letsencrypt #404 and redis / mariadb #414. Once this is integrated, I'm happy to give it another try - but it looks promising :-)

nachoparker, 2018-02-19:
@testbird 

I did play with the _macvlan_ driver when I was creating the docker version, but I decided that I liked the network isolation docker provides.

In any case, that was a quick decision and we can study the proposed changes which might very easily make more sense.

@Aeon512 

watchtower looks like exactly what we need!! that's awesome. I was thinking that we were going to implement something like that ourselves ^^

Aeon512, 2018-02-19:
I hope that with my latest pull requests the docker image can just be updated by watchtower. Will activate that on my side once the changes are in and will continue to test this in following updates...

testbird, 2018-02-19:
**macvlan:** I read somewhere, and noticed, that it does not allow any connectivity between host and container, unless enabled with some config option I think, only then it allows all ports at once and would require additional firewalling for selective access. Port forwarding from external router/firewall to separate IP works fine.
The main advantage I found with separate IPs were never to have port conflicts between container and host (especially 80,443). (All web interfaces can continue to be reached on standard ports internally, and external port forwardings (or even proxying) is possible, pointing to the separate internal IPs as required.)

nachoparker, 2018-02-22:
@Aeon512 have you tried watchtower with NCP?

Aeon512, 2018-02-22:
@nachoparker: what specially are you referring to?
Watchtower is an addditional docker image which checks every x minutes it a new nextcloudpi docker image is published at docker hub. If yes, it will pull the image stop the current container and start it again with new newly downloaded image. 

Hence, it changes the whole image, nothing particular to NCP.

nachoparker, 2018-02-23:
@Aeon512 I see... well, that means that you have to install it separatedly.

Updating of the docker container needs to be improved. While watchtower is amazing, it would be ideal to have a self-contained solution.

With regards to debian packages, the docker image should upgrade them automatically with 'unattended upgrades'. You should be able to verify this by looking at `/var/log/unattended-upgrades/unattended-upgrades.log`.

With regards to upgrading NCP itself, the same could be done with `nc-autoupdate-nc`, as it is done in the SD card version.

I don't think these solutions are super clean. I know that this is using docker more as a virtual machine that runs multiple processes than the way docker is designed to run. I believe it makes sense though in our case, but if the design fails to meet our needs, we might need to separate NCP docker into 3 components ( lamp, nextcloud, nextcloudpi )that can be upgraded individually. Still the problem would remain to upgrade without external watchtower like software, so as of today I am still leaning towards the one container solution.

With regards to Nextcloud version jumps, I want to automate this for the SD card version, and the docker image also could benefit from this. Just replacing the `/var/www/nextcloud` directory is not enough, as the config and database info needs to be migrated. Effectively, `occ upgrade` would need to be run, so this could be part of the provisioning process, so that whenever the docker pull updates to a new NC version, `occ upgrade` is run.

It shouldn't be too hard to code this, as we already have the backup and restore part covered. In essence, it is a matter of scripting [these steps](https://docs.nextcloud.com/server/12/admin_manual/maintenance/manual_upgrade.html)

If there's no good solution I guess we could always add _watchtower_ to the docker-compose file.

---

Going back to the original topic, what's your idea about @testbird proposals? I think we are fine without using _macvlan_, but maybe I am missing something.

testbird, 2018-02-23:
Without macvlan and an own IP the docker image is not usable (confflicts) on hosts that already have any sort or web access (ports 80/443). This includes any configuration web interface on the host.

Watchtower may not be necessary if it would could be sufficient to have a cron job check for updates. Either on the host, as I had though above in https://github.com/nextcloud/nextcloudpi/issues/381#issuecomment-362072170. Or, if there is a way to configure or script docker to automatically pull updates when re-running a new container, the cron might even be inside the ncp-container and just kill the container to get updates applied (so that the restart=always results in pulling in the updated image).

To handle updating the states within ncp maybe there are some ideas to take from this solus blog post, especially usysconf sounds like a clever solution based on a very simple mechanism: https://solus-project.com/2018/02/15/exploring-solus-architecture/

nachoparker, 2018-02-23:
> Without macvlan and an own IP the docker image is not usable (confflicts) on hosts that already have any sort of web access (ports 80/443). This includes any configuration web interface on the host.

Have you tried mapping to other ports, like `docker run -p 8080:80 ...`, or in docker compose file?

> Watchtower may not be necessary if it could be sufficient to have a cron job check for updates. Either on the host, as I had thought above in #381 (comment). Or, if there is a way to configure or script docker to automatically pull updates when re-running a new container, the cron might even be inside the ncp-container and just kill the container to get updates applied (so that the restart=always results in pulling in the updated image).

I don't think you can start the process from inside the docker container, but a script to run the container that also checks for updates and performs the container change would be possible. I don't know if that would be better than using watchtower.

> To handle updating the states within ncp maybe there are some ideas to take from this solus blog post, especially usysconf sounds like a clever solution based on a very simple mechanism (triggers based on a list of modification times): https://solus-project.com/2018/02/15/exploring-solus-architecture/

Interesting. Will have a look

testbird, 2018-02-23:
> Have you tried mapping to other ports, like docker run -p 8080:80 ..., or in docker compose file?

But, then one always needs to add the non-standard port to the ncp domains, I didn't think that would be nice, so I was happy I found macvlan to simply use another ip for the local ncp domain, and get rid of the cumbersome process of getting the container to know its host IP.

Concerning the update trigger mentioned in that blog post, I think the mechanism is that the update/migrate program gets updates and is always only run once after package updates and after booting to check known distributed files with its list of last modification times. The result is that it will update/migrate those files to newer states whenever old file versions are found (e.g. backup restored) or  the program was updated with new migrations (e.g. docker image updated).

Aeon512, 2018-02-23:
In my opinion the _correct_ way to solve this the docker way would be to run NextCloudPi on a different port as suggested by @nachoparker, e.g. 8080. Then have an additional docker image like _nginx-proxy_ running which actually listens on port 80 and, depending on the specified host name forwards the request to the correct internal upstream port (e.g. 8080 in this example).

If you want to dig deeper, the are actually automated process to configure the Nginx Proxy automatically depending on the started upstream docker images. See e.g. https://github.com/jwilder/nginx-proxy

To sum up, while macvlan would be an alternative approach to solve the problem described by @testbird, the common approach by the docker community is to use an additional upstream proxy and keep the actual docker image like NextCloudPi more general and independent.

testbird, 2018-02-23:
> approach by the docker community is to use an additional upstream proxy and keep the actual docker image like NextCloudPi more general and independent of such technologies.

How do non-standard ports, port-forwardings, and needing to pass the host IP make container be more general? The nginx-proxy container won't be able to run on a device that uses port 80 or 443 for its own config either.
How is depending on an additional proxy more independent?
What disadvantage do you see for macvlan, and having the container behave just like a normal ncp install with its own IP? (I see that I haven't read up on using dhcp for the container.)
Personally, I was happy finding macvlan, because I like solutions that make things simpler without adding unnecessary cruft.

Nevertheless, I understand that in servers with docker separation of services using the nginx-proxy container, the port forwading may be standard. So, as the config lines aren't too complicated, after all, a ncp-docker-run.sh script could maybe therefore support both options.

Aeon512, 2018-02-23:
Maybe I misunderstood your point. If macvlan yields just to another _docke-compose_ or _docker-run_ script, than sure - **just fill a PR with such a file and it can be included**.

Regarding your other questions:
If the hosts already run's a web service on port 80 or 443, then just use it as a proxy to forward to the non-default port on the docker image. Or change the port on the host and use a docker proxy to do the work for you. 

Main disadvantage for macvlan in my eyes is that it adds another independent (internal) IP address to the whole story. If you want to provide external access with a single (external) IP to which (internal) IP should you forward port 80 or port 443? If all services run on the same IP but you use different host names to differentiate, it's a more straight forward approach.

testbird, 2018-02-26:
Thanks for answering.

So, it seems it all depends on whether one needs to have multiple standard http paths reachable on an external IP (port 80/443). This will require to install, configure and maintain a proxy and multiple (dynDNS) domains.

To have multiple internal servers be reachable on the standard ports, and only expose a single server to the outside, it is sufficient to use separate IPs and configure the port forwarding on the router for a particular IP.

nachoparker, 2018-02-27:
Good discussion. I'm learning here :)

Well, both solutions seem fine to me. I guess that regarding to answering to the question "should we change this design"?, the answer would be "no, because there are at least two ways of setting up this configuration, one of which implies using a proxy and keeps the NCP dockerfile standard and flexible"

If that is the right answer, then if anyone wants to do a more complex setup, they have the options of using _macvlan_, or the proxy or whatever, but NCP remains general and standard.

I guess that under that point of view the situation is similar to people wanting to install pihole on the same pi as NCP... if they want to do non-stardard complex setups power to them, but they must expect to do more complex setups themselves.

Don't you think?

testbird, 2018-02-27:
As stated, I think letting the user have to configure forwarding ports, possibly changing the host configuration, and fiddling the host IP into the docker container is more complex then necessary, and certainliy more complex then running the ncp docker image in a conatainer with its own IP (just like on a separate RPi).

Adding the complexitiy and surface of a reverse nginx-proxy is completely unrelated IMHO, because it is supposed to work with macvlan just as well https://github.com/jwilder/nginx-proxy/issues/1050.

Aeon512, 2018-02-27:
IMHO there’s nothing wrong with adding a docker-compose-macvlan file to the repository. That way the user can choose what he wants with already preconfigured Files.

nachoparker, 2018-02-27:
@testbird, do you have a working docker-compose file to share?

testbird, 2018-02-27:
I used the docker commands in the second post, but this find seems to have a static composer file that is only missing the dns option from the posted commands: https://github.com/nextcloud/nextcloudpi/issues/381#issuecomment-368955927 Unfortunately, I don't know if/how the IPs and ranges in the compose file may be automatically determined from the host system as in the envisioned/wished bash script above.

nachoparker, 2018-04-10:
In-container updates have now been implemented. We need to go through some of structural changes explained in #380 in order to be able to cleanly just pull a new container, and support something like watchtower. It's in the roadmap :)

About the network configuration, I think we can just leave it to the user choice.

Thanks

closed 2018-04-10 by nachoparker

testbird, 2019-06-22:
Hello nachoparker, it's been a while. In  case you wondered; my attempts to work on the curl debian install had failed due to obscure systemd login reliability failures I could not debug, only working every so many boots. And as you said the problems with a co-existing webserver stack are manifold, and just scripting in fellt conflict prone, and just not perfectly right.

I remembered we were discussing about some "perfect build system" for images incl. docker, here. Finaly, I think I possibly may have come across seeing a sane solution for building and maintaining a set of architecture images, docker, tarballs, and also universal packages, including combineable webserver stack config (parts). Did you come across the GNU Guix package manager / linux distro yet?

(They recently released v1.0. It can make it "surprisingly easy to deploy otherwise complex services such as applications that depend on Web services. For instance, setting up CGit or Zabbix is a one-liner, even though behind the scenes that involves setting up nginx, fcgiwrap, etc. We’d love to see to what extent this helps people self-host services—sort of similar to what FreedomBox and YunoHost have been focusing on." https://www.gnu.org/software/guix/blog/2019/gnu-guix-1.0.0-released/)

It is usually presented as a univeral package manager or next-generation linux distro, but its declarative system and config definition and bulid artefact pruduction commands quite changes the way of software distribution (no package maintanance bottleneck).

It's does not seem to be that easy to get all benefits at once, but reading up it seems to solve well not only the problem of having images including dependencies while tracking and generating security updates for all included libs. 

Main features explained: https://ambrevar.xyz/guix-advance/

Technically, first a hint for a much saner syntax to avoid the parentheses headache scheme:
https://dustycloud.org/blog/wisp-lisp-alternative/

The get-in-fast intro to concepts-and-tech:
https://gitlab.com/rendaw/blog/blob/master/how_to_guix_for_those_who_dont.md

Mainline kernels: https://gitlab.com/nonguix

A project to produce SoC images, but with the Guix predecessor project NixOS:
https://github.com/telent/nixwrt

nachoparker, 2019-06-22:
Thanks for the info. I have been reading things about guix and it looks interesting, even though I haven't played with it.

Unfortunately I don't think NCP has the resources to reimplement everything using it.

Good read though, thanks!

testbird, 2019-06-23:
Reimplementing? I think it's rather trying out a new linux distro type, that happens to be able to provide declarative configuration as well as generators for portable packages, docker images, and SoC images (besides generating .iso images and system installations). 

And then to install ncp-web. The Ncp-web (docker image version) might possibly not even need writing of an own module definition, only inheriting and extending the nextcloud module so it gets installed in the same "package space" to manage nextcloud. In NixOS nextcloud is already available
https://nixos.org/nixos/manual/

And I think for SoCs without native distro support (yet) it should still be possible to install the universal guix or nix package say to armbian or raspian etc. systems, but now with a fully fledged update and rollback system.

testbird, 2019-06-23:
Maybe it's not reimplementing, but possibly using universal distro infrastructure instead, that can reduce the burden of maintaining own mechanisms?

testbird, 2019-07-02:
Found that NixOS actually already provides official rpi3 sdcard images, together with update channels that include pre-compiled binary packages.
https://nixos.wiki/wiki/NixOS_on_ARM
It's a peace of cake to get that running and declare `services.nextcloud.enable = true;` to pull everything in.