nextcloud/nextcloudpi #382: /etc/apache2/conf-enabled/security.conf

testbird, 2018-01-29:
The file does not seem to be adjusted for production, yet.

nachoparker, 2018-02-08:
could you be a bit more specific please ? 8P

thanks for the feedback

testbird, 2018-02-08:
Thinking the `ServerTokens Prod` is sure to work and protects version meta data from g**gle hacking , and enabling the clickjack protection "sameorigin" should also work.
The others could work, depending on the ncp configuration.

nachoparker, 2018-02-10:
do you have a PR for me? ;)

nachoparker, 2018-02-10:
I'll close

we can keep talking, and if there's work to be done I'll reopen

thanks

testbird, 2018-02-10:
Allright, I see the header options seem to be taken care of.
Only I'm not sure about the `ServerTokens Prod`apache option which avoids disclosing information on the error page if something triggers an error.

nachoparker, 2018-02-10:
that's cool, we'll include that one ;)

will you send a PR?

testbird, 2018-02-10:
I frankly don't know where and how this is to be edited in ncp.
(It's goes beyond changing an existing string. Currently there is just the debian default conf file. In FBX they seem to include parsers to edit the config files.)

nachoparker, 2018-02-10:
that's alright, I'll do it. It's pretty easy

```
ServerSignature Off
ServerTokens Prod
```

I will add these [here](https://github.com/nextcloud/nextcloudpi/blob/master/lamp.sh#L50), and in update.sh for older images

nachoparker, 2018-02-16:
implemented in `v0.46.12` 

thanks!

closed 2018-02-16 by nachoparker