nextcloud/nextcloudpi #383: docker security

testbird, 2018-01-29:
publish independently?
https://titanous.com/posts/docker-insecurity

(Link found with further things at https://github.com/wsargent/docker-cheat-sheet#security)

nachoparker, 2018-02-11:
thanks for the tips. I think we are ok with those. This being said, if anyone wants to propose specific solutions, those are very welcome. Bring those PRs!

I can add my own list of TODO things for docker security. Let's try to create a list.

[ ] try user namespaces ( --userns-remap=default, -u )
[ ] try using USER
[ ] review capabilities / seccomp filters and use the bare minimum

More links 

---

https://logz.io/blog/docker-security/

https://opensource.com/business/15/3/docker-security-tuning

https://docs.docker.com/engine/security/seccomp/

https://www.sumologic.com/blog/security/securing-docker-containers/

https://opensource.com/business/14/9/security-for-docker

https://docs.docker.com/engine/security/security/

https://sreeninet.wordpress.com/2016/03/06/docker-security-part-1overview/

https://outflux.net/teach-seccomp

https://docs.docker.com/engine/swarm/secrets/#simple-example-get-started-with-secrets

nachoparker, 2018-02-22:
I am pretty busy right now, but maybe someone can help ( @Aeon512 ? ). I'll add the tag.

Aeon512, 2018-02-22:
@nachoparker: the initial post just describes some general shortcomings of docker. Nothing particularly applying to NextcloudPi. Having said that I might look into improving the dockerfile if time permits. First of all we should get rid of the warnings before improving them...
However currently I’m rather busy with other topics...

nachoparker, 2018-02-23:
@Aeon512 that's great. It makes me feel better that there are some people looking at this. ATM I am not using the docker image, so somebody that actually uses it and is there to detect things is greately appreciated.

Of course, there is no rush. If I have time ( I don't think so ), I will look into those bullets I mentioned above ( seccomp, USER and such )

closed 2018-10-27 by nachoparker