nextcloud/nextcloudpi #384: I can not use a 1TB hard drive as the data location.

chiyi4488, 2018-01-30:
I can not use a 1TB hard drive as the data location.

Always say :  "ERROR: Count 0 device. Please insert only the USB drive to be formatted''

Paste the output of `sudo ncp-report`

Pant, 2018-01-31:
Asking for others to connect to your server is a very bad decision because he could install something malicious. Moreover this place is not for asking others to get things done, but to get help and do it on your own.

You should run from a terminal with ssh to the Raspberry Pi the command `sudo ncp-report` and paste here in a new comment the output in order for us to understand what is your issue.

I am closing this issue for now, but if you want help and come back with the output of the command, reopen this issue.

chiyi4488, 2018-01-31:
ok

chiyi4488, 2018-01-31:
ok,but i  need help.can you open  this issues.
my ncp-report no thing

chiyi4488, 2018-01-31:
ok,but i need help.can you open this issues.
i think i can't open this issues
my ncp-report no thing

Pant, 2018-01-31:
If you don't provide us info from your setup, configuration and logs we can't help you. You have to run `ncp-report` from ssh and paste here the output.

chiyi4488, 2018-02-01:
i know .
but i type `sudo ncp-report`in ssh
system give this

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

Nextcloud is in maintenance mode - no app have been loaded

chiyi4488, 2018-02-01:
So, i don't know i can do something

Pant, 2018-02-01:
Most probably you are stuck in maintenance mode.
Run the following through ssh.
```
sudo -u www-data php occ maintenance:mode --off
```
Then run `sudo ncp-report` again and post the output.

chiyi4488, 2018-02-02:
No, can't
System give this
`Could not open input file: occ`

chiyi4488, 2018-02-02:
I typed`sudo ncp-report` , the result is just the same.

chiyi4488, 2018-02-05:
i close maintenance mode

closed 2018-02-05 by chiyi4488

nachoparker, 2018-02-05:
hi,

please give us the output of `sudo ncp-diag`

chiyi4488, 2018-02-06:
NextCloudPi version|v0.41.0
NextCloudPi image|NextCloudPi_12-04-17
distribution|Raspbian GNU/Linux 9 \n \l
automount|yes
USB devices|sda
datadir|/media/myCloudDrive/ncdata
data in SD|no
data filesystem|btrfs
data disk usage|9.5G/932G
rootfs usage|1.9G/7.3G
sudo: /var/www/nextcloud/occ: command not found
Nextcloud check|error
HTTPD service|up
PHP service|up
MariaDB service|up
Redis service|up
Postfix service|up
internet check|ok
port check 80|open
port check 443|open
Device "" does not exist.
IP|
gateway|
interface|

chiyi4488, 2018-02-06:
I changed the transformer, it succeeded!

nachoparker, 2018-02-06:
good!!