nextcloud/nextcloudpi #385: Internal Server Error

jerhone, 2018-01-30:
Paste the output of `sudo ncp-report`
<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.44.15
NextCloudPi image    NextCloudPi_09-28-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/WDRedDrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      91G/1.8T
rootfs usage         1.9G/15G
swapfile             /var/swap
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.44
gateway              192.168.1.254
interface            eth0
certificates         none
certs due            none
NAT loopback         no
uptime               11min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
OCP\AppFramework\QueryException: Could not resolve OCA\DAV\HookManager! Class OCA\DAV\CardDAV\SyncService does not exist in /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php:102
Stack trace:
#0 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(117): OC\AppFramework\Utility\SimpleContainer->resolve('OCA\\DAV\\HookMan...')
#1 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCA\\DAV\\HookMan...')
#2 /var/www/nextcloud/lib/private/AppFramework/DependencyInjection/DIContainer.php(416): OC\ServerContainer->query('OCA\\DAV\\HookMan...')
#3 /var/www/nextcloud/apps/dav/lib/AppInfo/Application.php(77): OC\AppFramework\DependencyInjection\DIContainer->query('OCA\\DAV\\HookMan...')
#4 /var/www/nextcloud/apps/dav/appinfo/app.php(30): OCA\DAV\AppInfo\Application->registerHooks()
#5 /var/www/nextcloud/lib/private/legacy/app.php(209): require_once('/var/www/nextcl...')
#6 /var/www/nextcloud/lib/private/legacy/app.php(149): OC_App::requireAppFile(Object(OCA\DAV\AppInfo\Application))
#7 /var/www/nextcloud/lib/private/legacy/app.php(124): OC_App::loadApp('dav')
#8 /var/www/nextcloud/lib/private/Console/Application.php(104): OC_App::loadApps()
#9 /var/www/nextcloud/console.php(99): OC\Console\Application->loadCommands(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#10 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#11 {main}
```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Tue Jan 30 06:25:02.712543 2018] [ssl:warn] [pid 927:tid 1992105984] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Tue Jan 30 06:25:03.000645 2018] [mpm_event:notice] [pid 927:tid 1992105984] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Tue Jan 30 06:25:03.000779 2018] [core:notice] [pid 927:tid 1992105984] AH00094: Command line: '/usr/sbin/apache2'
[Tue Jan 30 21:29:17.367814 2018] [mpm_event:notice] [pid 927:tid 1992105984] AH00491: caught SIGTERM, shutting down
[Tue Jan 30 21:29:31.286690 2018] [ssl:warn] [pid 641:tid 1992179712] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[... 43 lines cut ...]

```

{"reqId":"WmtHrX8AAQEAACPziU0AAAUK","level":3,"time":"2018-01-26T15:22:25+00:00","remoteAddr":"192.168.1.237","user":"--","app":"no app in context","method":"PUT","url":"\/remote.php\/webdav\/Photos\/2017\/10\/17-10-29%2006-59-35%205761.heic","message":"Exception: {\"Exception\":\"Symfony\\\\Component\\\\Routing\\\\Exception\\\\RouteNotFoundException\",\"Message\":\"Unable to generate a URL for the named route \\\"theming.Theming.getLogo\\\" as such route does not exist.\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Route\\\/Router.php(339): Symfony\\\\Component\\\\Routing\\\\Generator\\\\UrlGenerator->generate('theming.Theming...', Array, 1)\\n#1 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Route\\\/CachingRouter.php(60): OC\\\\Route\\\\Router->generate('theming.Theming...', Array, false)\\n#2 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/URLGenerator.php(77): OC\\\\Route\\\\CachingRouter->generate('theming.Theming...', Array)\\n#3 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/theming\\\/lib\\\/ThemingDefaults.php(165): OC\\\\URLGenerator->linkToRoute('theming.Theming...')\\n#4 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/public\\\/Defaults.php(186): OCA\\\\Theming\\\\ThemingDefaults->getLogo(true)\\n#5 \\\/var\\\/www\\\/nextcloud\\\/core\\\/templates\\\/layout.guest.php(45): OCP\\\\Defaults->getLogo()\\n#6 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Template\\\/Base.php(176): include('\\\/var\\\/www\\\/nextcl...')\\n#7 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Template\\\/Base.php(151): OC\\\\Template\\\\Base->load('\\\/var\\\/www\\\/nextcl...', NULL)\\n#8 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/template.php(202): OC\\\\Template\\\\Base->fetchPage(NULL)\\n#9 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/template.php(224): OC_Template->fetchPage()\\n#10 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Template\\\/Base.php(132): OC_Template->fetchPage()\\n#11 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/template.php(342): OC\\\\Template\\\\Base->printPage()\\n#12 \\\/var\\\/www\\\/nextcloud\\\/remote.php(83): OC_Template::printExceptionErrorPage(Object(OCP\\\\AppFramework\\\\QueryException))\\n#13 \\\/var\\\/www\\\/nextcloud\\\/remote.php(165): handleException(Object(OCP\\\\AppFramework\\\\QueryException))\\n#14 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/symfony\\\/routing\\\/Generator\\\/UrlGenerator.php\",\"Line\":130}","userAgent":"Mozilla\/5.0 (iOS) Nextcloud-iOS\/2.19.1","version":"12.0.3.3"}
{"reqId":"WmtHrX8AAQEAACPziU8AAAUP","level":2,"time":"2018-01-26T15:22:25+00:00","remoteAddr":"192.168.1.237","user":"--","app":"no app in context","method":"PUT","url":"\/remote.php\/webdav\/Photos\/2018\/01\/18-01-24%2016-03-14%206902.heic","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (iOS) Nextcloud-iOS\/2.19.1","version":"12.0.3.3"}
{"reqId":"WmtHrX8AAQEAACPziU8AAAUP","level":3,"time":"2018-01-26T15:22:25+00:00","remoteAddr":"192.168.1.237","user":"--","app":"remote","method":"PUT","url":"\/remote.php\/webdav\/Photos\/2018\/01\/18-01-24%2016-03-14%206902.heic","message":"Exception: {\"Exception\":\"OCP\\\\AppFramework\\\\QueryException\",\"Message\":\"Could not resolve OCA\\\\DAV\\\\HookManager! Class OCA\\\\DAV\\\\CardDAV\\\\SyncService does not exist\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/AppFramework\\\/Utility\\\/SimpleContainer.php(117): OC\\\\AppFramework\\\\Utility\\\\SimpleContainer->resolve('OCA\\\\\\\\DAV\\\\\\\\HookMan...')\\n#1 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/ServerContainer.php(132): OC\\\\AppFramework\\\\Utility\\\\SimpleContainer->query('OCA\\\\\\\\DAV\\\\\\\\HookMan...')\\n#2 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/AppFramework\\\/DependencyInjection\\\/DIContainer.php(416): OC\\\\ServerContainer->query('OCA\\\\\\\\DAV\\\\\\\\HookMan...')\\n#3 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/AppInfo\\\/Application.php(77): OC\\\\AppFramework\\\\DependencyInjection\\\\DIContainer->query('OCA\\\\\\\\DAV\\\\\\\\HookMan...')\\n#4 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/appinfo\\\/app.php(30): OCA\\\\DAV\\\\AppInfo\\\\Application->registerHooks()\\n#5 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/app.php(209): require_once('\\\/var\\\/www\\\/nextcl...')\\n#6 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/app.php(149): OC_App::requireAppFile(Object(OCA\\\\DAV\\\\AppInfo\\\\Application))\\n#7 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/app.php(124): OC_App::loadApp('dav')\\n#8 \\\/var\\\/www\\\/nextcloud\\\/remote.php(147): OC_App::loadApps(Array)\\n#9 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/AppFramework\\\/Utility\\\/SimpleContainer.php\",\"Line\":102}","userAgent":"Mozilla\/5.0 (iOS) Nextcloud-iOS\/2.19.1","version":"12.0.3.3"}
{"reqId":"WmtHrX8AAQEAACPziU8AAAUP","level":3,"time":"2018-01-26T15:22:26+00:00","remoteAddr":"192.168.1.237","user":"--","app":"no app in context","method":"PUT","url":"\/remote.php\/webdav\/Photos\/2018\/01\/18-01-24%2016-03-14%2069 [... cut ...] y\\\/template.php(342): OC\\\\Template\\\\Base->printPage()\\n#12 \\\/var\\\/www\\\/nextcloud\\\/index.php(67): OC_Template::printExceptionErrorPage(Object(OCP\\\\AppFramework\\\\QueryException))\\n#13 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/symfony\\\/routing\\\/Generator\\\/UrlGenerator.php\",\"Line\":130}","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10_13_2) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.132 Safari\/537.36","version":"12.0.3.3"}

```

jerhone, 2018-01-30:
I am not able to open my nextcloudpi web ui after upgrading to v0.44.15... no changes made on the configs etc... and I could not really understand the PHP exception errors... Please Help!

nitind, 2018-01-30:
ncp-diag is telling me /media/USBdrive/ncdata doesn't exist, and indeed the USBdrive symlink is no longer there.

jerhone, 2018-01-30:
"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable" how do I fix that? 
My datadir is /media/WDRedDrive/ncdata...

jerhone, 2018-01-30:
There seem to be a problem with the SyncService 'OCA\DAV\CardDAV\SyncService does not exist in' but I dunno how to disable this...

nachoparker, 2018-02-05:
@nitind if your link dissappeared that seems like a different issue. Please, open a new issue

@jerhone something happened with your instance but it is Nextcloud related.

Never seen that issue before, but if it happened "without you doing anything", you should report it, or see if it has been reported.

I suggest that you investigate in the forums or in the Nextcloud issue tracker

https://help.nextcloud.com

I'll close, but please let us know how it goes. We can keep talking here

closed 2018-02-05 by nachoparker

jerhone, 2018-02-10:
@nachoparker I just upgraded to that newer version and everything went berserk... a bit hard to fix on my end been reading lots of stuff already but I will try to fix it again on my rest hours...

is there a way to re-install nextcloud-pi totally and just plug in my external HDD as before preserving the files there?

jerhone, 2018-02-10:
@nachoparker thanks a lot. I appreciate your reply on this.

nachoparker, 2018-02-10:
maybe [this](https://github.com/nextcloud/nextcloudpi/wiki/How-to-backup-and-restore-a-NextCloudPi-instance-using-nextcloudpi-config) will be useful

otherwise, just 

- install a new image
- follow the wizard
- copy your files manually to the data folder
- run `nc-scan` to make NC aware of your files

traori, 2018-10-06:
I got the same problem. What was the solution ? Can somebody reply please?

nachoparker, 2018-10-07:
@traori what is your issue? please output of `sudo ncp-report`

traori, 2018-10-11:
@nachoparker:
I reinstall the latest Version. Here the last ncp-report
https://help.nextcloud.com/t/nextcloudpi-please-help/38517/7

In the admin view, first i saw again this error but then stops. On the linked page i don`t get everything online. Here the new full ncp-report

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.62.6
NextCloudPi image    NextCloudPi_10-05-18
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      44G/917G
rootfs usage         2.2G/15G
swapfile             /media/4cbb233e-2788-47bc-8b73-3ba5f271bfae/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    14.0.1.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               2days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.178.xx",
            "2": "xxx.spdns.de",
            "4": "xxx.spdns.de",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "https:\/\/xxx.spdns.de\/",
        "dbtype": "mysql",
        "version": "14.0.1.1",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
[... 47 lines cut ...]
```

{"reqId":"W7uV7H8AAQEAABdl@XoAAABK","level":2,"time":"2018-10-08T17:37:48+00:00","remoteAddr":"77.185.166.20","user":"--","app":"no app in context","method":"GET","url":"\/index.php\/204","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Android) ownCloud-android\/3.3.0","version":"14.0.1.1"}
{"reqId":"W7uV7H8AAQEAABdl@XwAAABQ","level":2,"time":"2018-10-08T17:37:48+00:00","remoteAddr":"77.185.166.20","user":"--","app":"no app in context","method":"GET","url":"\/index.php\/204","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Android) ownCloud-android\/3.3.0","version":"14.0.1.1"}
{"reqId":"W7uV7H8AAQEAABdl@XsAAABT","level":2,"time":"2018-10-08T17:37:48+00:00","remoteAddr":"77.185.166.20","user":"--","app":"no app in context","method":"GET","url":"\/index.php\/204","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Android) ownCloud-android\/3.3.0","version":"14.0.1.1"}
{"reqId":"W7uV7n8AAQEAABdl@X4AAABR","level":2,"time":"2018-10-08T17:37:50+00:00","remoteAddr":"77.185.166.20","user":"--","app":"no app in context","method":"GET","url":"\/index.php\/204","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Android) ownCloud-android\/3.3.0","version":"14.0.1.1"}
{"reqId":"W7uV7n8AAQEAABdl@X0AAABS","level":2,"time":"2018-10-08T17:37:50+00:00","remoteAddr":"77.185.166.20","user":"--","app":"no app in context","method":"GET","url":"\/index.php\/204","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Android) ownCloud-android\/3.3.0","version":"14.0.1.1"}
{"reqId":"W7uV7n8AAQEAABdl@X4AAABW","level":2,"time":"2018-10-08T17:37:50+00:00","remoteAddr":"77.185.166.20","user":"--","app":"no app in context","method":"GET","url":"\/index.php\/204","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Android) ownCloud-android\/3.3.0","version":"14.0.1.1"}
{"reqId":"W7udJH8AAQEAABdl@Y4AAABJ","level":2,"time":"2018-10-08T18:08:36+00:00","remoteAddr":"77.185.166.20","user":"--","app":"no app in context","method":"GET","url":"\/index.php\/204","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Android) ownCloud-android\/3.3.0","version":"14.0.1.1"}
{"reqId":"W7udJH8AAQEAABdl@Y8AAABI","level":2,"time":"2018-10-08T18:08:36+00:00","remoteAddr":"77.185.166.20","user":"--","app":"no app in context","method":"GET","url":"\/index.php\/204","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Android) ownCloud-android\/3.3.0","version":"14.0.1.1"}
{"reqId":"W7udJn8AAQEAABdl@ZAAAABM","level":2,"time":"2018-10-08T18:08:38+00:00","remoteAddr":"77.185.166.20","user":"--","app":"no app in context","method":"GET","url":"\/index.php\/204","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Android) ownCloud-android\/3.3.0","version":"14.0.1.1"}
[... cut ...] ib\/base.php","line":987,"function":"match","class":"OC\\Route\\Router","type":"->","args":["\/apps\/files\/"]},{"file":"\/var\/www\/nextcloud\/index.php","line":42,"function":"handleRequest","class":"OC","type":"::","args":[]}],"File":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/Driver\/PDOStatement.php","Line":105}},"CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (Windows NT 6.1; Win64; x64; rv:62.0) Gecko\/20100101 Firefox\/62.0","version":"14.0.1.1"}

```
</details>