nextcloud/nextcloudpi #386: docker image observations

testbird, 2018-01-30:
little inconsistendies that I found

* the text boxes (stdout?) are often dropping at least one first character. "nabling module security2"
* nc-info throws errors
```
 rep: /usr/local/etc/nextcloudpi-config.d/nc-automount.sh: No such file or directory
./nc-info.sh line 23: column: command not found
```
before showing the following:
* nc-info advice to use dnsmasq is wrong  with mdns or router that supprtorts local dns resolution (and promotes 8.8.8.8 tracking). (docker independent)
* nc-info advice to use usb drive is incorrect if host running from hdd/ssd
* Later the nc-info info box is empty, even though dnsmasq or usb where not enabled.
* NC updates (betas) are not available
* I am missing a ncp-log function to view log (default filter errors?)
* there is no logout button, only the hazardous off button. ;-)

nachoparker, 2018-02-11:
I aggree that there are a lot of adjustments to be made for the docker image.

Unfortunatedly things will go slow unless we find help

Thanks for reporting. We'll keep a list for when we have time/resources to look at this

- [ ] add `column` command
- [ ] adjust nc-info output
- [ ] updates
- [ ] remove poweroff button

The log... I was thinking it would be nice to have a general ncp-log with all the operations, but that would be a separate feature request

nemo19, 2018-03-13:
I'd like to add to the list: From other Docker projects I'm used to all persistent data being stored in volumes, so that containers can be removed and images updated without losing anything. So I started nextcloudpi-armhf with the --rm option, which removed the container upon exit. When I started a new container Apache failed to start because it tried to access my letsencrypt cert and key in /etc which did not exist anymore. Strange thing is that I don't know why Apache was expecting any letsencrypt files, maybe the Apache config is generated from some file in ncdata?

Anyway, please either move all persistent files to ncdata (/data), or include additional volumes in the default `docker run` command for all important paths.

nachoparker, 2018-03-14:
the persistent volume does keep apache configuration, certificates, database and everything that needs to be persisted.

do we have steps to reproduce?

nemo19, 2018-03-14:
Currently I can't take my nextcloud offline to reproduce the problem, I'll try in a few days. What I did is this:

* Open ports 80 and 443 on my router
* Start nextcloudpi-armhf:latest with a previously created but empty volume (#437)
* Set up letsencrypt from the ncp web interface
* Log in to nextcloud, set up users, upload some files
* Stop and remove the container
* Start a new container using the same `docker run` command and same volume as before

Effects: Apache and mariadb failed to start. The log said something about mariadb not being able to connect to the database and reported an error in some PHP file. I could start mariadb manually using `/etc/init.d/mariadb start`. Then I tried to start apache from `init.d` which failed and reported an error in the apache config file in `/etc` at the line where it specified the letsencrypt cert and key. They were supposed to be located in `/etc/letsencypt` I think. I had to remove the lines from the configs for the ncp web interface and nextcloud and could finally start apache. I was able to access ncp web but requesting a new letsencrypt cert failed and I could not get access to the nextcloud webinterface.

At that point I removed the container and cleared the volume and started from scratch (which worked flawlessly, thank you very much!) and made sure not to remove the ncp container, but only start and stop it.

testbird, 2018-03-14:
@nemo19 Don't know if things have been fixed but the problems you found seem to confirm posts here: https://github.com/nextcloud/nextcloudpi/issues/395

nemo19, 2018-03-15:
The second one seems very similar indeed.

I looked into it a little and found that `/etc/apache2` is a symlink to the persistent storage on `/data`. When using letsencrypt the site configs `/etc/apache2/sites-available/{ncp,nextcloud}.conf` contain the keys `SSLCertificateFile` and `SSLCertificateKeyFile`, which contain the paths `/etc/letsencrypt/live/<your-domain>/fullchain.pem` and `/etc/letsencrypt/live/<your-domain>/privkey.pem`. That path is not symlinked, and thus vanishes when the container is removed. So to handle this consistently a symlink should be added for `/etc/letsencrypt`.

Another enhancement would be using pretty urls (without index.php) by default. That would require enabling the apache mod `env` with `a2enmod env`, adding `'htaccess.RewriteBase' => '/',` to `/data/app/config/config.php` and running `sudo -u www-data php /data/app/occ maintenance:update:htaccess`. This could probably be done by `ncp-init`?

I cannot promise anything but I might be able to create pull requests for those issues some time next week.

testbird, 2018-03-15:
There could be some more files not be linked into persistent storage.
https://github.com/nextcloud/nextcloudpi/search?utf8=%E2%9C%93&q=%2Fetc%2F&type=

nachoparker, 2018-03-15:
@nemo19 great! PRs are more than welcome... we are lacking manpower because of the many open fronts.

About the issue with letsencrypt... you can see [here](https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/letsencrypt.sh#L50) that it gets transfered to `/data`, so maybe what's happening is that you got stuck in a weird situation after doing a `docker pull` of a more recent container?

before pushing the latest image I am pretty sure that I tested letsencrypt

nachoparker, 2018-03-17:
@nemo19 about _pretty urls_, the main reason that I decided not to use that is because the pi is quite limited in resources so I try to disable every apache module that we don't use.

it is true that it is more pleasing to have nicer urls but I thought that speed comes first. Thoughts?

testbird, 2018-03-17:
> Thoughts?

[[ "$DOCKERBUILD" == 1 ]]

nachoparker, 2018-04-10:
most of this things are now implemented and I will push them soon, with the notable exception of a beter PID 1 process

closed 2018-04-10 by nachoparker