nextcloud/nextcloudpi #387: 503 Service Unavailable

rencire, 2018-01-30:
Hi there, I'm new to nextcloud and everything seemed to be working when I first installed it.  However, after I accidentally disconnected the external HDD and power, and powered on the raspberry pi again, I am unable to access the web server.

Every time I restart the device, and go to https://nextcloudpi.local, I get a 503 error. Any help/tips would be appreciated, thanks!

Here is the report:

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.44.15
NextCloudPi image    NextCloudPi_12-04-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      347M/932G
rootfs usage         2.6G/15G
swapfile             /media/USBdrive/swap
Nextcloud check      error
HTTPD service        up
PHP service          down
MariaDB service      up
Redis service        up
Postfix service      down
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.50.219
gateway              192.168.50.1
interface            wlan0
certificates         none
certs due            none
NAT loopback         no
uptime               1day

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

/usr/local/bin/ncp-report: line 48: 15104 Segmentation fault      sudo -u www-data php /var/www/nextcloud/occ config:list system

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Tue Jan 30 06:25:02.064507 2018] [ssl:warn] [pid 573:tid 1992790016] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Tue Jan 30 06:25:03.001110 2018] [mpm_event:notice] [pid 573:tid 1992790016] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Tue Jan 30 06:25:03.001189 2018] [core:notice] [pid 573:tid 1992790016] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

```
[... 5 lines cut ...]
```

{"reqId":"ItQNZgdVcfEf7vrWM9Jr","level":3,"time":"2018-01-28T06:15:04+00:00","remoteAddr":"","user":"--","app":"appstoreFetcher","method":"--","url":"--","message":"Exception: {\"Exception\":\"GuzzleHttp\\\\Exception\\\\ConnectException\",\"Message\":\"cURL error 6: Could not resolve host: apps.nextcloud.com\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(103): GuzzleHttp\\\\Exception\\\\RequestException::wrapException(Object(GuzzleHttp\\\\Message\\\\Request), Object(GuzzleHttp\\\\Ring\\\\Exception\\\\ConnectException))\\n#1 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(132): GuzzleHttp\\\\RequestFsm->__invoke(Object(GuzzleHttp\\\\Transaction))\\n#2 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/react\\\/promise\\\/src\\\/FulfilledPromise.php(25): GuzzleHttp\\\\RequestFsm->GuzzleHttp\\\\{closure}(Array)\\n#3 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/ringphp\\\/src\\\/Future\\\/CompletedFutureValue.php(55): React\\\\Promise\\\\FulfilledPromise->then(Object(Closure), NULL, NULL)\\n#4 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Message\\\/FutureResponse.php(43): GuzzleHttp\\\\Ring\\\\Future\\\\CompletedFutureValue->then(Object(Closure), NULL, NULL)\\n#5 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(134): GuzzleHttp\\\\Message\\\\FutureResponse::proxy(Object(GuzzleHttp\\\\Ring\\\\Future\\\\CompletedFutureArray), Object(Closure))\\n#6 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Client.php(165): GuzzleHttp\\\\RequestFsm->__invoke(Object(GuzzleHttp\\\\Transaction))\\n#7 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Client.php(125): GuzzleHttp\\\\Client->send(Object(GuzzleHttp\\\\Message\\\\Request))\\n#8 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Http\\\/Client\\\/Client.php(138): GuzzleHttp\\\\Client->get('https:\\\/\\\/apps.ne...', Array)\\n#9 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/App\\\/AppStore\\\/Fetcher\\\/Fetcher.php(99): OC\\\\Http\\\\Client\\\\Client->get('https:\\\/\\\/apps.ne...', Array)\\n#10 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/App\\\/AppStore\\\/Fetcher\\\/AppFetcher.php(66): OC\\\\App\\\\AppStore\\\\Fetcher\\\\Fetcher->fetch('\\\"2018-01-26 21:...', '[{\\\"id\\\":\\\"spreedm...')\\n#11 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/App\\\/AppStore\\\/Fetcher\\\/Fetcher.php(162): OC\\\\App\\\\AppStore\\\\Fetcher\\\\AppFetcher->fetch('\\\"2018-01-26 21:...', '[{\\\"id\\\":\\\"spreedm...')\\n#12 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Installer.php(394): OC\\\\App\\\\AppStore\\\\Fetcher\\\\Fetcher->get()\\n#13 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/updatenotification\\\/lib\\\/Notification\\\/BackgroundJob.php(258): OC\\\\Installer::isUpdateAvailable('files_trashbin', Object(OC\\\\App\\\\AppStore\\\\Fetcher\\\\AppFetcher))\\n#14 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/updatenotification\\\/lib\\\/Notification\\\/BackgroundJob.php(155): OCA\\\\UpdateNotification\\\\Notification\\\\BackgroundJob->isUpdateAvailable('files_trashbin')\\n#15 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/updatenotification\\\/lib\\\/Notification\\\/BackgroundJob.php(79): OCA\\\\UpdateNotification\\\\Notification\\\\BackgroundJob->checkAppUpdates()\\n#16 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/BackgroundJob\\\/Job.php(59): OCA\\\\UpdateNotification\\\\Notification\\\\BackgroundJob->run(NULL)\\n#17 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/BackgroundJob\\\/TimedJob.php(54): OC\\\\BackgroundJob\\\\Job->execute(Object(OC\\\\BackgroundJob\\\\JobList), Object(OC\\\\Log))\\n#18 \\\/var\\\/www\\\/nextcloud\\\/cron.php(124): OC\\\\BackgroundJob\\\\TimedJob->execute(Object(OC\\\\BackgroundJob\\\\JobList), Object(OC\\\\Log))\\n#19 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Exception\\\/RequestException.php\",\"Line\":49}","userAgent":"--","version":"12.0.4.3"}
{"reqId":"ItQNZgdVcfEf7vrWM9Jr","level":3,"time":"2018-01-28T06:15:04+00:00","remoteAddr":"","user":"--","app":"appstoreFetcher","method":"--","url":"--","message":"Exception: {\"Exception\":\"GuzzleHttp\\\\Exception\\\\ConnectException\",\"Message\":\"cURL error 6: Could not resolve host: apps.nextcloud.com\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(103): GuzzleHttp\\\\Exception\\\\RequestException::wrapException(Object(GuzzleHttp\\\\Message\\\\Request), Object(GuzzleHttp\\\\Ring\\\\Exception\\\\ConnectException))\\n#1 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(132): GuzzleHttp\\\\RequestFsm->__invoke(Object(GuzzleHttp\\\\Transaction))\\n#2 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/react\\\/promise\\\/src\\\/FulfilledPromise.php(25): GuzzleHttp\\\\RequestFsm->GuzzleHttp\\\\{closure}(Array)\\n#3 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/ringphp\\\/src\\\/Future\\\/CompletedFutureValue.php(55): React\\\\Promise\\\\FulfilledPromise->then(Object(Closure), NULL, NULL)\\n#4 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Message\\\/FutureResponse.php(43): GuzzleHttp\\\\Ring\\\\Future\\\\CompletedFutureValue->then(Object(Closure), NULL, NULL)\\n#5 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(134): GuzzleHttp\\\\Message\\\\FutureResponse::proxy(Object(GuzzleHttp\\\\Ring\\\\Future\\\\CompletedFutureArray), Object(Closure))\\n#6 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Client.php(165): GuzzleHttp\\\\RequestFsm->__invoke(Object(GuzzleHttp\\\\Transaction))\\n#7 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Client.php(125): GuzzleHttp\\\\Client->send(Object(GuzzleHttp\\\\Message\\\\Request))\\n#8 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Http\\\/Client\\\/Client.php(138): GuzzleHttp\\\\Client->get('https:\\\/\\\/apps.ne...', Array)\\n#9 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/App\\\/AppStore\\\/Fetcher\\\/Fetcher.php(99): OC\\\\Http\\\\Client\\\\Client->get('https:\\\/\\\/apps.ne...', Array)\\n#10 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/App\\\/AppStore\\\/Fetcher\\\/AppFetcher.php(66): OC\\\\App\\\\AppStore\\\\Fetcher\\\\Fetcher->fetch('\\\"2018-01-26 21:...' [... cut ...] \\/private\\\/BackgroundJob\\\/TimedJob.php(54): OC\\\\BackgroundJob\\\\Job->execute(Object(OC\\\\BackgroundJob\\\\JobList), Object(OC\\\\Log))\\n#18 \\\/var\\\/www\\\/nextcloud\\\/cron.php(124): OC\\\\BackgroundJob\\\\TimedJob->execute(Object(OC\\\\BackgroundJob\\\\JobList), Object(OC\\\\Log))\\n#19 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Exception\\\/RequestException.php\",\"Line\":49}","userAgent":"--","version":"12.0.4.3"}

```
</details>

ovpc, 2018-01-31:
If no progress on this issue.
Thanks for reporting, please post terminal output of:

`df -h`
and
`ls -lh /media/`

rencire, 2018-02-01:
Not much progress.  Here's the output:

```
pi@nextcloudpi:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G  2.6G   12G  19% /
devtmpfs        460M     0  460M   0% /dev
tmpfs           464M     0  464M   0% /dev/shm
tmpfs           464M  6.3M  458M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1   41M   21M   21M  52% /boot
/dev/sda1       932G  346M  930G   1% /media/myCloudDrive
tmpfs            93M     0   93M   0% /run/user/1000

pi@nextcloudpi:~ $ ls -lh /media
total 20K
drwxr-xr-x 1 root root   32 Jan 22 08:22 myCloudDrive
drwxr-xr-x 2 root root 4.0K Jan 22 08:22 USBdrive
```

@ovpc any insights/potential leads to the problem?

nachoparker, 2018-02-05:
looking at your `ncp-report` output, your PHP is down

```
PHP service          down
```
What is the output of 

```
systemctl status php7.0-fpm
```

rencire, 2018-02-13:
Thanks for the tip.  I had to re-flash the image anyway, so did not need to troubleshoot this further.  Closing ticket.

closed 2018-02-13 by rencire