nextcloud/nextcloudpi #388: avoid ssh configuration sins

testbird, 2018-01-31:
I assume the shell scripts in /data/ncp are written by ncp-web, that they contain the state of the configuration options as well as (customiable?) functions, and that they are sourced for presenting the options and executing state changing functions. (My assumption may not be correct, if you could put an expaining comment into those files that would be nice.)

SSH.sh configure() automatically creates new user accounts but never removes them. So if one makes a test or configures temporal access with simple combination like "temp" password "temp", it won't be removed and continues to be active even is ssh is eventually configured with a proper account.

testbird, 2018-02-01:
The purpose of the file might also be more obvious if you move the scripts in the docker volume to /data/etc/nextcloudpi-config.d (relating to the regular image).

nachoparker, 2018-02-08:
Your assumptions are correct

> SSH.sh configure() automatically creates new user accounts but never removes them. So if one makes a test or configures temporal access with simple combination like "temp" password "temp", it won't be removed and will be re-activated if ssh is eventually configured with a proper account.

that's true... that feature creates, but doesn't delete of course, maybe we should re-think it and make it work without creating the user if it doesn't exist

> The purpose of the file might also be more obvious if you move the scripts in the docker volume into /data/etc/nextcloudpi-config.d/ (relating closer to the regular image location).

I guess we could do that :D

nachoparker, 2018-02-08:
In any case, I think we should ask ourselves... is that ( a bit twisted ) use case really sensible? I mean, if we create users that we consider for _temporary access_, then it should be our duty to disable them, delete them or invalidating their login before leaving

testbird, 2018-02-08:
I was surprised the username was actually changeable, so also consolidating this feature to just enable/disable and update the "ncp" user's password should be ok, I guess.

nachoparker, 2018-02-16:
Implemented `v0.46.11`. It will just fail if the user doesn't exist

closed 2018-02-16 by nachoparker