nextcloud/nextcloudpi #390: docker image: nc-scan-auto missing

testbird, 2018-02-01:
Please ship the nc-scan-auto feature with the docker image. 

The docker "ncdata" volume can easily be made accessible to other docker containers (--volume-from). For example to the syncthing/syncthing docker. Syncthing is much better for synchronization than nextclouds "csync". One would add syncthings sync folders in the ncdata volume as external filesystem folders to nextcloud, to have web access etc.

However, nextcloud still needs automatic periodical scanning to catch up with file changes, because nobody has merged the integrations yet: https://forum.syncthing.net/t/integrating-syncthing-with-owncloud/6332?source_topic_id=10499

nachoparker, 2018-02-11:
Sure, 

There's some functionality that relies on systemd timers. When adapting to docker, I realized that we don't have systemd in the docker container, only cron. 

Therefore some extras need adaptation. You can see the exclude list and the causes here

https://github.com/nextcloud/nextcloudpi/blob/master/update.sh#L45

testbird, 2018-02-12:
Ok, I see. Cron is more univerally portable.

nachoparker, 2018-02-12:
yes, well systemd doesn't work on docker at all

nachoparker, 2018-02-16:
I'll close and compile all enhancenments like this in a list #386

closed 2018-02-16 by nachoparker