nextcloud/nextcloudpi #391: NextCloudPi "hangs" every few weeks (see #353)

RalfJahns, 2018-02-01:
Hello,

so, it happened again. Took 18 days this time.

First my phone informs me, that it can not sync my calendar any more:
[image]

I try to login to the Web Gui:

[image]

Apache is obviously running.
I try to login to the management Web Gui on port 4443:

[image]

I try to login by ssh:

[image]

It doesn't tell me "wrong password" (or something like that), it closes the connection right away. I think that's not the normal behaviour.

So, down the stairs into the basement, to connect real Keyboard and Monitor to the Raspberry Pi:

[image]

But it does not output any video signal to the HDMI port. No matter what I do, the monitor keeps black. (Of course I pressed keys on the keyboard).

From my experience: Everything will be up and running fine, if I reboot. But last time I was told here not to do it.

Any idea???

Best regards
Ralf Jahns

new-on-github, 2018-02-03:
Same problem here. 

> From my experience: Everything will be up and running fine, if I reboot. But last time I was told here not to do it.<

Normaly this works for me, but last time I plugged the raspberry out, the nextcloud instance isn't working anymore. 

ncp-report says:
sudo ncp-report
<–! Paste this in GitHub report -->

details
NextCloudPi diagnostics

Cannot load Zend OPcache - it was already loaded
PHP Fatal error: Interface ‘OCP\L10N\IFactory’ not found in /var/www/nextcloud/lib/private/L10N/Factory.php on line 41

deljones, 2018-02-03:
Yes same here.. Every few days works fine then 'bang' no access.. A new one I've recently noticed is that. Also sometimes logging in on a browser, the norm is to go to Files, as set in config.php, sometimes it opens in Announcements, an app/service that I employ on my instance. Has anyone else seeing a different app opening other than the one set in config.php?

Del

new-on-github, 2018-02-03:
But a reboot helps always?

Because my instance isn't working any more since the last "down"-time

nachoparker, 2018-02-05:
Shutting down the pi in a non clean way is not a good idea. Soon there will be data corruption of some sort and things will stop working, or even we could damage the SD card.

I think that you need to have the monitor connected at the time of boot up for it to work on the pi. I don't know if this is an option, but you could leave an old monitor plugged in (turned off) and next time you will probably be able to access by plugging in a keyboard.

Still puzzled that some people are seeing this. My instance is always up for months with no issues.

Do you have good quality SD cards and power sources? that is the number one cause of "unexplainable" things normally

Need a `ncp-report` or similar to start digging

new-on-github, 2018-02-06:
Thanks for your help:

I have a transcend  sd card with 64GB. Should be ok. Or do you recommend something else?

I also tried 2 different power supplies. One from Samsung for mobile phones. What do you recommend?

I would try to give you a ncp-report, but my cloud after unplugging it no totally down:
https://help.nextcloud.com/t/problem-nexcloudpi-doesnt-start-anymore/26820

I think I have to set it up form scratch again...

Ricardosgeral, 2018-02-06:
this one is good
https://www.amazon.de/Raspberry-Offizielles-Pi-Netzteil-schwarz/dp/B01DP8O5A4/ref=sr_1_1?ie=UTF8&qid=1517920286&sr=8-1&keywords=raspberry+pi+3+power+supply

nachoparker, 2018-02-06:
yes, that's important... many use cheap chinese adaptors or even phone chargers and the pi is a fragile cheap board that needs a good power supply

if you want to check your SD card for damage, a simple way of doing this is explained [here](https://github.com/nextcloud/nextcloudpi/wiki/FAQ#why-is-nextcloud-so-slow--failing-randomly--getting-server-errors-even-with-a-fresh-image)

RalfJahns, 2018-02-08:
Added "hdmi_safe=1" to /boot/config.txt. This allows the pi to start without monitor and have a picture if you connect it later. Now I wait for the problem to occur again.

new-on-github, 2018-02-08:
Ok, I will try the recommended power supply. Hopefully it helps.

My external HDD has no external power supply, therefore the pi had to deliver the energy. But this should be a problem, or?

theCalcaholic, 2018-02-10:
Is there any reason to not set `hdmi_safe=1` by default?

nachoparker, 2018-02-14:
I'm going to close because we are unsure about wether it is a HW issue. We can still talk and we will reopen if there is something to fix.

closed 2018-02-14 by nachoparker