nextcloud/nextcloudpi #393: Cannot install from script

jsfrederick, 2018-02-04:
Trying to install on a Rock64 using Ayufan's Debian Stretch image using the nextcloudpi install script.  It fails.  I tried to run the script manually.  IT fails at this point:

root@AyufanTest:~# git clone -q --depth 1 https://github.com/nextcloud/nextcloudpi.git /tmp/nextcloudpi
fatal: unable to access 'https://github.com/nextcloud/nextcloudpi.git/': gnutls_handshake() failed: Error in the pull function.

Paste the output of `sudo ncp-report`

nachoparker, 2018-02-05:
hi, do you have internet access from the board?

jsfrederick, 2018-02-05:
Yes, I do.  I am going to bring up a new Raspberry Pi with Raspbian Lite and see if the script works.  Wondering if it's an issue with the Ayufan .06.19 image on the Rock64.  Will update you later this week once I've had a chance to test.

I see a gnutls_handshake error.  Maybe teh Rock64 image is missing some packages?

nachoparker, 2018-02-05:
yes, if you have connectivity then you must be missing some package I guess

I found (this)[https://askubuntu.com/questions/186847/error-gnutls-handshake-failed-when-connecting-to-https-servers#187199] but looks too complicated

Ino from the Armbian forums installed NCP on the ayufan image, so you can ask him for details.

BTW, do you know if the armbian image for Debian 9 is ready and usable?

jsfrederick, 2018-02-05:
Thanks, I'll check it out.

I do not know about the Armbian Stretch image.  I only see the Ubuntu image for Rock64.  Let me see what I can find out on this one.

jsfrederick, 2018-02-09:
Ok, an Update.

I tried this today on a Rpi and it worked fine.  So I went back to the Rock64 and it WORKED!  Must have been some issue on the GitHub side last week.  Will continue testing NextcloudPi on the Rock64 using Ayufan's Stretch images.  Hopefully we can call this working shortly.

closed 2018-02-09 by jsfrederick

nachoparker, 2018-02-09:
awesome

well, if you wanted to contribute a ready to use image based on ayufan (as long as armbian doesn't support it), I know of some people that would be quite happy ;D