nextcloud/nextcloudpi #394: Running nc-audit gives very long list of vulnerabilities

brotonmail, 2018-02-07:
Hi,

Today I ran nc-audit to check an NCP with lynis. I was shocked to see that there were so many unpatched vulnerabilities. I tried updating the system but I couldn't find any newer version of the affected packages. What can I do to mitigate?

Known vulnerabilities in this system
CVE-2017-8923 php7.0-mysql (remotely exploitable, high urgency)
CVE-2017-17087 vim-tiny (low urgency)
CVE-2017-16879 libncurses5 (remotely exploitable, medium urgency)
CVE-2017-10268 mariadb-common (low urgency)
CVE-2017-10378 mariadb-common (remotely exploitable, medium urgency)
CVE-2017-15365 mariadb-common
CVE-2017-3636 mariadb-common (medium urgency)
CVE-2017-3641 mariadb-common (remotely exploitable, medium urgency)
CVE-2017-3653 mariadb-common (remotely exploitable, low urgency)
CVE-2017-12613 libapr1 (low urgency)
CVE-2017-16879 ncurses-term (remotely exploitable, medium urgency)
CVE-2017-16879 libtinfo5 (remotely exploitable, medium urgency)
CVE-2017-15908 libudev1 (remotely exploitable, medium urgency)
CVE-2018-1049 libudev1
CVE-2018-1000021 git-man
CVE-2017-7516 cpio
CVE-2017-11462 libk5crypto3 (remotely exploitable, low urgency)
CVE-2018-5709 libk5crypto3 (remotely exploitable, medium urgency)
CVE-2018-5710 libk5crypto3 (remotely exploitable, medium urgency)
CVE-2017-10268 mariadb-client-10.1 (low urgency)
CVE-2017-10378 mariadb-client-10.1 (remotely exploitable, medium urgency)
CVE-2017-15365 mariadb-client-10.1
CVE-2017-3636 mariadb-client-10.1 (medium urgency)
CVE-2017-3641 mariadb-client-10.1 (remotely exploitable, medium urgency)
CVE-2017-3653 mariadb-client-10.1 (remotely exploitable, low urgency)
CVE-2017-9417 firmware-realtek (remotely exploitable, high urgency)
CVE-2017-11671 gcc-4.8-base (low urgency)
CVE-2017-8923 php7.0-readline (remotely exploitable, high urgency)
CVE-2016-10517 redis-tools (remotely exploitable, medium urgency)
CVE-2018-6594 python-crypto
CVE-2016-10228 libc6-dbg (remotely exploitable, medium urgency)
CVE-2017-1000408 libc6-dbg
CVE-2017-1000409 libc6-dbg
CVE-2017-12132 libc6-dbg (remotely exploitable, medium urgency)
CVE-2017-12133 libc6-dbg (remotely exploitable, medium urgency)
CVE-2017-15670 libc6-dbg (remotely exploitable, high urgency)
CVE-2017-15671 libc6-dbg (remotely exploitable, medium urgency)
CVE-2017-15804 libc6-dbg (remotely exploitable, low urgency)
CVE-2017-16997 libc6-dbg (remotely exploitable, high urgency)
CVE-2017-8804 libc6-dbg (remotely exploitable, high urgency)
CVE-2018-1000001 libc6-dbg
CVE-2018-6485 libc6-dbg
CVE-2017-15908 libpam-systemd (remotely exploitable, medium urgency)
CVE-2018-1049 libpam-systemd
CVE-2017-8923 php7.0-opcache (remotely exploitable, high urgency)
CVE-2017-15232 libjpeg62-turbo (remotely exploitable, low urgency)
CVE-2017-10268 mariadb-server-core-10.1 (low urgency)
CVE-2017-10378 mariadb-server-core-10.1 (remotely exploitable, medium urgency)
CVE-2017-15365 mariadb-server-core-10.1
CVE-2017-3636 mariadb-server-core-10.1 (medium urgency)
CVE-2017-3641 mariadb-server-core-10.1 (remotely exploitable, medium urgency)
CVE-2017-3653 mariadb-server-core-10.1 (remotely exploitable, low urgency)
CVE-2017-8923 php7.0-mbstring (remotely exploitable, high urgency)
CVE-2017-17087 xxd (low urgency)
CVE-2017-8923 php7.0 (remotely exploitable, high urgency)
CVE-2017-16879 libncursesw5 (remotely exploitable, medium urgency)
CVE-2017-12618 libaprutil1-dbd-sqlite3 (low urgency)
CVE-2017-3144 isc-dhcp-common
CVE-2016-2779 libuuid1 (high urgency)
CVE-2016-10209 libarchive13 (remotely exploitable, medium urgency)
CVE-2016-10349 libarchive13 (remotely exploitable, medium urgency)
CVE-2016-10350 libarchive13 (remotely exploitable, medium urgency)
CVE-2017-14166 libarchive13 (remotely exploitable, medium urgency)
CVE-2017-14501 libarchive13 (remotely exploitable, medium urgency)
CVE-2017-14502 libarchive13 (remotely exploitable, medium urgency)
[... 294 lines cut ...]
CVE-2018-6485 libc6-dev
CVE-2016-10517 redis-server (remotely exploitable, medium urgency)
CVE-2017-9525 cron (medium urgency)
CVE-2018-1000030 python2.7-dev
CVE-2016-10228 locales (remotely exploitable, medium urgency)
CVE-2017-1000408 locales
CVE-2017-1000409 locales
CVE-2017-12132 locales (remotely exploitable, medium urgency)
CVE-2017-12133 locales (remotely exploitable, medium urgency)
CVE-2017-15670 locales (remotely exploitable, high urgency)
CVE-2017-15671 locales (remotely exploitable, medium urgency)
CVE-2017-15804 locales (remotely exploitable, low urgency)
CVE-2017-16997 locales (remotely exploitable, high urgency)
CVE-2017-8804 locales (remotely exploitable, high urgency)
CVE-2018-1000001 locales
CVE-2018-6485 locales
CVE-2018-1000021 git
CVE-2018-1000030 libpython2.7-stdlib
CVE-2017-14062 libidn11 (remotely exploitable, high urgency)
CVE-2016-2568 libpolkit-gobject-1-0 (medium urgency)
CVE-2016-2779 util-linux (high urgency)
CVE-2017-11671 gcc-5-base (low urgency)
CVE-2018-1000030 python2.7-minimal
CVE-2016-2779 libsmartcols1 (high urgency)
CVE-2017-12618 libaprutil1-ldap (low urgency)

PascalHeynmoeller, 2018-02-07:
Most of them aren't relevant for Nextcloudpi.
Further Readings at https://ownyourbits.com/2017/12/23/security-audit-your-arm-board-with-lynis/

>>"... We can see that some of these warnings are taylored towards fat x86 servers and might not make sense in our home ARM setups where we don’t have resources to spare, and we have some different constrains and usage cases.
As with anything else in security, there are compromises to me made between security and inconvenience. For this reason, I add these rules to my custom profile ...."<<

This is about lynis but I think it's also an statement about the the log you posted from debsecan.
Also It would be good to make these checks on an updated system.

Addional readings on https://ownyourbits.com

Hope this helps.

nachoparker, 2018-02-08:
Yes, this only helps us realize how important security is and to take it seriously. In any case, there's little we can do, as this is the job of the Debian/Raspbian maintainers.

As long as we don't disable unattended upgrades not much more we can do, but it could be a useful development tool to analyze the vulnerabilities and think if there's ways in which some can be mitigated

closed 2018-02-08 by nachoparker