nextcloud/nextcloudpi #395: NextCloudPi updated to NC13

user-jan, 2018-02-08:
Hi all,
I recently updated my NCP to v0.46.3 and then saw Nacho's new article on https://ownyourbits.com. As a tech-layperson I would like to know whether my whole system is now fully updated or whether I need to re-install because on the "NextCloudPi system info" it still says:   

**Nextcloud version    12.0.3.3**

Thanks for any clarification on this matter :-)
Jan

**Here's the system info print out:**

[image]

Paste the output of `sudo ncp-report`

ovpc, 2018-02-08:
v0.46.3 is latest version of ncp-web and nextcloudpi-config
You can update from NC12.0.3.3 to NC12.0.4 (and 5)  through the admin panel of your Nextcloud instance, when branch is set to stable.
NC13.0.0 should appear there as soon as devs are confident about release being stable.

user-jan, 2018-02-08:
Great!!! Thanks alot for your help ;-)

testbird, 2018-02-08:
Too bad... docker image updates are missing

Still ncp 0.41.0 and nc 12.0.4, should that be updated through the NC internal updater? (non-automatically)

nachoparker, 2018-02-08:
please be patient. I just fixed them for NC13 and will upload them in a few hours after a bit of testing

updating the docker instance is not a finished and polished experience yet... sadly things go slow if there is not many people involved, but it will come!

closed 2018-02-08 by nachoparker

testbird, 2018-02-08:
> updating the docker instance is not a finished and polished experience yet...

No problem, what do you think about the idea to put the isnstructions in a script:
https://github.com/nextcloud/nextcloudpi/issues/381#issuecomment-362072170

I am a little confused about the ways to update NC itself. Are there actually (at least) two? Through ncp (new beta support) AND within NC? When to use which one?

nachoparker, 2018-02-08:
I still haven't read that issue, I'll get to it

right now you should update docker from NC itself. The other one you mention ( `nc-nextcloud` ) cleans your current instance and install some version (whatever you select) in its place, so it's not really an 'update'

nachoparker, 2018-02-08:
btw, docker NC13 already pushed for x86 and building for armhf

testbird, 2018-02-08:
Ah ok, so to understand the difference, the nc-nextcloud is like a "nc-reset" that would discard the configuration DB? 
And beta testing with docker would have been possible without nc-nextcloud, using the NC's build-in updater?

And doing another docker pull/run cycle would be comparable to getting ncp updates + the NC internal stable updates?

testbird, 2018-02-08:
A draft table with differences, for your Download page?

```
                           auto-updates             ncp-updates          NC-update chanels
NCP (NC for RPi)              system (raspbian)    ncp-web/nc-reinstall  yes
NCP-HC1 (NC for HC1)          system (armian)       ncp-web/nc-reinstall  yes
NCP-Docker (NC Provisioning)  system (?)             no                     yes (transient restart)
NCP-install (NC Provisioning)     ?                   ?                     ?
```

nachoparker, 2018-02-09:
> Ah ok, so to understand the difference, the nc-nextcloud is like a "nc-reset" that would discard the configuration DB?

yes... more or less.

The point of nc-init is to 'reset' the instance, so clean configuration, database... just as if you just installed.

The point of nc-nextcloud is to *download* a different nextcloud version. You can go back to NC12 if you want, or to NC14 when it comes out. By doing this, it also removes completely the current version, including database and is replaced by the downloaded version.

After running nc-nextcloud, your instance is *not* configured, so it doesn't have any base configuration like cron jobs or what database to use. It is not yet usable. You need to run nc-init afterwards to do this initial setup.

I hope it is clear now

> And beta testing with docker would have been possible without nc-nextcloud, using the NC's build-in updater?

At this moment yes. I need to work a bit more on making this more smooth in the future

> And doing another docker pull/run cycle would be comparable to getting ncp updates + the NC internal stable updates?
We need more work in order to be able to upgrade this way (docker pull). If we are jumping a major version ( like 12 to 13 ) it probably would break the instance. Sadly, at this moment it is best to export files, calendars, contacts..., clean the volume, pull and re-import.

> A draft table with differences, for your Download page?
Sounds good. Maybe for the wiki? it is open to anyone to edit. Probably in the Downloads page would be a good idea too.

I would like to find more people to help me with building and/or with some appliances (such as docker, or armbian), because my hands are a bit full and some things progress slow
---

Thanks!

testbird, 2018-02-09:
Thanks for your explanations! I think I got it now, luckily I didn't accidentially wipe my install as I was definitely confused.

I think I would have understood that if nc-nextcloud would be called nc-wipe-reinstall and nc-init would be called nc-wipe-reconfigure (and let nc-wipe-reinstall automatically call nc-wipe-reconfigure).

>>   And beta testing with docker would have been possible without nc-nextcloud, using the NC's build-in updater?

> At this moment yes. I need to work a bit more on making this more smooth in the future

Tried that now, and NC updated to v.13 without me noticing any problem. :-)

> If we are jumping a major version ( like 12 to 13 ) it probably would break the instance. Sadly, at this moment it is best to export files, calendars, contacts..., clean the volume, pull and re-import.

OK. What problems do you have in mind that could happen? In the run-time updated NC container, the only thing I noticed till now is that some Apps want a newer PHP, and the entire app is in /data/app, or not?.

> I would like to find more people to help me with building and/or with some appliances (such as docker, or armbian), because my hands are a bit full and some things progress slow

With the new versions out, maybe you could try to package the ncp-installer (and nc-* skripts) into a first crude "NC Provisioning (NCP)" .deb package. I mean just depend on the debian packages freedombox-setup and plinth, and start alongside their web stack.

Then contact the freedombox folks. I mean sure everybody would love to have NC in debian someday, but a NC provisioner seems an option for now, they have experience and expressed willingness to help. https://salsa.debian.org/freedombox-team/plinth/issues/1200#note_13425 (I'm sure they would install, test and help with a ncp.dep, further as well.) FBX has own image builder, so maybe there is room to collaborate there as well.

user-jan, 2018-02-11:
Hi Nacho,
Do I understand correctly that it is not enough to just run 'nc-update' if I want to go from NC12 to NC13, but that I have to do a new install of the latest NC version?
Cheers,
Jan

nachoparker, 2018-02-11:
@testbird 

Initially nc-nextcloud and nc-init were part of the same script, but there are cases where you want them separate so that's what I did. For example, the first time you run the docker container, nc-init is run

About packaging... there's already somebody packaging NC debs

https://help.nextcloud.com/t/stable-debian-repository/1662/50

> OK. What problems do you have in mind that could happen?

We need to think about updates... if we docker pull the NC13 version, database changes will not be taken care off, that's what I mean. It's better to run the NC updater for this reason at the moment.

@user-jan 

You don't have to do a new install, just upgrade to NC13 from the Nextcloud web. It will tell you that there is an update available, or otherwise go to settings as the admin user and upgrade from there.

testbird, 2018-02-12:
Since the run-time-only update to NC13 with the build-in updater should already have done the DB migrations, I stopped and renamed the container, then did a new pull and run.

But now the browser is not able to connect to ncp nor NC in the container, despite the run command not showing any error. 

Any idea what I should do?

nachoparker, 2018-02-12:
the database is in the `ncdata` volume, so if you went through the upgrade and it worked the updated database should be there

what volume did you use for the latest pulled container?

testbird, 2018-02-12:
I used `-v ncdata:/data`, and just verified with `docker exec -ti nextcloudpi bash` that I see the user dirs in /data/app/data/.

This directory also contains an error.log and it has such lines:
`[authz core:error] ... AH01630:: client denied by server configuration: /var/www/nextcloud/data/.ocdata`
The nextcloud dir is a symlink to /data/app, and .ocdata is a zero size file.

nachoparker, 2018-02-12:
it's normal that `.ocdata` is zero size.

in principle your idea _should work_, but without trying myself I wouldn't know where things are going wrong.

what do you get from 

```
sudo ncp-diag
```

what error message you get? are the services up?

nachoparker, 2018-02-14:
in order to make your apache work again, remove the non existing certificate lines from `/etc/apache@/sites-enabled`

the fix in #402 has already been pushed

testbird, 2018-02-14:
Great, only I did not get any update yet, using docker pull.

With the three SSL lines removed from the nexcloud.conf and ncp.conf the browser now shows:
SSL_ERROR_RX_RECORD_TOO_LONG

nachoparker, 2018-02-14:
Write these three lines

https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-nextcloud.sh#L192

testbird, 2018-02-14:
Ok, getting certs seems to have worked after apache restart.

But in ncp-conf I saw this, and the browser now shows an empty page instead of the nextcloud login (ncp-web works):

```
An unhandled exception has been thrown:
RedisException: Failed to AUTH connection in /data/app/lib/private/Memcache/Redis.php:54
Stack trace:
#0 /data/app/lib/private/Memcache/Redis.php(54): Redis->get('3a6d14f45e54b6f...')
#1 /data/app/lib/autoloader.php(146): OC\Memcache\Redis->get('OC_User')
#2 [internal function]: OC\Autoloader->load('OC_User')
#3 /data/app/lib/private/User/Session.php(214): spl_autoload_call('OC_User')
#4 /data/app/lib/private/legacy/app.php(334): OC\User\Session->getUser()
#5 /data/app/lib/private/legacy/app.php(113): OC_App::getEnabledApps()
#6 /data/app/lib/base.php(661): OC_App::loadApps(Array)
#7 /data/app/lib/base.php(1080): OC::init()
#8 /data/app/console.php(46): require_once('/data/app/lib/b...')
#9 /data/app/occ(11): require_once('/data/app/conso...')
#10 {main}An unhandled exception has been thrown:
RedisException: Failed to AUTH connection in /data/app/lib/private/Memcache/Redis.php:54
Stack trace:
#0 /data/app/lib/private/Memcache/Redis.php(54): Redis->get('3a6d14f45e54b6f...')
#1 /data/app/lib/autoloader.php(146): OC\Memcache\Redis->get('OC_User')
#2 [internal function]: OC\Autoloader->load('OC_User')
#3 /data/app/lib/private/User/Session.php(214): spl_autoload_call('OC_User')
#4 /data/app/lib/private/legacy/app.php(334): OC\User\Session->getUser()
#5 /data/app/lib/private/legacy/app.php(113): OC_App::getEnabledApps()
#6 /data/app/lib/base.php(661): OC_App::loadApps(Array)
#7 /data/app/lib/base.php(1080): OC::init()
#8 /data/app/console.php(46): require_once('/data/app/lib/b...')
#9 /data/app/occ(11): require_once('/data/app/conso...')
#10 {main}Done. Press any key...
```

testbird, 2018-02-14:
Another catch, but maybe not related to my current problem: The ncp user credentials were not saved persistently in /data.

testbird, 2018-02-14:
In /var/log/apache2/error.log I found:
`localhost:4443:0 server certificate does NOT include an ID which matches the server name`
and [lines like these for "localhost"](http://e-mats.org/2013/12/ah01753-access-check-of-127-0-0-1-to-xxx-failed-reason-unable-to-get-the-remote-host-name/).

Is there a reason you did not symlink the whole /etc to /data/etc?

testbird, 2018-02-14:
Now I dumped the container and retried after docker pull loaded a new image and redid the cert exchange, but now letsencrypt configuration fails with "No internet connectivity". I thought that test was gone and it now uses the return value to print a different message.

nachoparker, 2018-02-15:
we're close. Make sure that the password for redis in `/var/www/nextcloud/config/config.php` is the same as the one in `/etc/redis/redis.conf`

testbird, 2018-02-15:
Ok, needed to fix the file ownership in docker after copying the edited files, then redis worked.

The letsencrypt.sh worked after I manually applied the changes to the connection check. There is no /etc/nextcloudpi-conf.d/ symlink in docker, I had to edit /data/ncp. So updates are not applied there?

As I wrote the "Require host localhost" apache lines produced errors (name resulution fails), and may only slow down things, isnt't just "ip 127" better?

testbird, 2018-02-15:
Oh, and I saw /var/log/redis/redis-server.log has some general complains about some config options.

nachoparker, 2018-02-15:
cool, I'll have a look at those things next time I improve the docker container