nextcloud/nextcloudpi #396: request

wisdomlight, 2018-02-08:
Hi

I'd like to share my experience of using nextcloudpi website/github from the point of view of a complete beginner with some but not much linux technical expertise.

I am not sure if you are interested but I'd like to possibly offer my help - as a complete beginner on how to make the experience of using NextCloudpi more smooth for a beginner.

If you are interested - is there a more private way to discuss?   

Many thanks

Sherab

ovpc, 2018-02-08:
Hi there,
Thanks for your offer, all help is welcomed.
You can join us at [Telgram](https://t.me/NextCloudPi), which is public, and go private from there, if you want.
Or leave your comments/suggestions here or at help.nexcloud.com when appropriate.

nachoparker, 2018-02-08:
hi @wisdomlight 

thank you very much for your offer... we need feedback from more "beginner" and less "highly technical" people in order to understand better where things are not clear.

The correct channels to talk about this are the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm), as well as the Telegram channel `@NextCloudPi`

closed 2018-02-08 by nachoparker