nextcloud/nextcloudpi #400: fix uPnP with UFW

nachoparker, 2018-02-09:
Still have to verify this, but I have a report that it doesn't work. Probably forgot to open the required ports

PascalHeynmoeller, 2018-02-13:
I reported this to you on Telegram but I thought about it and don't think its an bug, because UFW is useful if ncpi is an exposed Host, then I didn't want to have UPNP ports opened.

Else UPNP on ncpi is used behind an NAT Router to open ports of the Router (NAT) packet filter while UFW isn't needed because it's protected by the NAT and packet filter of the router.

For me it make sense to not open the UPNP ports when UFW is in use.
I stumbled about it, but it's ok because this makes sense to me.

Maybe a hint should be there or activating UPNP on ncpi-web should check if UFW is running and open the ports temporarily for UPNP.

theCalcaholic, 2018-02-14:
@PascalHeynmoeller Well, they can work together, so if this problem exists it should be fixable.

nachoparker, 2018-02-14:
Thanks both for your opinions. I also think it can  be easily fixed, so that all of NCP features work well together.

It's only a matter of opening the UPnP ports

nachoparker, 2018-02-19:
it should be a matter of using a rule similar to

```
sudo ufw allow proto udp from 192.168.0.0/24 port 1900:1901 to any port 1900:1901
```

nachoparker, 2018-02-19:
fixed `v0.46.22`

I had to open all UDP port communication in the local network, because unicast UDP responses were on random ports, even though multicast UDP announcements were on port 1900

closed 2018-02-19 by nachoparker