nextcloud/nextcloudpi #402: LetsEncrypt Certs not available after docker-compose up / down

Aeon512, 2018-02-09:
Hi, not sure if I'm doing something wrong but lets start from scratch

1. `docker-compose -f <docker-compose-arm.yml> up -d `
2. access ncp and enable lets encrypt
3. access via https and see that its works
4. `docker-compose -f <docker-compose-arm.yml> down `
5. `docker-compose -f <docker-compose-arm.yml> up -d `

When looking at `dockers logs` I see a PHP stack-trace but the actual error at the very beginning is an error in Apache loading the configuration since the /etc/letsencrypt/live folder with the certificates no longer exists.

As far as I can tell, the step 2) creates the necessary certificate data in /etc/letsencrypt/live and /etc/letsencrypt/archive but these folders are not saved via the library functions in the permanent /data/etc folder.

Wouldn't it make more sense to save this folder permanently as well?

nachoparker, 2018-02-10:
good catch.

you have to add a line to the `000ncp` file

Aeon512, 2018-02-10:
Sure I'm here to help :-)
But isn't the `000ncp` file also part of the docker-image and hence changes to it are not persistent?

nachoparker, 2018-02-10:
you are absolutely right, the code is there (didn't remember), but what we need is to activate it when launching _letsencrypt_

Aeon512, 2018-02-10:
I just couldn't find anything in the early startup phase where you check for an active letsencrypt configuration

in particular since the `is_active()` method in the file mentioned above just tests for the existence of `/etc/letsencrypt/live` which yields an hen-egg problem :-)

If you have something which I should test, or if I can assist otherwise, let me know.

nachoparker, 2018-02-10:
you are right... the problem is that we need to persist the whole `/etc/services-enabled` folder ( with the `_persistent_cfg_()` trick that you have already seen, because otherwise, once you remove the container it won't remember what services should be enabled

if you feel like playing with this you can add that to the _debian-ncp_ container (or wherever it's best, talking from memory) and rebuild the container. It should work.

there's instructions on ownyourbits.com, basically you do

```
make nextcloudpi-x86
```
and that will rebuild with the changes

Aeon512, 2018-02-10:
I would vote to add it in `nextcloudpi/docker-armhf/debian-ncp/run-parts.sh` and persist the cfg in the enabled / disabled case as well as just before iterating through the enabled services.

However, the only interaction with docker is on my raspberry pi, so I guess it has to be `armhf` but also needs to be somehow cross compiled or?

nachoparker, 2018-02-11:
> But it letsencrypt actually enabled? Only seen it the available section

You are right, it needs to be "enabled". Probably the best place is in `letsencrypt.sh`, at the end of the install() section.

> However, the only interaction with docker is on my raspberry pi, so I guess it has to be armhf but also needs to be somehow cross compiled or?

I suggest you test it in docker x86 because it's faster. When you are happy with the result we will merge it and we can generate the armhf version.

If you want to "develop" in the pi you would have to 

- login in the pi
- install docker
- clone the repo
- make changes in the repo
- build the armhf image with `make`

nachoparker, 2018-02-11:
I appreciate the help. My hands are pretty full with all the different platforms and it's hard to keep up.

Aeon512, 2018-02-11:
I tried to change some files, e.g.
- `docker-armhf/debian-ncp/run-parts.sh`
- `docker-armhf/nextcloudpi/000ncp`
- `etc/nextcloudpi-config.d/letsencrypt.sh`

and then tried to build it with `make nextcloudpi-x86`. However, when I run this image, I don't see that changes aas made in the files above.
E.g. the `\run_parts.sh' is still the old one, the `service-available.d\100letsencrypt` which gets created by the file above has still the old contents, etc. Do you know why? What am I doing wrong?

nachoparker, 2018-02-11:
Did it build everything? it takes a while. You can inspect the Makefile to see the different steps.

> The corresponding Dockerfile referenced the docker and not the docker-armhf directory.

Yes, that's wrong, but actually it worked fine because both files are identical. Should be changed, though.

Aeon512, 2018-02-11:
Could you give me a hint where you `install` the `letsencrypt.sh`?

nachoparker, 2018-02-11:
yes, in `nextcloudpi.sh`, it calls the remote update from github

https://github.com/nextcloud/nextcloudpi/blob/master/nextcloudpi.sh#L174

nachoparker, 2018-02-13:
fixed by PR #407 

thanks!

closed 2018-02-13 by nachoparker

testbird, 2018-02-13:
Ok, looks as if at least part of my problem from https://github.com/nextcloud/nextcloudpi/issues/395 is this issue.

I found apache and mariaDB where not up. mysql service could be started manually, but apache failed missing certs.

Thanks to you! I just wait until the image is updated, but I guess I need to request new certs, how can I do that, and work around apache not starting up?

testbird, 2018-02-16:
Thank you, got it working that way. 
(In case you get into the same situation, there were some more things I posted in #395)