nextcloud/nextcloudpi #404: letsencrypt timeout issue

brotonmail, 2018-02-10:
My letsencrypt cert recently expired. I was surprised that it didn't autorenew so I tried to do it myself. Here is the error I received:

Launching letsencrypt
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Cert is due for renewal, auto-renewing...
Renewing an existing certificate
Performing the following challenges:
http-01 challenge for <censored>
Using the webroot path /var/www/nextcloud for all unmatched domains.
Waiting for verification...
Cleaning up challenges
Failed authorization procedure. <censored> (http-01): urn:acme:error:connection :: The server could not connect to the client to verify the domain :: Fetching http://<censored>/.well-known/acme-challenge/HulZw-NjvOOBx7ctPVn1GkpLC7cRgDJ4Rfn6D8hi1jI: Timeout

IMPORTANT NOTES:
 - The following errors were reported by the server:

   Domain: <censored>
   Type:   connection
   Detail: Fetching
   http://<censored>/.well-known/acme-challenge/HulZw-NjvOOBx7ctPVn1GkpLC7cRgDJ4Rfn6D8hi1jI:
   Timeout

   To fix these errors, please make sure that your domain name was
   entered correctly and the DNS A/AAAA record(s) for that domain
   contain(s) the right IP address. Additionally, please check that
   your computer has a publicly routable IP address and that no
   firewalls are preventing the server from communicating with the
   client. If you're using the webroot plugin, you should also verify
   that you are serving files from the webroot path you provided.
Done. Press any key...

I checked my Firewall and port 80 seems to be open and NATed. Any idea what causes this?

theCalcaholic, 2018-02-10:
The domain name seems to be misconfigured. The output should contain `http://<your-domain-name>/well-known/acme-challenge/<token>`. In your case domain name appears to be blank. Is it correctly displayed in the DOMAIN field under letsencrypt in the web panel?

brotonmail, 2018-02-10:
Yes sorry I had it as <xxx> so it didn't show up. Just changed.

theCalcaholic, 2018-02-10:
What does `/var/log/letsencrypt/letsencrypt.log` read after the failed attempt?

nachoparker, 2018-02-11:
are your ports open? please, always include the output of

```
sudo ncp-report
```
, or in ncp-web

```
nc-info
```

brotonmail, 2018-02-11:
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v0.46.6
NextCloudPi image    untagged
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          none
datadir              /var/nextcloud/datadir
data in SD           yes
data filesystem      ext2/ext3
data disk usage      7.7G/15G
rootfs usage         7.7G/15G
swapfile             /var/swap
Nextcloud check      ok
Nextcloud version    12.0.0.29
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   10.0.0.2
gateway              10.0.0.1
interface            enxb827ebc2e8f5
certificates         xxx 
certs due            none
NAT loopback         no
uptime               1day

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "10.0.0.2",
            "2": "xxx",
            "4": "xxx"
        },
        "datadirectory": "\/var\/nextcloud\/datadir",
        "overwrite.cli.url": "https:\/\/xxx",
        "dbtype": "mysql",
        "version": "12.0.0.29",
        "dbname": "nextcloud",
        "dbhost": "localhost",
        "dbport": "",
        "dbtableprefix": "oc_",
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "ocbx7lnvodo4",
        "loglevel": "2",
        "log_type": "file",
[... 34 lines cut ...]
```

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"WTsGUn8AAQEAAAQ4lJYAAA8W","level":2,"time":"2017-06-09T20:34:26+00:00","remoteAddr":"192.168.14.27","user":"admin","app":"core","method":"GET","url":"\/index.php\/settings\/users","message":"Login failed: 'admin' (Remote IP: '192.168.14.27')","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10.12; rv:53.0) Gecko\/20100101 Firefox\/53.0","version":"12.0.0.29"}
{"reqId":"WTwAGn8AAQEAAAZt100AAEwP","level":4,"time":"2017-06-10T14:20:21+00:00","remoteAddr":"79.253.39.136","user":"admin","app":"webdav","method":"PUT","url":"\/remote.php\/dav\/addressbooks\/users\/admin\/contacts\/3b5d7e27-c493-404b-b8f2-2c0e509c80f5.vcf","message":"Exception: {\"Message\":\"HTTP\\\/1.1 415 This resource only supports valid vCard or jCard data. Parse error: End of document reached prematurely\",\"Exception\":\"Sabre\\\\DAV\\\\Exception\\\\UnsupportedMediaType\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/CardDAV\\\/Plugin.php(316): Sabre\\\\CardDAV\\\\Plugin->validateVCard('BEGIN:VCARD\\\\r\\\\nVE...', false)\\n#1 [internal function]: Sabre\\\\CardDAV\\\\Plugin->beforeCreateFile('addressbooks\\\/us...', 'BEGIN:VCARD\\\\r\\\\nVE...', Object(OCA\\\\DAV\\\\CardDAV\\\\AddressBook), false)\\n#2 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/event\\\/lib\\\/EventEmitterTrait.php(105): call_user_func_array(Array, Array)\\n#3 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(1093): Sabre\\\\Event\\\\EventEmitter->emit('beforeCreateFil...', Array)\\n#4 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/CorePlugin.php(525): Sabre\\\\DAV\\\\Server->createFile('addressbooks\\\/us...', 'BEGIN:VCARD\\\\r\\\\nVE...', NULL)\\n#5 [internal function]: Sabre\\\\DAV\\\\CorePlugin->httpPut(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#6 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/event\\\/lib\\\/EventEmitterTrait.php(105): call_user_func_array(Array, Array)\\n#7 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(479): Sabre\\\\Event\\\\EventEmitter->emit('method:PUT', Array)\\n#8 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(254): Sabre\\\\DAV\\\\Server->invokeMethod(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#9 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Server.php(253): Sabre\\\\DAV\\\\Server->exec()\\n#10 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/appinfo\\\/v2\\\/remote.php(33): OCA\\\\DAV\\\\Server->exec()\\n#11 \\\/var\\\/www\\\/nextcloud\\\/remote.php(162): require_once('\\\/var\\\/www\\\/nextcl...')\\n#12 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/CardDAV\\\/Plugin.php\",\"Line\":356,\"User\":\"admin\"}","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10_12_5) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/59.0.3071.86 Safari\/537.36","version":"12.0.0.29"}
{"reqId":"WTxqin8AAQEAAAZlrbgAABMJ","level":4,"time":"2017-06-10T21:54:19+00:00","remoteAddr":"84.16.226.51","user":"lenard","app":"webdav","method":"MKCOL","url":"\/remote.php\/webdav\/Senior%20Year%20Memories\/Ethan","message":"Exception: {\"Message\":\"HTTP\\\/1.1 405 The resource you tried to create already exists\",\"Exception\":\"Sabre\\\\DAV\\\\Exception\\\\MethodNotAllowed\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/CorePlugin.php(590): Sabre\\\\DAV\\\\Server->createCollection('Senior Year Mem...', Object(Sab [... cut ...] acintosh) mirall\/2.2.4 (build 1) (Nextcloud)","version":"12.0.0.29"}
{"reqId":"WT1F7H8AAQEAAARuPLgAAABI","level":3,"time":"2017-06-11T13:30:21+00:00","remoteAddr":"79.253.20.9","user":"lenard","app":"PHP","method":"MOVE","url":"\/remote.php\/webdav\/Location%20Of%20Oil%20Fields.docx.bc","message":"Undefined offset: 3 at \/var\/www\/nextcloud\/apps\/activity\/lib\/FilesHooks.php#620","userAgent":"Mozilla\/5.0 (Macintosh) mirall\/2.2.4 (build 1) (Nextcloud)","version":"12.0.0.29"}

```
</details>

theCalcaholic, 2018-02-11:
Are any requests to /.well-known/... documented in /var/log/apache2/error.log or in /var/log/apache2/access.log after the attempt?

If not, are you sure you configured dns correctly (can you reach your cloud under that domain) and are there definitely no typos in the url in the letsencrypt configuration?

nachoparker, 2018-02-11:
your ports are open, so it must be a DDNS problem. What's your provider? no-ip? freeDNS?

I also wonder why you have that funky interface name :P

brotonmail, 2018-02-11:
I've configured my DDNS with nsupdate.info. I checked the site and it seemed it was pointing to the same IP Address as I got when I entered "curl ipinfo.io". I'll try switching providers and if that doesn't work I'll try reinstalling on a fresh image.

nachoparker, 2018-02-11:
that's weird. It's going to be one of

- network connectivity outside ( can you ping google from the pi? )
- DDNS config ( seems ok )
- port forwarding

about this last one, it's worth mentioning that we use an online service to check if the ports are open. This service *caches* the results, so if you now close the ports, the check will return a false positive.

Maybe your router configuration reseted, or was reseted by your ISP? It's worth checking.

If you know your public IP you can try accessing by IP from outside ( for instance, mobile phone not connected to wifi ).

Also, something that can happen is that your ISP internally changed you to be behind a CG-NAT. In that case you need to contact them and ask them to take you out of the CG-NAT range. Not all providers will do this for you.

brotonmail, 2018-02-12:
Hey guys. Just double checked my Firewall. I had put in a rule allowing HTTP but hadn't NATed it properly. NextcloudPi was not at fault, it was my own stupidity :D Thank you for your great work!

theCalcaholic, 2018-02-12:
Glad you could solve it. :D

closed 2018-02-12 by theCalcaholic