nextcloud/nextcloudpi #406: Apache - server reached MaxRequestWorkers

nero120, 2018-02-11:
Hi guys, I upgraded to NextCloud 13 yesterday and NCP 0.46.6, the upgrades went well but now I'm noticing that NextCloud becomes unresponsive after a short time - attempting to access the website (or the NCP admin site) just hangs. I checked the apache logs and I see the following:

```
[mpm_event:error] [pid 921:tid 1991921664] AH00484: server reached MaxRequestWorkers setting, consider raising the MaxRequestWorkers setting
```

as well as lots of `The timeout specified has expired` and `child process xxxx still did not exist, sending a SIGTERM`.

Restarting the apache service does not bring back NextCloud, but rebooting the machine does bring the sites back. But a short while afterwards the same errors return.

I looked into `MaxRequestWorkers` but apprently it's configured within the `/etc/apache2/mods-enabled/mpm_prefork.conf` file which doesn't exist on my installation.

Any ideas?

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v0.46.6
NextCloudPi image    NextCloudPi_09-12-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/lambdaCloudData/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      48G/916G
rootfs usage         2.0G/2.7G
swapfile             /media/lambdaCloudData/swap
Nextcloud check      ok
Nextcloud version    13.0.0.14
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.16
gateway              192.168.1.1
interface            eth0
certificates         lambda-cloud.online 
certs due            88 days 
NAT loopback         no
uptime               2:50

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.1.16",
            "4": "lambda-cloud.online"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "https:\/\/lambda-cloud.online",
        "dbtype": "mysql",
[... 103 lines cut ...]
```

{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Run OCA\\Files\\BackgroundJob\\CleanupFileLocks job with ID 7","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Finished OCA\\Files\\BackgroundJob\\CleanupFileLocks job with ID 7 in 0 seconds","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Run OCA\\Activity\\BackgroundJob\\EmailNotification job with ID 10","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Finished OCA\\Activity\\BackgroundJob\\EmailNotification job with ID 10 in 0 seconds","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Run OCA\\Files_Versions\\BackgroundJob\\ExpireVersions job with ID 12","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Finished OCA\\Files_Versions\\BackgroundJob\\ExpireVersions job with ID 12 in 0 seconds","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Run OCA\\Files_Trashbin\\BackgroundJob\\ExpireTrash job with ID 14","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Finished OCA\\Files_Trashbin\\BackgroundJob\\ExpireTrash job with ID 14 in 0 seconds","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Run OC\\Authentication\\Token\\DefaultTokenCleanupJob job with ID 15","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Invalidating session tokens older than 2018-02-10T20:30:03+00:00","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Invalidating remembered session tokens older than 2018-01-27T20:30:03+00:00","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Finished OC\\Authentication\\Token\\DefaultTokenCleanupJob job with ID 15 in 0 seconds","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Run OCA\\UpdateNotification\\ResetTokenBackgroundJob job with ID 2517","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Finished OCA\\UpdateNotification\\ResetTokenBackgroundJob job with ID 2517 in 0 seconds","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Run OC\\Authentication\\Token\\DefaultTokenCleanupJob job with ID 2522","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Invalidating session tokens older than 2018-02-10T20:30:03+00:00","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Invalidating remembered session tokens older than 2018-01-27T20:30:03+00:00","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Finished OC\\Authentication\\Token\\DefaultTokenCleanupJob job with ID 2522 in 0 seconds","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Run OC\\Log\\Rotate job with ID 2523","userAgent":"--","version":"13.0.0.14"}
{"reqId":"eY0jpJgWZPwxVZJhXyWg","level":0,"time":"2018-02-11T20:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Finished OC\\Log\\Rotate job with ID 2523 in 0 seconds","userAgent":"--","version":"13.0.0.14"}

```
</details>

nachoparker, 2018-02-13:
have you been playing with nc-limits? what's the value of `pm.max_children`, `spare_servers` and `start_servers` in `/etc/php/7.0/fpm/pool.d/www.conf`

nero120, 2018-02-15:
@nachoparker thanks for the response. No I hadn't touch nc-limits, but strangely the system is working fine now without me having changed anything. Perhaps it was a side affect of the upgrade? In any case it seems stable and I haven't seen the error in the log for a few days so no much point in keeping this open. 😄

nero120, 2018-02-20:
@nachoparker I've noticed this is continuing to happen intermittently each time the RP3 is rebooted. Anything look strange about these values?

```
pm.max_children = 3
pm.start_servers = 2
pm.min_spare_servers = 1
pm.max_spare_servers = 3
```

Thanks.

nero120, 2018-02-20:
Having done some experimentation, I believe this only occurs once I start browsing photos, if I just navigate files the issue does not occur and the site does not stop responding. I have noticed over time that nextcloud is not particularly efficient or effective at displaying photos (when compared to google drive or onedrive), which is one of the main reasons I run a nextcloud server.

I noticed [this](https://github.com/nextcloud/server/issues/6459) issue in the nextcloud server repo relating to redis server, could very well be what I'm experiencing.

nachoparker, 2018-02-21:
Well... it seems than indeed the photo handling is not very well optimized. Most people find problem with preview generation when there are many new (big) images, but it seems like you are also having problems displaying pictures.

Can't talk about NC13 but in the past I have never had issues looking at my photos like what you are experiencing.

In any case, maybe letting nextcloud-server people that there is room for improvement here would not be a bad idea.

Actually those PHP values came from [this issue](https://github.com/nextcloud/nextcloudpi/issues/146)

nero120, 2018-02-21:
Ah that's interesting. I think the PHP values work well, in general I haven't had issues with image generation, I use the [Preview Generator](https://github.com/rullzer/PreviewGenerator) app and I have thousands of large photos, generating previews doesn't seem to kill the RP3 with this configuration. However, as you say, viewing photos does seem to degrade the performance until the site simply dies. Weirdly, even restarting the apache service doesn't bring it back, I have to restart the Pi.

As far as I'm concerned, NCP has been a game changer for me, making the process of running a personal nextcloud server far easier and more practical for someone who is not as experienced in running linux servers. Thanks very much for all your hard work @nachoparker, now if only the nextcloud devs could sort out the gallery I'd be a happy camper! 😄

nachoparker, 2018-02-21:
thanks, that makes me happy

well, it would be interesting to see what is exactly happening, but it doesn't happen to me so I can't reproduce it.

have you tried running `sudo iotop` and the load average ( `uptime` or `top` ) to see if the system is stalled in I/O? that might be the case, because restarting apache does not solve the issue.

nero120, 2018-02-21:
No I haven't actually, thanks, I'll give it a try when I get home from work tonight.

rmufr, 2018-02-27:
Hi @nero120, @nachoparker ,
Maybe you'd have a look at that one #7269
NC server triggers full preview folders rescan when browsing galleries or uploading files (sync client or webUI)
Not directly related to the PreviewGenerator app (I disabled it) but the more previews you have the more performance downgrade you will experience, up to clients time-outs.

nero120, 2018-02-27:
Thank you @rmufr, that's it. I've subscribed to that issue so I'll follow there from now on.

nachoparker, 2018-02-27:
suscribed too. 99% of the reports here are about NCP hogging down while managing pictures.

nero120, 2018-02-27:
@nachoparker manging my photo library is the main reason I run a nextcloud instance, and I'm sure it's a key feature for many users, it really does need some significant time investment from the NC dev team. Given the complexity of the entire application stack I'm not surprised many issues get logged in the wrong repo!

nero120, 2018-03-08:
@nachoparker I'm still getting the issue where both the NC Web UI and the NCP admin site load initially but then shortly afterwards will stop responding then eventually both return 503 errors.

`top` doesn't show anything untoward:

[image]

and nothing in the apache error log:

[image]

(I just restarted the service but no errors since it restarted, yet both sites are still failing to respond).

This can't be related to images as this occurs when I haven't even browsed any images, either via the Web UI or via apps. I haven't customised the config outside of what you get with the standard NCP build, currently on v0.46.6.

I'm not an experienced linux admin, but I don't understand why both sites would fail to respond when there are no errors in the log and the resources on the server aren't being taxed. Is anyone else seeing these kinds of issues, or provide any insight as to how I might diagnose the cause?

nachoparker, 2018-03-14:
when you are in this state, you can try to read the database, redis and php logs, and also systemctl status and journalctl to see if we discover something

seems like it is not an apache issue, so maybe try restarting database alone, or database+php in this order

```
systemctl stop php7.0-fpm
systemctl stop mysqld

systemctl start php7.0-fpm
systemctl start mysqld
```

nero120, 2018-03-14:
@nachoparker boom! Restarting just the php daemon brings the sites back up immediately! Is there some configuration change that could prevent the service from becoming stuck, or do you think upgrading php to 7.2 would help to mitigate this?

Sorry to keep hounding you on this but it feels like we're getting closer to sorting this. 😄

nachoparker, 2018-03-15:
> Sorry to keep hounding you on this but it feels like we're getting closer to sorting this. smile

no problem, it would make me feel better if this gets fixed

whenever this happens again, look at the php logs, and `systemctl status php7.0-fpm`

nero120, 2018-03-15:
Thanks @nachoparker. I see the following error in the php log:

```
WARNING: [pool www] server reached pm.max_children setting (3), consider raising it
```

After a bit of reading around the web of people having this issue of their pis, I've upped the numbers in `/etc/php/7.0/fpm/pool.d/www.conf` to:

```
pm.max_children = 7
pm.start_servers = 4
pm.min_spare_servers = 2
pm.max_spare_servers = 4
```

Just trying this config to see if it works better in my particular situation. Will report back next week with the results.

nachoparker, 2018-03-16:
Yes, those values were tuned after a similar issue #146 

Let us know! if you find better default values we'll change defaults

nachoparker, 2018-03-18:
I'll close for now because I still don't see any clear thing that we need to fix. We can still talk here of course.

nero120, 2018-03-20:
Ok so I've played with the numbers, and I'm currently at:

```
pm.max_children = 10
pm.start_servers = 4
pm.min_spare_servers = 4
pm.max_spare_servers = 8
```

I upped the numbers gradually, but each time I would see `WARNING: [pool www] server reached pm.max_children setting, consider raising it` in the php log after a short time. Note, this is under very light usage, one user and just browsing, no uploads.

These numbers appeared to work best so far, with the sites (NC and NCP config) available until this morning and now they are timing out. The thing that does my head in is that running `top` shows the cpu barely under any load! Perhaps the issue lies with the db but I simply don't know much about the db architecture or how to profile it.

@nachoparker if you can give me some pointers as to where to start I'll reset the php numbers back to default and have a play with the db config and see if this garners better results. I mean, considering my pi is getting no requests (whilst I'm asleep!) you'd think it wouldn't be falling over! 😉

nachoparker, 2018-03-20:
I don't see that message in `/var/log/apache2/*.log /var/log/php7.0-fpm.log`. Probably not a good idea to have more threads that CPUs in my opinion, because the thread context switch can get the pi stuck badly.

I wonder why you are seeing that. I am just browsing my files, using the news app and the tasks app and I don't get that.

Also, I don't think it's database related. That warning sounds like at some point, NC is trying to open more PHP threads than we are allowing, so probably that means that some threads have to wait, that's all.

nero120, 2018-03-20:
It's strange. I don't tax my pi at all, I'm the only user and under the default config both sites are unresponsive soon after the first requests are received by apache. Top shows no load, but there must be background processes that are holding up php and causing the max_children issue - could it be redis?

nachoparker, 2018-03-20:
not redis ( we all are using redis... ) just NC I guess :S 

what does the `wa` field in `top` say?

nero120, 2018-03-21:
[image]

When using the NC web UI, between 0 and 3 for normal browsing, spikes up to around 12 when browsing my photo library in gallery mode. The good thing about these new config settings is if I select Gallery and browse to my photo library root folder, it is super quick to render 144 folders with the preview images! The whole web UI feels much more snappy, the only problem is that eventually I get the  max_children issue. Would be great to know what in the pipeline is causing this. 😕

nachoparker, 2018-03-22:
That seems to be an improvement, does your instance become slow/unusable/unresponsive/dead when you see the warning? or nothing "bad" really happens?

If nothing happens, it could be a good idea to measure page load time (you can hit F12 in firefox for the dev tools) with and without cache with different numbers and see what's the best response time.

My first intuition is that same children threads as CPUs would be the optimal setting because there are fewer process context switches. That being said, because there is some I/O wait ( you mentioned `wa=12` ), maybe having some more threads would benefit us so other processing can be made while some of the threads are stuck waiting for the data.

What in any case I would do is to find some way of measuring performance
- `wa`
- CPU
- time to load the file browser
- time to load a galery page

Maybe you could try to measure that? it would be interesting.

nachoparker, 2018-03-22:
I'll actually reopen, other people might be interested

nero120, 2018-03-24:
Usually when I see the warning, that's when the sites start timing out. So I'm definitely finding that with these settings it takes longer before I see the warning and the sites start timing out (days rather than minutes), and that when browsing the NC Web UI, the file view and gallery view both feel much more snappy than with the current default config.

The problem is I'm not familiar enough with this architecture to understand where the bottleneck is occurring that causes the sites to become unresponsive. It definitely has something to do with PHP FPM as resetting the PHP service brings the sites back immediately, I just don't know what the root cause is and none of the metrics or logs I can see really point me to anything obvious.

For example, the sites has now become unresponsive, any requests to them time out and my NC sync client can't connect. If I now run `top` this is what I see:

[image]

Nothing obvious from what I can see, CPU looks fine, memory using more swap but otherwise fine. Interestingly I don't have *any* recent errors or warnings in the php log, and nothing but timeout errors in the apache log and nextcloud log. So what could be causing PHP FPM to block apache? As soon as I restart PHP, I see many more PHP FPM processes appear in top - could that mean when the sites are unresponsive, all the available FPM processes have gone to sleep and refusing to handle any more requests, which causes the sites to timeout?

nachoparker, 2018-03-24:
maybe you can try playing with `max_execution_time` and `max_input_time` in `/var/www/nextcloud/.user.ini`

nero120, 2018-03-24:
> maybe you can try playing with max_execution_time and max_input_time in /var/www/nextcloud/.user.ini

Having done some reading and playing around with `top`, I can see that most of the php-fpm processes are idle, yet they are still not receiving any requests to process. Changing the execution time might not make a difference in this case. I'm just testing an alternative config which is looking good so I'll report back tomorrow.

nero120, 2018-03-24:
Actually I don't need to wait until tomorrow! So I switched `pm` from `dynamic` to `ondemand` as I wanted the processes to be killed when they were sat idle for a short period, in this case 10 secs by virtue of setting `pm.process_idle_timeout = 10s`. I've also set `pm.max_children = 10` so that the server has some memory spare when all 10 php-fpm processes are active (typically I keep 80-100MB free under this setting).

Initially the response is great, the NC Web UI is quick and responsive, gallery renders quickly. When watching top, I see php-fpm processes spawning then being killed when idle for a brief period. However, after a few hours (of inactivity, no browsing or uploading on my part), I see this:

[image]

We can see the max amount of php-fpm processes are running, and using about 30% of the server memory. Note that they are *all* inactive (pressing `i` when running top makes them all disappear). When I try to access the NC and NCP config sites both are unresponsive and time out.

Clearly, something is keeping php-fpm processes alive and not allowing them to be killed when idle. I suspect there must be some bug in NC, perhaps unclosed connections to unmanaged resources? I'm not a php dev so I couldn't say, but I don't see why these processes would not be terminating if idle, unless NC were keeping them open and it's not likely to do that deliberately, especially when there are no requests coming in!

It is consistent with experience when running `pm = dynamic`, the common problem is php-fpm processes not being released, regardless of whether the processes are being actively managed by php.

@nachoparker do you think you could try running with `pm = ondemand` and the settings above? Watch the php-fpm processes (press `o` in `top` and enter `COMMAND=php` to filter to those processes) over time and see if there are being terminated or if they get stuck like mine. I'm guessing you won't see this problem as if you were you would also be seeing the Web UI time out and become unresponsive. It's confusing though as I'm not doing anything else on the server besides running NCP...

nachoparker, 2018-03-27:
Do you need to do anything to get to the situation where your system stops responding? (like browsing pics)

In my system (doesn't get stuck or unresponsive) I have my four configured PHP processes that are inactive most of the time. I think that situation is normal to the MPM mechanics of PHP.

```
1090 www-data  20   0  235696  39184  30036 S   0.0  4.1  38:09.56 php-fpm7.0                                                                                                                        
 1087 www-data  20   0  235008  34816  26492 S   0.0  3.7  38:31.85 php-fpm7.0                                                                                                                        
 1088 www-data  20   0  234908  34608  26168 S   0.0  3.6  38:38.56 php-fpm7.0                                                                                                                        
 1089 www-data  20   0  234864  33200  24876 S   0.0  3.5  37:59.49 php-fpm7.0
```

I think it is ok: PHP spawns X threads that are ready and waiting for action and they process all needed PHP code requests. It is consistent with the way MPM works.

In my experience with PHP the system gets stuck when you have all threads busy with some execution that takes a long time.

For example, if you don't close properly your requests you might get the whole server stuck after hitting the refresh key four times because all four processes are busy doing nothing and there's no other processes left to handle the new requests.

In order to prevent this for being a serious issue, we have the `max_execution_time` configuration parameter, so these bad behaving scripts can be killed and the resources freed.

So I think we might get a better insight by directly asking the devs, because we are a bit lost here.

Again, in order to reproduce... what do you need to do? because this doesn't happen to me, or anybody else up to this point, at least nobody reported it.

Thanks for the feedback

nachoparker, 2018-03-28:
I am testing with my picture collection on a clean instance and yes, the preview generation is slow the first time, but everything seems to work and there's no timeouts.

I think that we first need some steps to reproduce.

Also I am thinking of two possibilities:

- include the preview generator with NCP
- disable previews

In any case, also if we find good PHP values (like you did) we could also change the defaults

nachoparker, 2018-03-30:
You should probably test first without the pregeneration of thumbnails activated in cron.

I would start as simple as possible and test adding only one thing at a time.

nachoparker, 2018-04-10:
Let's move this conversation to #470

closed 2018-04-10 by nachoparker

nero120, 2018-05-08:
Returning to this after a while as I've been distracted by my own project, just a final note to say that since I removed the preview generator app I have not suffered with any timeouts, so it seems the cause of the problems is clear. Also, running these php fpm config settings seems to be working very well:

```
pm = ondemand
pm.max_children = 10
pm.process_idle_timeout = 10s;
pm.max_requests = 500
```

nachoparker, 2018-05-09:
thanks for your feedback, I will do my own testing with those parameters too

ryangorley, 2018-06-08:
@nachoparker Just for the record, in my own case this is not a preview generator specific issue. I am receiving the same "server reached MaxRequestWorkers setting, consider raising the MaxRequestWorkers setting" errors and crashes just browsing to a directory in the web UI that contains 30 or so large image files. Not using the Gallery plugin either.

nero120, 2018-06-08:
@ryangorley out of interest have you tried running using the php fpm config settings I mentioned above?

ryangorley, 2018-06-09:
@nero120 Not yet. I just detached the S3 external storage. I'm going to keep an eye on whether this happens with locally stored files. If so, I'll test the php-fpm stuff. Thank you for the response!

runasas2017, 2018-07-20:
I've tested:
 pm = ondemand
pm.max_children = 10
pm.process_idle_timeout = 10s;
pm.max_requests = 500
Better!
But in 2 cpu 3clients( 2 web+1mobile -just text chat) load up to 30%
Need to path something in NC.

feutl, 2018-09-04:
I am having the same issues described here.
BUT on a Debian 9.5 64Bit having NextcloudPi installed using the script provided on the website. Looked around for issues, first thought its the DB for whatever reason. Afterwards I figured out it is the FPM process issue.

I have 2 cores celeron something and 4GB of RAM, CPU is not the real issue and RAM either based on `top`

So something else is bugging here, I had an ubuntu 16.04 with everything setup manually first which didn't run into that issues also using fpm, but I needed a new installation and thought I go with nextcloudpi on debian, also considering switching to docker once it is working ;)

I am still having issues even I am using these settings for fpm config
`
pm.max_children = 40
pm.start_servers = 15
pm.min_spare_servers = 15
pm.max_spare_servers = 25
pm.max_requests = 500
`

rest is default, perhaps somebody can help us out because I think there could be something else bugging.

feutl, 2018-09-04:
I tested my settings today from inside the network where the server is located as well.
30GB 80.000 files and surfing the nextcloud website and it rocked :)
RAM was about 2 GB max CPU usage was over 200% so around 3,8 for a 2 core machine ;) but not timeouts. I ran into 2 file lock issues while uploading, which I assume was based on the CPU usage, redis could not deal with.

Anyhow, I would like to see the option of adjusting the FPM setting using the webinterface of ncadmin, like memory and stuff.
Perhaps with some kind of automatizm, like the average memory usage for the FPM processes (which was around 50 to 60 MB on my site) and the overall RAM minus 1GB for the rest to propose some FPM settings "automagically".

I know for Raspis thats not an option, but the Nextcloudpi insallation could be the easy to use alternative to the official VM, also for docker an any other machine out there besides raspies.
I mean for hitting the masses the docker compose file with 7 container is just a joke, honestly, but a single docker container or the script for a single (old) PC on debian is just perfect for everyone.

THANKS FOR THE SCRIPT 👍

feutl, 2018-09-10:
Perhaps nextcloudpi can integrate some automatic calculation for FPM processes

I used to calculate average fpm usage using

> ps -C php-fpm7.0 -o rss= | awk '{ sum+=$1 } END { printf ("%d%s\n", sum/NR/1024,"M") }'

If slightly adjusted you get the total memory usage instead of the avarage one

> ps -C php-fpm7.0 -o rss= | awk '{ sum+=$1 } END { printf ("%d%s\n", sum/1024,"M") }'

Anyhow, it would be great to have this information somehow monitored for a longer period of time to better calculate the "maximum" FPM process number. This usage (average/total) in combination with the "free/usable" RAM could lead to some automatic script for Nextcloudpi in the future.

nachoparker, 2018-09-10:
Right now the maximum available memory for PHP is automatically set at 75% of the total memory upon boot, and the number of threads is equal to the number of cores.

If you set that manually it will get overwritten on next reboot. The NCP way is to set this through `nc-limits`

feutl, 2018-09-10:
I see - I did my adjustments with nc-limits - thanks for the hint ;)

But the 2 process limit is not scaling - I tested it by opening the gallery. I immediatly hit the wall and ran into a timeout on the machine and got a service unavailable.
I get the idea theoretical idea of 2 processes per cores - but for my setup (and others it seems as well) this doesn't work as intended. As I said, with the static values around 50 this system is stable, fast and rock solid ;)

Two values I am missing in nc-limits
pm = dynamic / static
pm.max_requests = 500 (in my case)

Also a "hint" to the autoconfig values would be great - like 75% of the memory and the rest for the settings.
I thing, this nextcloudpi setup / script could be the easiest way to grab an old hardware, install debian and run nextcloud within 30 minutes - solid and without issues.
But these settings could hinder people at sticking with it, I assume.

nachoparker, 2018-09-11:
yeah the gallery is demanding on resources, this is an old topic. If I have time one day I will test your settings. In the mean time it would be great if others could test this values, specially on the pi.

so which one are you using? static or dynamic?

feutl, 2018-09-11:
I started with dynamic to see how this works out and RAM usage and all the stuff works and ended up with using static.
But only because this machine I am using is dedicated to nextcloud an nothing else is running on it.

nero120, 2018-09-12:
> Perhaps nextcloudpi can integrate some automatic calculation for FPM processes

To be fair, I'd much prefer that nextcloudpi was tuned specifically for the RP platform given it's hardware limitations. Running nextcloud on a regular server with decent specs gives you far more rope, and I never saw freezing when I used to run on a normal machine. Gallery has always been a dog and this is far more a software issue than a hardware one in my opinion.

feutl, 2018-09-12:
@nero120 
I think as well that nextcloudpi should be "adjusted" basically for the PI, but because the script is available for other plattforms and installations as well, at a certain point it should give the option to change some very important stuff directly from the admin interface or even better, give some advice based on the "hardware" available.

For me that means, allowing to change php settings for threads (already there), changing the mode and the max request time. It allows me to stick with the GUI instead of the CLI which makes sense.
The same applies to the nextcloud pi docker images, there are x86 builds, therefore there should be an option to change these settings in the webui, easily.

nachoparker, 2018-09-12:
I mean, NCP is by default taylored to the pi, but you are able to easily adjust stuff. Moreover, with automatic settings the hope is that you normally don't need to do even that ;)