nextcloud/nextcloudpi #408: access attemps using external IP

testbird, 2018-02-12:
I found many unsolicited access attemps even in the NC admin interface log complaining about "secure domain" failures. These were requests using the external IP as domain.

Instead of having apache listening and NC block these request, maybe apached could be configured right away to only answer requests using IPs (instead of dns domain names) if they are  local IPs.

Here is an example restricting request based on the source IP, which is another thing, but maybe the same ranges can be used to bypass a "apache secure domains configuration"
https://salsa.debian.org/freedombox-team/plinth/commit/21d6174ba052e22926f05b0a6806e3fa7f83c8f1

nachoparker, 2018-02-13:
hi, that is exactly [what ncp-web does](https://github.com/nextcloud/nextcloudpi/blob/master/nextcloudpi.sh#L74) but... I do want to access to NextCloudPi from outside my local IPs.

I want to access from my phone, from work, and so on.

testbird, 2018-02-13:
> I want to access from my phone, from work, and so on.

Sure. But you know and use your (dyn)DNS name to access, while portscanners and search bots just try IPs (http://1.2.3.4 etc.) and spider whatever they find. What I thought about would have a similar result as if NC would return nothing in response to requests for "non-trusted domains" instead of the info page, ideally it would be as if the port wasn't opened at all in the firewall.

Its like the solution to this question (actual config of the solution is at the duplicate question)
https://serverfault.com/questions/786809/block-direct-ip-connections-to-apache-web-server

nachoparker, 2018-02-13:
ok, I get you now... well that is interesting then, I just wonder if there will be use cases of people wanting to access through external IP address.

otherwise, I think it's worth implementing it

testbird, 2018-02-13:
Yes, the use case would be those with a static IP.

testbird, 2018-02-13:
You don't necessarily need to ask. I do make PRs if I think it's reachable for me and I'm able to.
Still struggling with my docker container.

(In looking for above, I came accross a potentially vast apache config
https://askubuntu.com/questions/922085/i-need-rules-to-drop-some-malicious-apache-connection)

nachoparker, 2018-02-13:
haha it was worth a try ;D

theCalcaholic, 2018-02-14:
It should definitely disabled if no domain is defined and probably not be enabled by default. I'm using NCP behind a static IP for instance - probably a few others as well.

nachoparker, 2018-02-14:
> It should definitely disabled if no domain is defined and probably not be enabled by default. I'm using NCP behind a static IP for instance - probably a few others as well.

Do you access through static IP, not by DDNS domain name? It is true that we would block many attack attempts by implementing this.

nachoparker, 2018-02-14:
Regardless of wether we do it or not, I am wondering if the following would work ( in `/etc/apache2/sites-enabled/nextcloud.conf` )

```
<RequireAny>
      Require host *
      Require local
      Require ip 192.168
      Require ip 172
      Require ip 10
</RequireAny>
```
We would need to test if `host *` would capture any hostname, but not direct access by IP. The other lines I know that work because are copied from `ncp.conf`

theCalcaholic, 2018-02-14:
@nachoparker I do for testing/debugging purposes mainly (port/DNS setup and such).

testbird, 2018-02-14:
I think you'll be able test the port forwarding as you see a "not allowed" message from apache.

testbird, 2018-02-14:
From what I read in the links above "host *" would have to be in the first (used as default) virtual host stanza (whichever that may be with multiple files).

nachoparker, 2018-02-15:
in a quick test, `host *` it didn't work

testbird, 2018-02-15:
maybe it a regexp is expected? .*

testbird, 2018-02-15:
No, i actually just commented out "Require host localhost", it checks the domain of the ip the request is comming from, and reverse lookup does not work for every ip, and not at all in the docker container.

The solution to this issue is, as in the linked answers above, have a first "dummy" virtualhost block everything (it gets used as fallback for non defined virtual hosts).

nachoparker, 2018-02-22:
> The solution to this issue is, as in the linked answers above, have a first "dummy" virtualhost block everything (it gets used as fallback for non defined virtual hosts).

I aggree, it would be nice if somebody could come up with this configuration, as my hands are pretty full right now

Aeon512, 2018-02-23:
Just wanted to throw in my opinion on this.
Why not using the geoip database, and overing an option in NCP to specify which countries are allowed to externally access NextCloud?

Usually, there are only a few countries where you usually want to have access from, so you would filter out a lot of unwanted accesses.

Having said that, the best approach is still to enabled the two factor authentication plugin in NextCloud and enforce this for all users.

nachoparker, 2018-02-26:
Thanks

> Usually, there are only a few countries where you usually want to have access from, so you would filter out a lot of unwanted accesses.

This would prevent us from using Tor

szaimen, 2020-11-28:
Hey, maybe something here can be useful?
https://github.com/nextcloud/vm/blob/master/network/geoblock.sh