nextcloud/nextcloudpi #409: I can't get Pico CMS app to work

nhall123, 2018-02-12:
I've installed the Pico CMS app through the Admin section, but can't get it to work. When I create a site and try to navigate to it, I get the following:
"File not found The specified document has not been found on the server."
The only setup instructions I could find are those relating to the apache configuration: https://github.com/nextcloud/cms_pico/wiki/Admin-Settings#apache-configuration
I've tried implementing each of the suggested settings in /etc/apache2/sites-available/nextcloud.conf but nothing is working.
Is there anything particular to nextcloudpi that would be getting in the way?
Has anyone seen this issue and worked out a solution?
Does anyone have a detailed setup guide for Pico CMS in nextcloudpi?

ovpc, 2018-02-12:
There quite a few reports on issues with cms_pico and NC integration, I tried once, it did not work out of the box, I did not investigate further. Did you check these?

- [Issues at NC Help](https://help.nextcloud.com/t/picocms-step-by-step-instructions-for-installation-and-configuration/22643)

- [NC-cms_pico Issues on Github](https://github.com/nextcloud/cms_pico/issues)

Doesn't sound like this has anything to do with NCP.

I use leightweight [CMSimple](https://www.cmsimple.org/en/url) for CMS (without DB) on RPI

nachoparker, 2018-02-13:
Hi, please use the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) for help and technical conversation and this for bugs.

You can link your forum post here for continuity if you want

Thanks for reporting

closed 2018-02-13 by nachoparker