nextcloud/nextcloudpi #410: NCP doesn't detect available updates / The git clone command failed: No connectivity to https://github.com

nachoparker, 2018-02-12:
These two symptoms refer to a bug that was introduced to the NCP updater the 8th of february.

People that updated between the 8th of february (`v0.64.4`) and the 12th of february (`v0.64.6`) (including autoupdates) need to manually update to get the fix

- run `sudo ncp-update`
- run `nc-update` in ncp-config
- run `nc-update` in ncp-web

eogut, 2018-02-14:
Hi Nacho,

the update broke my Raspi's internet connection completely. When I run `sudo ncp-update` I get:
`Downloading updates
fatal: unable to access 'https://github.com/nextcloud/nextcloudpi.git/': Could not resolve host: github.com
No internet connectivity`

`sudo ncp-report` provides:

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v0.46.3
NextCloudPi image    untagged
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          sda
datadir              /media/Festplatte/Nextcloud/Data
data in SD           no
data filesystem      ext2/ext3
data disk usage      186G/874G
rootfs usage         2.8G/15G
swapfile             /media/Festplatte/swap
Nextcloud check      ok
Nextcloud version    12.0.2.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       no }
port check 80        open
port check 443       open
IP                   192.168.178.39
gateway              192.168.178.1
interface            enxb827eb09ee11
certificates         none
certs due            none
NAT loopback         no
uptime               1day

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.178.39",
            "4": "kushwraboowymgkl.myfritz.net"
        },
        "datadirectory": "\/media\/Festplatte\/Nextcloud\/Data",
        "overwrite.cli.url": "https:\/\/kushwraboowymgkl.myfritz.net",
        "dbtype": "mysql",
        "version": "12.0.2.0",
        "dbname": "nextcloud",
        "dbhost": "localhost",
        "dbport": "",
        "dbtableprefix": "oc_",
[... 48 lines cut ...]
```

{"reqId":"xxkJBr3zecOfDLeParT3","level":3,"time":"2018-02-14T07:30:07+00:00","remoteAddr":"","user":"--","app":"appstoreFetcher","method":"--","url":"--","message":"Exception: {\"Exception\":\"GuzzleHttp\\\\Exception\\\\ConnectException\",\"Message\":\"cURL error 6: Could not resolve host: apps.nextcloud.com\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(103): GuzzleHttp\\\\Exception\\\\RequestException::wrapException(Object(GuzzleHttp\\\\Message\\\\Request), Object(GuzzleHttp\\\\Ring\\\\Exception\\\\ConnectException))\\n#1 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(132): GuzzleHttp\\\\RequestFsm->__invoke(Object(GuzzleHttp\\\\Transaction))\\n#2 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/react\\\/promise\\\/src\\\/FulfilledPromise.php(25): GuzzleHttp\\\\RequestFsm->GuzzleHttp\\\\{closure}(Array)\\n#3 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/ringphp\\\/src\\\/Future\\\/CompletedFutureValue.php(55): React\\\\Promise\\\\FulfilledPromise->then(Object(Closure), NULL, NULL)\\n#4 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Message\\\/FutureResponse.php(43): GuzzleHttp\\\\Ring\\\\Future\\\\CompletedFutureValue->then(Object(Closure), NULL, NULL)\\n#5 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(134): GuzzleHttp\\\\Message\\\\FutureResponse::proxy(Object(GuzzleHttp\\\\Ring\\\\Future\\\\CompletedFutureArray), Object(Closure))\\n#6 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Client.php(165): GuzzleHttp\\\\RequestFsm->__invoke(Object(GuzzleHttp\\\\Transaction))\\n#7 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Client.php(125): GuzzleHttp\\\\Client->send(Object(GuzzleHttp\\\\Message\\\\Request))\\n#8 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Http\\\/Client\\\/Client.php(138): GuzzleHttp\\\\Client->get('https:\\\/\\\/apps.ne...', Array)\\n#9 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/App\\\/AppStore\\\/Fetcher\\\/Fetcher.php(99): OC\\\\Http\\\\Client\\\\Client->get('https:\\\/\\\/apps.ne...', Array)\\n#10 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/App\\\/AppStore\\\/Fetcher\\\/AppFetcher.php(66): OC\\\\App\\\\AppStore\\\\Fetcher\\\\Fetcher->fetch('\\\"2018-02-10 02:...', '[{\\\"id\\\":\\\"spreedm...')\\n#11 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/App\\\/AppStore\\\/Fetcher\\\/Fetcher.php(162): OC\\\\App\\\\AppStore\\\\Fetcher\\\\AppFetcher->fetch('\\\"2018-02-10 02:...', '[{\\\"id\\\":\\\"spreedm...')\\n#12 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Installer.php(394): OC\\\\App\\\\AppStore\\\\Fetcher\\\\Fetcher->get()\\n#13 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/updatenotification\\\/lib\\\/Notification\\\/BackgroundJob.php(258): OC\\\\Installer::isUpdateAvailable('systemtags', Object(OC\\\\App\\\\AppStore\\\\Fetcher\\\\AppFetcher))\\n#14 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/updatenotification\\\/lib\\\/Notification\\\/BackgroundJob.php(155): OCA\\\\UpdateNotification\\\\Notification\\\\BackgroundJob->isUpdateAvailable('systemtags')\\n#15 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/updatenotification\\\/lib\\\/Notification\\\/BackgroundJob.php(79): OCA\\\\UpdateNotification\\\\Notification\\\\BackgroundJob->checkAppUpdates()\\n#16 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/BackgroundJob\\\/Job.php(59): OCA\\\\UpdateNotification\\\\Notification\\\\BackgroundJob->run(NULL)\\n#17 \\ [... cut ...] : 'admin' (Remote IP: '31.18.113.217')","userAgent":"Mozilla\/5.0 (Android) Nextcloud-android\/3.0.0","version":"12.0.2.0"}
{"reqId":"WoP9SX8AAQEAAC79q8sAAABH","level":2,"time":"2018-02-14T09:12:08+00:00","remoteAddr":"31.18.113.217","user":"--","app":"core","method":"HEAD","url":"\/remote.php\/webdav\/SofortUpload\/WhatsApp%20Video\/","message":"Login failed: 'admin' (Remote IP: '31.18.113.217')","userAgent":"Mozilla\/5.0 (Android) Nextcloud-android\/3.0.0","version":"12.0.2.0"}

```
</details>

nachoparker, 2018-02-14:
@eogut your internet connection is fine. Type

```
sudo ncp-update
```
to get the fix

eogut, 2018-02-14:
Hi nacho,

I expect NCP to have caused the issue as the last automatic update to v0.46.3 worked and I haven't touched my Raspi ever since.

Anyways, pinging works, I suppose:

<details>
<summary>8.8.8.8 ping statistics</summary>

```
36 packets transmitted, 36 received, 0% packet loss, time 35053ms
rtt min/avg/max/mdev = 20.665/21.379/22.625/0.503 ms
```

</details>

I'm puzzled, what's going on? :( Thanks for your help in advance.

nachoparker, 2018-02-14:
Ok, you have internet, let's see your DNS. Your error log above says

```
Could not resolve host: github.com
```

So try pinging github.com, and also google.com

```
ping github.com
ping google.com
```

eogut, 2018-02-15:
Result: Temporary failure in name resolution

nachoparker, 2018-02-15:
your DNS is not set correctly. Review your router configuration/ISP. Seems like this has nothing to do with NCP

closed 2018-02-26 by nachoparker