nextcloud/nextcloudpi #411: [UPGRADE] Step 5 is currently in process. Please reload this page later.

piexad, 2018-02-12:
Hello,
I'm trying to upgrade from nextcloudpi 12.0.4 to 13 but when I click on "Open upgrade system" (I'm french so I don't know exactly how it is written "Ouvrir le système de mise à jour") I get this message : 
`Step 5 is currently in process. Please reload this page later.`

What to think about that ? 
My ncp-report says that : 
... 
`Cannot load Zend OPcache - it was already loaded`

piexad, 2018-02-13:
My ncp-report : 
<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v0.46.7
NextCloudPi image    NextCloudPi_12-04-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive1/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      3.4G/932G
rootfs usage         3.5G/30G
swapfile             /var/swap
Nextcloud check      ok
Nextcloud version    12.0.4.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.xx
gateway              192.168.1.1
interface            eth0
certificates         none
certs due            none
NAT loopback         no
uptime               1min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "192.168.1.xx",
            "6": "xxxxxxx.fr"
        },
        "datadirectory": "\/media\/USBdrive1\/ncdata",
        "overwrite.cli.url": "https:\/\/192.168.1.xx",
        "dbtype": "mysql",
        "version": "12.0.4.3",
        "dbname": "nextcloud",
        "dbhost": "localhost",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "ocxxxxxxxxxx",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
[... 63 lines cut ...]
<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"WoJ4hX8AAQEAAHhzeWYAAEIB","level":2,"time":"2018-02-13T05:32:53+00:00"                                                                                        ,"remoteAddr":"192.168.1.1","user":"--","app":"no app in context","method":"PROP                                                                                        FIND","url":"\/remote.php\/dav\/calendars\/famille\/personal\/","message":"Tempo                                                                                        rary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userA                                                                                        gent":"DAVdroid\/1.9.10-ose (1\/20\/18; dav4android; okhttp3) Android\/6.0","ver                                                                                        sion":"12.0.4.3"}
{"reqId":"WoJ51H8AAQEAAHhykaMAAAEB","level":2,"time":"2018-02-13T05:38:28+00:00"                                                                                        ,"remoteAddr":"192.168.1.47","user":"--","app":"no app in context","method":"GET                                                                                        ","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"                                                                                        Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","                                                                                        userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:58.0) Gecko\/20100101                                                                                         Firefox\/58.0","version":"12.0.4.3"}
{"reqId":"WoJ51H8AAQEAAHhykaQAAAEC","level":2,"time":"2018-02-13T05:38:29+00:00"                                                                                        ,"remoteAddr":"192.168.1.47","user":"pierre","app":"no app in context","method":                                                                                        "POST","url":"\/index.php\/heartbeat","message":"Temporary directory \/media\/US                                                                                        Bdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windo                                                                                        ws NT 10.0; Win64; x64; rv:58.0) Gecko\/20100101 Firefox\/58.0","version":"12.0.                                                                                        4.3"}
{"reqId":"WoJ51n8AAQEAAHhykaUAAAAC","level":2,"time":"2018-02-13T05:38:30+00:00"                                                                                        ,"remoteAddr":"192.168.1.1","user":"--","app":"no app in context","method":"GET"                                                                                        ,"url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/                                                                                        tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.                                                                                        3 (build 1) (Nextcloud)","version":"12.0.4.3"}
[... cut ...]                                                                                       PROPFIND","url":"\/remote.php\/dav\/files\/xxxxxxx\/","message":"Temporary direct                                                                                        ory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozi                                                                                        lla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"12.0.4.3"}

nachoparker, 2018-02-13:
maybe it takes a while, I was able to upgrade from 12.0.4 to 13. I suggest you try the official forums if you cannot do it for any reason.

https://help.nextcloud.com/c/support/appliances-docker-snappy-vm

ovpc, 2018-02-13:
One of my 'testpi' upgrades could not complete at once, others went fine, it did not  turn off maintenance mode.
Also completed fine though, after doing:
```
sudo su -
cd /var/www/nextcloud
sudo -u www-data php occ maintenance:mode --off
exit
```

piexad, 2018-02-13:
@nachoparker thank you, I'll try the official forum if we can't find a solution here but I thought it would be better to start here

@ovpc Well it's not in maintenance mode and when I try `sudo -u www-data php occ maintenance:mode --off` it says the same thing as previously: `Cannot load Zend OPcache - it was already loaded`
Weird!

closed 2018-02-13 by nachoparker

piexad, 2018-02-13:
Why is my issue closed ?

nachoparker, 2018-02-13:
because you are going to search for help somewhere else, and because this is not an NCP issue. 

even though the issue is closed, we can still talk here without problems, but I know that upgrades to NC13 work in NCP in general and I don't think there is nothing to fix from our side so I will just close it so the issues are more manageable for us and this issue won't distract our attention.

please, let us know how it goes, if we can help we will

denics, 2018-07-31:
Anyway @piexad just delete the file `.step` located here:
`/nextcloud-data/updater-somerandomstring/.step`
it solved the issue for me.

fhiernaux, 2020-10-28:
thanks @denics for this info about the file .step,  I've also update the memory_limit to -1 and now it works

RebelCoderRU, 2020-12-19:
I am having this issue too and none of the solutions I found, worked. @denics (and a few others on NC forums) suggested deleting `.step` file. My NC folder does not even have that file, and I am also stuck on `Step 5 is currently in process. Please reload this page later.`

To be very honest with you, this is the worst experience with Open-Source software I have ever had. I asked on forums and got totally ignored by the team. 50% of the people could not solve it with proposed solutions and are still having this issue. This seems to be a bug but this issues is closed.

Example of this not being solved at all:

- https://help.nextcloud.com/t/stuck-on-update-process-step-5-is-currently-in-process-please-reload-this-page-later/58371/12
- https://help.nextcloud.com/t/step-5-is-currently-in-process-please-reload-this-page-later-with-12-0-5-updater/30716
- https://help.nextcloud.com/t/update-from-20-0-2-to-20-0-4-fails-step-5-is-currently-in-process-please-reload-this-page-later/101553/3

How can this be used in a team environment if something breaks, and you get almost 0 support?

PrzemekSkw, 2021-01-15:
Hello, I have the same problem on Dietpi (RaspberryPi4). I don't have any .step file to delete and I have memory -1 set. What else I can do? I try upgrade from 19.0.5 to 19.0.7.
Regards,
Przemek

jonathanjsimon, 2022-02-12:
I found this article while trying to solve the issue. The updater-foo folders are created in the data folder *where your sync data actually is* not the data folder in the nextcloud installation folder.

https://alaskalinuxuser3.ddns.net/2021/07/29/nextcloud-step-5-is-currently-in-process/

JVKeller, 2023-03-13:
> I found this article while trying to solve the issue. The updater-foo folders are created in the data folder _where your sync data actually is_ not the data folder in the nextcloud installation folder.
> 
> https://alaskalinuxuser3.ddns.net/2021/07/29/nextcloud-step-5-is-currently-in-process/

Thanks for pointing this out!
`sudo su
cd /var/www/nextcloud****/data/`
there you will find the updater-**** to delete