nextcloud/nextcloudpi #412: Install torrent client and seed ncp's and other torrents

ovpc, 2018-02-14:
Installing a torent client on rpi is quite simple following this [how to](http://kamilslab.com/2017/04/21/how-to-build-a-raspberry-pi-torrentbox/)

It would be a nice addition if this could be installed, configured and activated from ncp

I have mine set up to seed latest from ownyourbits ;-)

Paste the output of `sudo ncp-report`
pi@nextcloudpi:~ $ sudo ncp-report
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded

theCalcaholic, 2018-02-14:
This would be very nice, if it
1) was disabled by default (opt-in)
2) could be run headless and be controlled via web interface or ncp-config

sunjam, 2018-02-21:
[OcDownloader](https://apps.nextcloud.com/apps/ocdownloader) is available in the Nextcloud appstore and can use aria2c for torrents, which is also a popular  linux terminal download manager (think wget) that can handle multiple files + formats.  Could definitely be used to seed ncp torrents.  Setup requires installing and setting up aria2c, youtube-dl and optionally curl, which could all be automated through nextcloudpi-config.  Here is the [firefox extension](https://addons.mozilla.org/en-US/firefox/addon/ocdownloader-extension/?src=cb-dl-created) and [chrome extension](https://chrome.google.com/webstore/detail/ocdownloader/dgemhnddkdkailclfekamalbkaccckop?hl=en-US) so you can send it your download links.

nachoparker, 2018-04-10:
I am afraid this won't get done any time soon. It is a bit out of scope, and we really don't have the resources to accomplish it.

Thankfully, the torrents have more than enough seeds. We now have the steps to do it if anyone wants to do it manually like @ovpc did.

We can reopen if we change our minds.

Thanks as always

closed 2018-04-10 by nachoparker