nextcloud/nextcloudpi #413: NextCloudPi Panel password

wildthing1, 2018-02-14:
In the NextCloudPi panel, tab "nc-passwd" it's not possible to set passwords that contain "!" (exclamation mark). Not a big issue though...

theCalcaholic, 2018-02-14:
This is currently not easily resolvable, due to the design of the ncp-web backend, but might get a fix if/when we start refactoring ncp-web.

theCalcaholic, 2018-02-14:
For documentation purposes: The problem is the regex testing of config values in ncp-launcher.php

nachoparker, 2018-02-14:
We need to carefully sanitize any input to prevent code injection. `!` means history expansion in bash and is really dangerous.

Whenever we refactor this this will be fixed, but it is not going to happen in the near future.

Thanks for reporting

closed 2018-02-14 by nachoparker