nextcloud/nextcloudpi #416: PhP error ?

Iolaum, 2018-02-15:
### PROBLEMS

I am not sure if this is a problem with ncp, php/raspbian or not a problem at all. Simply running `php -v` gives an error:

```
$ php -v
Cannot load Zend OPcache - it was already loaded
PHP 7.0.27-0+deb9u1 (cli) (built: Jan  5 2018 13:51:52) ( NTS )
Copyright (c) 1997-2017 The PHP Group
Zend Engine v3.0.0, Copyright (c) 1998-2017 Zend Technologies
    with Zend OPcache v7.0.27-0+deb9u1, Copyright (c) 1999-2017, by Zend Technologies
```

Looking into some other log files as well, the error gets thrown at sys log every 30 min:

```
root@nextcloudpi:/var/log# cat syslog | grep OPcache
Feb 15 06:39:09 nextcloudpi sessionclean[10872]: Cannot load Zend OPcache - it was already loaded
Feb 15 06:39:09 nextcloudpi sessionclean[10872]: Cannot load Zend OPcache - it was already loaded
Feb 15 07:09:09 nextcloudpi sessionclean[11044]: Cannot load Zend OPcache - it was already loaded
Feb 15 07:09:09 nextcloudpi sessionclean[11044]: Cannot load Zend OPcache - it was already loaded
Feb 15 07:39:09 nextcloudpi sessionclean[11204]: Cannot load Zend OPcache - it was already loaded
Feb 15 07:39:10 nextcloudpi sessionclean[11204]: Cannot load Zend OPcache - it was already loaded
Feb 15 08:09:09 nextcloudpi sessionclean[11315]: Cannot load Zend OPcache - it was already loaded
...
```

**Paste the output of `sudo ncp-report`:**

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v0.46.10
NextCloudPi image    NextCloudPi_12-04-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      182G/917G
rootfs usage         1.7G/3.6G
swapfile             /media/wd1/swap
Nextcloud check      ok
Nextcloud version    12.0.4.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.20
gateway              192.168.1.1
interface            eth0
certificates         $HOSTNAME
certs due            25 days
NAT loopback         no
uptime               2days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

[... 73 lines cut ...]
```

{"reqId":"WoCkVX8AAQEAAAG4u6UAAA4B","level":2,"time":"2018-02-11T20:15:18+00:00","remoteAddr":"137.226.113.20","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"137.226.113.20\" tried to access using \"79.75.141.71\" as host.","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:50.0) Gecko\/20100101 Firefox\/50.0","version":"12.0.4.3"}
{"reqId":"WoCkVn8AAQEAAAG4u60AAA4J","level":2,"time":"2018-02-11T20:15:19+00:00","remoteAddr":"137.226.113.20","user":"--","app":"core","method":"GET","url":"\/index.php\/js\/core\/merged-template-prepend.js?v=2793255a2c31f4b4c2eab906af15ce25-1","message":"Trusted domain error. \"137.226.113.20\" tried to access using \"79.75.141.71\" as host.","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:50.0) Gecko\/20100101 Firefox\/50.0","version":"12.0.4.3"}
{"reqId":"WoDQD38AAQEAAAHFrL4AAABS","level":2,"time":"2018-02-11T23:21:51+00:00","remoteAddr":"107.170.227.152","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"107.170.227.152\" tried to access using \"79.75.141.71\" as host.","userAgent":"Mozilla\/5.0 zgrab\/0.x","version":"12.0.4.3"}
{"reqId":"WoF0O38AAQEAAB-8900AAAAA","level":3,"time":"2018-02-12T11:02:19+00:00","remoteAddr":"210.16.189.29","user":"--","app":"PHP","method":"GET","url":"\/admin\/config.php","message":"Exception: The requested uri(\/admin\/config.php) cannot be processed by the script '\/core\/templates\/404.php') at \/var\/www\/nextcloud\/lib\/private\/AppFramework\/Http\/Request.php#729","userAgent":"curl\/7.19.7 (i386-redhat-linux-gnu) libcurl\/7.19.7 NSS\/3.27.1 zlib\/1.2.3 libidn\/1.18 libssh2\/1.4.2","version":"12.0.4.3"}
{"reqId":"WoHJ6H8AAQEAAB-893IAAAAO","level":2,"time":"2018-02-12T17:07:53+00:00","remoteAddr":"164.52.6.150","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"164.52.6.150\" tried to access using \"79.75.151.21:443\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; rv:52.0) Gecko\/20100101 Firefox\/52.0","version":"12.0.4.3"}
{"reqId":"WoHjsn8AAQEAAB-8948AAAAS","level":2,"time":"2018-02-12T18:57:55+00:00","remoteAddr":"209.126.136.7","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"209.126.136.7\" tried to access using \"79.75.151.21\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/57.0.2987.133 Safari\/537.36","version":"12.0.4.3"}
{"reqId":"WoH69X8AAQEAACAK1dIAAABE","level":2,"time":"2018-02-12T20:37:09+00:00","remoteAddr":"139.162.78.135","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"139.162.78.135\" tried to access using \"79.75.151.21\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; WOW64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/45.0.2454.101 Safari\/537.36","version":"12.0.4.3"}
{"reqId":"WoJC9H8AAQEAACAK9L8AAABL","level":2,"time":"2018-02-13T01:44:20+00:00","remoteAddr":"34.214.38.179","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"34.214.38.179\" tried to access using \"92.4.153.185:443\" as host.","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10_11_5) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/50.0.2661.102 Safari\/537.36","version":"12.0.4.3"}
{"reqId":"WoLhKn8AAQEAAD5A4SIAAABL","level":2,"time":"2018-02-13T12:59:22+00:00","remoteAddr":"139.162.78.135","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"139.162.78.135\" tried to access using \"92.4.153.185\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; WOW64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/45.0.2454.101 Safari\/537.36","version":"12.0.4.3"}
{"reqId":"WoLrIn8AAQEAAD4wuFkAAAAH","level":2,"time":"2018-02-13T13:41:54+00:00","remoteAddr":"184.105.139.69","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"184.105.139.69\" tried to access using \"92.4.153.185\" as host.","userAgent":"--","version":"12.0.4.3"}
{"reqId":"WoMy238AAQEAAD4wxZkAAAAS","level":2,"time":"2018-02-13T18:47:55+00:00","remoteAddr":"107.170.227.37","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"107.170.227.37\" tried to access using \"79.75.128.116\" as host.","userAgent":"Mozilla\/5.0 zgrab\/0.x","version":"12.0.4.3"}
{"reqId":"WoOZQX8AAQEAAANxI7kAAAAT","level":2,"time":"2018-02-14T02:04:49+00:00","remoteAddr":"89.248.160.199","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"89.248.160.199\" tried to access using \"79.75.128.116\" as host.","userAgent":"libwww-perl\/6.31","version":"12.0.4.3"}
{"reqId":"WoQxAn8AAQEAAA2rWeEAAABQ","level":2,"time":"2018-02-14T12:52:19+00:00","remoteAddr":"184.105.139.69","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"184.105.139.69\" tried to access using \"79.75.128.116\" as host.","userAgent":"--","version":"12.0.4.3"}
{"reqId":"WoR7bH8AAQEAAA2rWg8AAABV","level":2,"time":"2018-02-14T18:09:48+00:00","remoteAddr":"60.191.38.77","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"60.191.38.77\" tried to access using \"79.75.128.116:443\" as host.","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10.11; rv:47.0) Gecko\/20100101 Firefox\/47.0","version":"12.0.4.3"}
{"reqId":"WoSXW38AAQEAAA2rWicAAABE","level":3,"time":"2018-02-14T20:08:59+00:00","remoteAddr":"163.172.192.121","user":"--","app":"PHP","method":"GET","url":"\/a2billing\/common\/javascript\/misc.js","message":"Exception: The requested uri(\/a2billing\/common\/javascript\/misc.js) cannot be processed by the script '\/core\/templates\/404.php') at \/var\/www\/nextcloud\/lib\/private\/AppFramework\/Http\/Request.php#729","userAgent":"python-requests\/2.18.4","version":"12.0.4.3"}
{"reqId":"WoT5@38AAQEAAA2rWzoAAABJ","level":2,"time":"2018-02-15T03:09:47+00:00","remoteAddr":"139.162.78.135","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"139.162.78.135\" tried to access using \"79.75.128.116\" as host.","userAgent":"M [... cut ...] 18.206.66\" tried to access using \"79.75.156.41\" as host.","userAgent":"--","version":"12.0.4.3"}
{"reqId":"WoX18X8AAQEAACmpzJsAAAAH","level":2,"time":"2018-02-15T21:04:50+00:00","remoteAddr":"60.191.38.77","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"60.191.38.77\" tried to access using \"79.75.156.41:443\" as host.","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10.11; rv:47.0) Gecko\/20100101 Firefox\/47.0","version":"12.0.4.3"}

```
</details>

nachoparker, 2018-02-16:
That log is not the mysql log, but the apache log. See this

https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-nextcloud.sh#L190

You are right, my mysql log is also empty... let's see if we can find it. Are you using ramlogs? does it come back after doing `systemctl restart mysqld`?

nachoparker, 2018-02-16:
I think I found it. Some update duplicated the symlink for the opcache configuration

```
# ls -l /etc/php/7.0/fpm/conf.d/ | grep opc
lrwxrwxrwx 1 root root 39 Dec  4 13:29 10-opcache.ini -> /etc/php/7.0/mods-available/opcache.ini
lrwxrwxrwx 1 root root 39 Feb  7 10:47 20-opcache.ini -> /etc/php/7.0/mods-available/opcache.ini
```

Can you replicate this finding?

It is just a harmless warning.

Iolaum, 2018-02-16:
Yes I can replicate that:

```
$ pwd
/etc/php/7.0/fpm/conf.d
$ ll | grep opcache
lrwxrwxrwx 1 root root   39 Dec  4 13:29 10-opcache.ini -> /etc/php/7.0/mods-available/opcache.ini
lrwxrwxrwx 1 root root   39 Jan 14 22:49 20-opcache.ini -> /etc/php/7.0/mods-available/opcache.ini
```
Is the solution to just delete the second symlink?

nachoparker, 2018-02-16:
yes, I fixed it in the update, please reopen if it still happens after `v0.46.16`

closed 2018-02-16 by nachoparker

Iolaum, 2018-10-27:
It looks like this bug reappeared as we moved to php7.2:

```
$ echo "$( ls -l /etc/php/7.2/fpm/conf.d/*-opcache.ini |  wc -l )"
2
$ echo "$( ls -l /etc/php/7.2/cli/conf.d/*-opcache.ini |  wc -l )"
2

```

ovpc, 2018-10-27:
Strange, it only happens on my rpi3B+ , not on my Debian9 install

Iolaum, 2018-10-27:
I think this bug creeped in after nacho enabled the buster sources, not php7.2 but I am not 100% sure

ovpc, 2018-11-17:
2 simlinks in /etc/php/7.2/cli/conf.d/  (10- and 20-) both pointing to    /etc/php/7.2/mods-available/opcache.ini

eogut, 2018-11-23:
> Back again ;-( 2 simlinks in /etc/php/7.2/cli/conf.d/ (10- and 20-) both pointing to /etc/php/7.2/mods-available/opcache.ini

I have that error, too!

nachoparker, 2018-11-23:
should be fixed now again. Note that this is a harmless warning, that appears with updates. Not much we can do about it.

saidogoid, 2018-12-21:
Hi,

In amazon Ami2 solved to rename opcache.ini to 10-opcache.ini.
Go to /etc/php.d/ and rename it. 
See if you have here a duplicated opcache too.

Cheers

amilopowers, 2019-04-22:
This still happens in v1.11.2 on ODROID XU4.

miri64, 2019-10-09:
Had the same problem in v1.17.1 (after an update on Sep 30th) and removing one of the symlink solved it

Iolaum, 2019-10-09:
@miri64 which symlink? I noticed in my ncp gives that warning again.

miri64, 2019-10-09:
See https://github.com/nextcloud/nextcloudpi/issues/416#issuecomment-366183822. I just deleted the older of the two to silence the warnings.

Iolaum, 2019-10-09:
Hm, so it's the same issue again. Thought that the upgrade to buster would have solved that.

ovpc, 2019-11-15:
Returned again
Fix is
> sudo rm /etc/php/7.3/cli/conf.d/20-opcache.ini
> sudo rm /etc/php/7.3/fpm/conf.d/20-opcache.ini

ferderm, 2021-02-19:
Hi there,

with the updates to `php7.3-opcache armhf 7.3.27-1~deb10u1` / `php7.3 all 7.3.27-1~deb10u1` the issue re-appaered again. 

> sudo rm /etc/php/7.3/cli/conf.d/20-opcache.ini
> sudo rm /etc/php/7.3/fpm/conf.d/20-opcache.ini
 
did it.

ovpc, 2021-02-19:
yep, same here.

Iolaum, 2021-02-20:
It appears a PR similar to [this one](https://github.com/nextcloud/nextcloudpi/pull/1031/files) is needed for this re-appearing in php7.3 series. If no-one is up for it I 'll try to find some time and pick it up.

FadeFx, 2022-07-17:
seems to me the problem is back, i just restored my nextcloud onto a fresh installation of ncp after apt-get upgrade (raspberrypi) and see this during most operations.
ncp version 1.48.1
php 7.4

thatgeek, 2022-10-02:
for me, this came up with a webuzo update to php. the extra.ini file at /usr/local/apps/php74/etc/php.d/extra.ini had two lines duplicated:
```
zend_extension=ioncube_loader_lin_7.4.so
zend_extension=opcache.so
```

removing the duplicated lines solved the warnings.

theCalcaholic, 2022-10-02:
@thatgeek What's a webuzo update? 🤔

deathontheshore, 2023-02-07:
Seems to have reappeared:
`lrwxrwxrwx 1 root root  39 Jan 22 02:31 10-opcache.ini -> /etc/php/8.1/mods-available/opcache.ini
lrwxrwxrwx 1 root root  39 Feb  6 11:46 20-opcache.ini -> /etc/php/8.1/mods-available/opcache.ini`

token0, 2023-07-25:
Manual fix.
Find all occurrences:
```
# for f in $(find /etc/php/ 2>/dev/null -name "*opc*") ; do ls -laFh $f;done
lrwxrwxrwx 1 root root 39 Dec 29  2022 /etc/php/7.3/cli/conf.d/20-opcache.ini -> /etc/php/7.3/mods-available/opcache.ini
lrwxrwxrwx 1 root root 39 Jul  6  2020 /etc/php/7.3/cli/conf.d/10-opcache.ini -> /etc/php/7.3/mods-available/opcache.ini
lrwxrwxrwx 1 root root 39 Dec 29  2022 /etc/php/7.3/cgi/conf.d/20-opcache.ini -> /etc/php/7.3/mods-available/opcache.ini
lrwxrwxrwx 1 root root 39 Jul  6  2020 /etc/php/7.3/cgi/conf.d/10-opcache.ini -> /etc/php/7.3/mods-available/opcache.ini
lrwxrwxrwx 1 root root 39 Dec 29  2022 /etc/php/7.3/fpm/conf.d/20-opcache.ini -> /etc/php/7.3/mods-available/opcache.ini
```
And then remove oldest duplicates. In my case those that was dated 2020.
```
rm /etc/php/7.3/cli/conf.d/10-opcache.ini
rm /etc/php/7.3/cgi/conf.d/10-opcache.ini
```

miri64, 2024-06-10:
This showed up again after updating the PHP packages to 8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e last Saturday (2024-06-08). I am on NCP v1.52.2:

```console
$ ls -l /etc/php/8.1/*/conf.d/*opcache*.ini
lrwxrwxrwx 1 root root 39 May  2 03:15 /etc/php/8.1/cli/conf.d/10-opcache.ini -> /etc/php/8.1/mods-available/opcache.ini
lrwxrwxrwx 1 root root 39 Jun  8 20:44 /etc/php/8.1/cli/conf.d/20-opcache.ini -> /etc/php/8.1/mods-available/opcache.ini
lrwxrwxrwx 1 root root 39 May  2 03:15 /etc/php/8.1/fpm/conf.d/10-opcache.ini -> /etc/php/8.1/mods-available/opcache.ini
lrwxrwxrwx 1 root root 39 Jun  8 20:44 /etc/php/8.1/fpm/conf.d/20-opcache.ini -> /etc/php/8.1/mods-available/opcache.ini
```

janvlug, 2026-10-17:
I saw the "Cannot load Zend OPcache - it was already loaded" already for a long time, but only today took action to investigate.

I have:
```
$ ls -l /etc/php/*/*/conf.d/ | grep opc
lrwxrwxrwx 1 root root 39 Sep  8  2024 10-opcache.ini -> /etc/php/8.3/mods-available/opcache.ini
lrwxrwxrwx 1 root root 39 Aug 24 13:10 20-opcache.ini -> /etc/php/8.3/mods-available/opcache.ini
lrwxrwxrwx 1 root root  39 Sep  8  2024 10-opcache.ini -> /etc/php/8.3/mods-available/opcache.ini
lrwxrwxrwx 1 root root  39 Aug 24 13:10 20-opcache.ini -> /etc/php/8.3/mods-available/opcache.ini
```
Note that the `20-opcache.ini` is the newest. I would expect it is best to remove the oldest link. But in the comments above some people remove the newest link. What is recommended?

Should this issue be reopened?

Knubbel, 2026-10-17:
>          Should this issue be reopened?

I guess so. I let DeepSeek do an analysis on my installation and here is the summary of findings and advice:

| | |
|---|---|
| **System** | Raspberry Pi 4B, NextcloudPi (NCP v1.58.0) |
| **OS** | Armbian 26.8.3, Debian 13 "trixie" |
| **PHP** | 8.3 from `deb.sury.org` (`8.3.x-1+0~<date>...+debian13~1`) |
| **Web server** | Apache 2.4.68 with `php8.3-fpm` |
| **Observed** | on every `php8.3-*` package upgrade since the NCP install (2024-09-24) |
| **Status** | root cause identified, fixed locally and in the NCP template, fix verified |

---

## TL;DR

NCP writes `/etc/php/8.3/mods-available/opcache.ini` **without the `; priority=10` header line**
that the original sury template contains. `php-helper` falls back to priority **20** when the
header is missing, so the `php8.3-opcache` `postinst` creates an additional `20-opcache.ini`
symlink next to the existing `10-opcache.ini` on the next package update. Both symlinks resolve
to the same file and both contain `zend_extension=opcache.so` → PHP loads OPcache twice and
prints the warning.

**Permanent fix:** add `; priority=10` as the first line of
`/etc/php/8.3/mods-available/opcache.ini` (and to the NCP template
`/usr/local/etc/ncp-templates/php/opcache.ini.sh`). After that, `phpenmod` updates the existing
10-priority symlink instead of creating a 20-priority duplicate.

The warning itself is **harmless** (no performance loss, OPcache stays active) — it is log noise
plus recurring manual cleanup after each PHP update.

---

## 1. Symptom

PHP prints a startup warning whenever a SAPI initialises:

```
PHP Warning:  Cannot load Zend OPcache - it was already loaded in Unknown on line 0
```

Where it shows up:

- `php -v` output (CLI), e.g. when cron or monitoring scripts invoke PHP
- `journalctl -u php8.3-fpm` (at every FPM start / reload)
- syslog — on this box roughly every 30 minutes (periodic CLI invocations)
- Nextcloud admin log is **not** affected by this warning

The duplicate symlinks are visible directly:

```bash
ls -l /etc/php/8.3/*/conf.d/*opcache*
```

```
/etc/php/8.3/cli/conf.d/10-opcache.ini -> /etc/php/8.3/mods-available/opcache.ini
/etc/php/8.3/cli/conf.d/20-opcache.ini -> /etc/php/8.3/mods-available/opcache.ini
/etc/php/8.3/fpm/conf.d/10-opcache.ini -> /etc/php/8.3/mods-available/opcache.ini
/etc/php/8.3/fpm/conf.d/20-opcache.ini -> /etc/php/8.3/mods-available/opcache.ini
```

Impact assessment:

- **No functional impact.** OPcache is loaded (once), `php -v` still reports
  `with Zend OPcache v8.3.x`.
- **Log noise**, both in journal and syslog, forever.
- **Recurring maintenance:** after every PHP package update the duplicate is back and has to be
  removed manually (`sudo rm …/20-opcache.ini`) — unless the actual cause is fixed.

[... 163 lines cut ...]
| Syslog | `grep -c 'already loaded' /var/log/syslog` | no new entries |
| Services | `systemctl is-active apache2 mariadb redis-server php8.3-fpm AdGuardHome kopia-server` | all `active` |
| Workers | `pgrep -c php8.3-fpm` | 25 (1 master + 24 workers) |
| Nextcloud | `curl -sk https://localhost/status.php` | HTTP 200, `"installed":true`, ~80 ms |
| User test | file upload in browser + sync from mobile app | passed, no new errors in NC log |
| Reboot | — | not required for a pure PHP update |

No configuration prompts appeared during the upgrade; the fix has held since.
The same pre-emptive fix was also applied to a second NCP box before its distribution upgrade
(NCP v1.57.1 → v1.58.0, Debian 12 → 13); no duplicate symlink occurred there either.

---

## 7. Suggested upstream change

The bug is not in Debian's `php-helper` (defaulting to 20 for a missing header is documented
behaviour) and not in the sury package (its original template contains the header). The
divergence is in the NCP template:

**`ncp-templates/php/opcache.ini.sh` should emit `; priority=10` as its first line**, matching
the sury template. That is a one-line change and fixes the issue for every NCP installation
using sury PHP, regardless of PHP version.

Optional hardening: after a PHP update, NCP could check for duplicate `*opcache*` symlinks in
`conf.d` and remove stale ones — but with the header in place this should never be necessary.

ph818, 2026-10-27:
This bug appeared for me after running `ncp-dist-upgrade` from bookwork to trixie. I noticed the problem because of all the new root mail. Thank you for the fixes -- I am implementing them on my system.

@Knubbel 
> ### 5.3 Permanent fix – NCP template (survives template re-runs)
> 
> `/usr/local/etc/ncp-templates/php/opcache.ini.sh` writes the file during NCP operations (Nextcloud update, `ncp-restore`, data directory change). Add the same header there, as the first line emitted by the heredoc, directly before `zend_extension=opcache.so`:
> 
> sudo cp /usr/local/etc/ncp-templates/php/opcache.ini.sh \
>         /usr/local/etc/ncp-templates/php/opcache.ini.sh.bak-$(date +%Y%m%d)
> sudo nano /usr/local/etc/ncp-templates/php/opcache.ini.sh
> # inside the "cat <<EOF" block, insert as first line:
> # ; priority=10
> 
> Then verify:
> 
> grep -n 'priority' /usr/local/etc/ncp-templates/php/opcache.ini.sh

I would like to verify, the text to insert into `/usr/local/etc/ncp-templates/php/opcache.ini.sh` is

```
# ; priority=10
``` 

with the # commenting it out, and not

```
; priority=10
``` 

without the comment?

Thank you :)