nextcloud/nextcloudpi #417: NextCloud admin pwd not as documented, reset fails in img NextCloudPi_02-06-18

wahboh, 2018-02-16:
Hi,

It seems as if the NextCloud admin password in distribution img NextCloudPi_02-06-18 either is not initiated, or the password is NOT 'ownyourbits' as documented in https://ownyourbits.com/2017/02/13/nextcloud-ready-raspberry-pi-image/

Trying to login to NextCloud with admin / ownyourbits says "wrong password", i e credentials seems not to be as in the figure "Nextcloud default configuration"

When trying to reset the password with the occ command, I get:

root@nextcloudpi:/var/www/nextcloud# sudo -u www-data php occ user:resetpassword admin
User does not exist

Logging in with NextCloud user ncp / ownyourbits works well, but then I can't create new users.

Best regards
Bo W

wahboh, 2018-02-16:
root@nextcloudpi:/var/www/nextcloud# sudo -u www-data php occ user:list
  - ncp: ncp

Does it mean that no admin user has been configured?

Best // Bo

nachoparker, 2018-02-16:
`admin` doesn't exist anymore (this was updated in ownyourbits.com). For consistency, the admin user is now called `ncp`. See [the announcement](https://ownyourbits.com/2018/02/06/nextcloudpi-updated-to-nc13-brings-ufw-security-improvements-auto-snapshots-and-more/)

What do you mean that you cannot create new users? please, use this [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) for questions and conversation, and only open an issue here when we identify a bug.

Thanks for reporting.

closed 2018-02-16 by nachoparker