nextcloud/nextcloudpi #418: NCP report coulnd't find db log file - DB misconfiguration ?

Iolaum, 2018-02-16:
### PROBLEMS

While creating the report the other issue i created I noticed that the ncp-report didn't find my database log. I was kind of surprised to I wondered how to look for it. Following [these instructions](https://serverfault.com/a/738061) I did:

```
root@nextcloudpi:/media/wd1/ncdata# mysql
Welcome to the MariaDB monitor.  Commands end with ; or \g.
Your MariaDB connection id is 3474
Server version: 10.1.23-MariaDB-9+deb9u1 Raspbian 9.0

Copyright (c) 2000, 2017, Oracle, MariaDB Corporation Ab and others.

Type 'help;' or '\h' for help. Type '\c' to clear the current input statement.

MariaDB [(none)]> show global variables like 'log_error';
+---------------+--------------------------+
| Variable_name | Value                    |
+---------------+--------------------------+
| log_error     | /var/log/mysql/error.log |
+---------------+--------------------------+
1 row in set (0.01 sec)

MariaDB [(none)]> quit
Bye
```
So I went to the above location to look at the specified directory:

```
root@nextcloudpi:/var/log# ll | grep mys
drwxr-xr-x  2 root  root     40 Feb 13 21:37 myslq/
```
 Now I was dumbfounded! Not only there's isn't a directory like the one specified but there's one that 's a mispelling of it? (I looked, it's empty.) What's going on?

Since my ncp instance is working there should be a log file somewhere. Knowing that I moved the data and database folders to the 1Tb drive of the nextcloud box I went to look there:

```
root@nextcloudpi:/var/log# cd /media/wd1/ncdata
root@nextcloudpi:/media/wd1/ncdata# ll | grep .log
-rw-r--r--  1 www-data root          0 Dec  4 14:06 access.log
-rw-r--r--  1 www-data root       3680 Dec  9 01:16 error.log
-rw-r-----  1 www-data www-data 202506 Feb 15 21:04 nextcloud.log
root@nextcloudpi:/media/wd1/ncdata# cat error.log 
[Mon Dec 04 14:06:04.844570 2017] [ssl:warn] [pid 471:tid 3069799520] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Mon Dec 04 14:06:09.979783 2017] [ssl:warn] [pid 533:tid 3069799520] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Mon Dec 04 14:09:54.881803 2017] [ssl:warn] [pid 1400:tid 3069455456] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Mon Dec 04 14:09:55.885226 2017] [ssl:warn] [pid 1402:tid 3069455456] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Mon Dec 04 15:07:47.687364 2017] [ssl:warn] [pid 4697:tid 3069963360] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Mon Dec 04 15:07:51.565935 2017] [ssl:warn] [pid 4699:tid 3069963360] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Mon Dec 04 15:16:30.562691 2017] [ssl:warn] [pid 4699:tid 3069963360] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Mon Dec 04 15:19:38.076144 2017] [ssl:warn] [pid 4699:tid 3069963360] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Mon Dec 04 15:46:37.688474 2017] [ssl:warn] [pid 479:tid 3069650960] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Mon Dec 04 15:46:43.332215 2017] [ssl:warn] [pid 553:tid 3069650960] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Mon Dec 04 15:56:14.826706 2017] [ssl:warn] [pid 479:tid 3069553760] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Mon Dec 04 15:56:20.526687 2017] [ssl:warn] [pid 552:tid 3069553760] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Mon Dec 04 15:59:21.183964 2017] [ssl:warn] [pid 610:tid 1992753152] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Mon Dec 04 15:59:22.086713 2017] [ssl:warn] [pid 818:tid 1992753152] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Sat Dec 09 00:54:38.232193 2017] [ssl:warn] [pid 546:tid 1992736768] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Sat Dec 09 00:54:39.076715 2017] [ssl:warn] [pid 754:tid 1992736768] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Sat Dec 09 01:08:14.141497 2017] [ssl:warn] [pid 560:tid 1992294400] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Sat Dec 09 01:08:15.083376 2017] [ssl:warn] [pid 774:tid 1992294400] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Sat Dec 09 01:10:23.703296 2017] [ssl:warn] [pid 596:tid 1991991296] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Sat Dec 09 01:10:24.103585 2017] [ssl:warn] [pid 778:tid 1991991296] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Sat Dec 09 01:16:17.259935 2017] [ssl:warn] [pid 605:tid 1992278016] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Sat Dec 09 01:16:18.084846 2017] [ssl:warn] [pid 807:tid 1992278016] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name

```

I have no idea how or when the database log got moved there and or what's happened with the directories but I decided to do a little digging following [this way to search a git repo](https://stackoverflow.com/a/2929502):

```
[... 132 lines cut ...]
```

{"reqId":"WoCkVX8AAQEAAAG4u6UAAA4B","level":2,"time":"2018-02-11T20:15:18+00:00","remoteAddr":"137.226.113.20","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"137.226.113.20\" tried to access using \"79.75.141.71\" as host.","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:50.0) Gecko\/20100101 Firefox\/50.0","version":"12.0.4.3"}
{"reqId":"WoCkVn8AAQEAAAG4u60AAA4J","level":2,"time":"2018-02-11T20:15:19+00:00","remoteAddr":"137.226.113.20","user":"--","app":"core","method":"GET","url":"\/index.php\/js\/core\/merged-template-prepend.js?v=2793255a2c31f4b4c2eab906af15ce25-1","message":"Trusted domain error. \"137.226.113.20\" tried to access using \"79.75.141.71\" as host.","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:50.0) Gecko\/20100101 Firefox\/50.0","version":"12.0.4.3"}
{"reqId":"WoDQD38AAQEAAAHFrL4AAABS","level":2,"time":"2018-02-11T23:21:51+00:00","remoteAddr":"107.170.227.152","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"107.170.227.152\" tried to access using \"79.75.141.71\" as host.","userAgent":"Mozilla\/5.0 zgrab\/0.x","version":"12.0.4.3"}
{"reqId":"WoF0O38AAQEAAB-8900AAAAA","level":3,"time":"2018-02-12T11:02:19+00:00","remoteAddr":"210.16.189.29","user":"--","app":"PHP","method":"GET","url":"\/admin\/config.php","message":"Exception: The requested uri(\/admin\/config.php) cannot be processed by the script '\/core\/templates\/404.php') at \/var\/www\/nextcloud\/lib\/private\/AppFramework\/Http\/Request.php#729","userAgent":"curl\/7.19.7 (i386-redhat-linux-gnu) libcurl\/7.19.7 NSS\/3.27.1 zlib\/1.2.3 libidn\/1.18 libssh2\/1.4.2","version":"12.0.4.3"}
{"reqId":"WoHJ6H8AAQEAAB-893IAAAAO","level":2,"time":"2018-02-12T17:07:53+00:00","remoteAddr":"164.52.6.150","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"164.52.6.150\" tried to access using \"79.75.151.21:443\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; rv:52.0) Gecko\/20100101 Firefox\/52.0","version":"12.0.4.3"}
{"reqId":"WoHjsn8AAQEAAB-8948AAAAS","level":2,"time":"2018-02-12T18:57:55+00:00","remoteAddr":"209.126.136.7","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"209.126.136.7\" tried to access using \"79.75.151.21\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/57.0.2987.133 Safari\/537.36","version":"12.0.4.3"}
{"reqId":"WoH69X8AAQEAACAK1dIAAABE","level":2,"time":"2018-02-12T20:37:09+00:00","remoteAddr":"139.162.78.135","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"139.162.78.135\" tried to access using \"79.75.151.21\" as host.","userAgent [... cut ...] 18.206.66\" tried to access using \"79.75.156.41\" as host.","userAgent":"--","version":"12.0.4.3"}
{"reqId":"WoX18X8AAQEAACmpzJsAAAAH","level":2,"time":"2018-02-15T21:04:50+00:00","remoteAddr":"60.191.38.77","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"60.191.38.77\" tried to access using \"79.75.156.41:443\" as host.","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10.11; rv:47.0) Gecko\/20100101 Firefox\/47.0","version":"12.0.4.3"}

```
</details>

nachoparker, 2018-02-16:
That log is not the mysql log, but the apache log. See this

https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-nextcloud.sh#L190

You are right, my mysql log is empty... let's see if we can find it. Are you using ramlogs? does it come back after doing `systemctl restart mysqld`?

Iolaum, 2018-02-16:
Restarted the process and nothing happened. Restarted rpi as well and no log appeared.

nachoparker, 2018-02-16:
are you using ramlogs?

Iolaum, 2018-02-18:
Yes I am using ramlogs. I am wondering though if it would be better to just symlink the log files on files located at the hard drive.

nachoparker, 2018-02-19:
can you try disabling ramlogs and rebooting, see it they show up?

Iolaum, 2018-03-01:
Yes, after disabling ramlogs the log file `/var/log/myql/error.log` appeared on the filesystem.

After I re-enabled ramlogs the file is still there.

nachoparker, 2018-03-03:
> After I re-enabled ramlogs the file is still there.

That is a bit weird hehe... anyhow, it is time to improve ramlogs because of things like this 

#309

nachoparker, 2018-03-03:
fixed in `v0.46.37`

closed 2018-03-03 by nachoparker