nextcloud/nextcloudpi #419: Error reading data directory - and USBdrive is mounted 8 times

RyanTG, 2018-02-16:
I might have a couple issues going on here. I started having problems last summer, but I didn't have time to troubleshoot it - but I stayed up to date with updates the whole time. I first noticed the issue when I couldn't get a connection to the server (and nextcloudpi.local doesn't load).

One symptom seems to be that the datadir "doesn't exist". 

I'm using a usbdrive for the database and datadir. 

Each time I reboot the raspberry, I get TWO new USBdrives. Check it out:

````
pi@raspberrypi:/media $ ls
cloud  USBdrive  USBdrive1  USBdrive2  USBdrive3  USBdrive4  USBdrive5  USBdrive6  USBdrive7  USBdrive8
````
Something weird with automount?

The database is in cloud (which is what I originally named the device):

````
pi@raspberrypi:/media/cloud/ncdatabase $ ls
aria_log.00000001  ibdata1      multi-master.info   nextcloud
aria_log_control   ib_logfile0  mysql               performance_schema
debian-10.1.flag   ib_logfile1  mysql_upgrade_info  tc.log
````
and for some reason it's also in USBdrive:

````
pi@raspberrypi:/media/USBdrive/ncdatabase $ ls
aria_log.00000001  ibdata1      multi-master.info   nextcloud
aria_log_control   ib_logfile0  mysql               performance_schema
debian-10.1.flag   ib_logfile1  mysql_upgrade_info  tc.log
````

and 'data' is in USBdrive8 (even though USBdrive8 didn't exist before I recently rebooted):

````
pi@raspberrypi:/media/USBdrive8 $ ls
data  lost+found  nextcloud-bkp_20170825.tar
````

But if I try to change the datadir to USBdrive8, I get:

````
Cannot load Zend OPcache - it was already loaded
Error reading data directory. Is NextCloud running and configured?
````

I'm totally fine with just reinstalling everything (especially if there's a way to do it without reflashing the SD). THANKS!

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

cat: /usr/local/etc/ncp-baseimage: No such file or directory
Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v0.46.10
NextCloudPi image
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/USBdrive/data (doesn't exist)
rootfs usage         2.1G/7.3G
swapfile             /var/swap
Nextcloud check      error
HTTPD service        down
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      down
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.102
gateway              192.168.1.1
interface            eth0
certificates         none
certs due            none
NAT loopback         no
uptime               11:49

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
Your data directory is invalid
Please check that the data directory contains a file ".ocdata" in its root.

Cannot create "data" directory
This can usually be fixed by giving the webserver write access to the root directory. See https://docs.nextcloud.com/server/12/go.php?to=admin-dir_permissions

An unhandled exception has been thrown:
Exception: Environment not properly prepared. in /var/www/nextcloud/lib/private/Console/Application.php:145
Stack trace:
#0 /var/www/nextcloud/console.php(91): OC\Console\Application->loadCommands(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#1 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#2 {main}
```
</details>

<details>
<summary>HTTPd logs</summary>

```

```
</details>

<details>
<summary>Database logs</summary>

```

2018-02-16  6:49:32 1989070848 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2018-02-16  6:49:32 1989070848 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2018-02-16  6:49:32 1989070848 [Note] InnoDB: Compressed tables use zlib 1.2.8
2018-02-16  6:49:32 1989070848 [Note] InnoDB: Using Linux native AIO
2018-02-16  6:49:32 1989070848 [Note] InnoDB: Using generic crc32 instructions
2018-02-16  6:49:32 1989070848 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2018-02-16  6:49:32 1989070848 [Note] InnoDB: Completed initialization of buffer pool
2018-02-16  6:49:33 1989070848 [Note] InnoDB: Highest supported file format is Barracuda.
2018-02-16  6:49:36 1989070848 [Note] InnoDB: 128 rollback segment(s) are active.
2018-02-16  6:49:36 1989070848 [Note] InnoDB: Waiting for purge to start
2018-02-16  6:49:36 1989070848 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 126035164
2018-02-16  6:49:36 1420985152 [Note] InnoDB: Dumping buffer pool(s) not yet started
2018-02-16  6:49:36 1989070848 [Note] Plugin 'FEEDBACK' is disabled.
2018-02-16  6:49:36 1989070848 [Note] Server socket created on IP: '127.0.0.1'.
2018-02-16 06:49:36 61bdf340 InnoDB: Error: Column last_update in table "mysql"."innodb_table_stats" is INT UNSIGNED NOT NULL but should be BINARY(4) NOT NULL (type mismatch).
2018-02-16 06:49:36 61bdf340 InnoDB: Error: Fetch of persistent statistics requested for table "mysql"."gtid_slave_pos" but the required system tables mysql.innodb_table_stats and mysql.innodb_index_stats are not present or have unexpected structure. Using transient stats instead.
2018-02-16  6:49:36 1989070848 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0
2018-02-16 06:49:50 61bae340 InnoDB: Error: Column last_update in table "mysql"."innodb_table_stats" is INT UNSIGNED NOT NULL but should be BINARY(4) NOT NULL (type mismatch).
2018-02-16 06:49:50 61bae340 InnoDB: Error: Fetch of persistent statistics requested for table "nextcloud"."oc_appconfig" but the required system tables mysql.innodb_table_stats and mysql.innodb_index_stats are not present or have unexpected structure. Using transient stats instead.

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/media/USBdrive/data/nextcloud.log' for reading: No such file or directory

```
</details>

nachoparker, 2018-02-16:
I can see that you have a very old image and at the very least you didn't update it regularly in the past.

All that stuff comes because you are mixing the old and the new automount features (because of not regular upgrades), so you have a mess going on.

Automount is pretty stable and has been working fine for a long time. I suggest that you

- run `ncp-update`
- fix this just enough to be able to run nextcloud
- run `nc-backup` and `nc-export`
- flash the latest image
- restore everything, as per [this instructions](https://github.com/nextcloud/nextcloudpi/wiki/How-to-backup-and-restore-a-NextCloudPi-instance-using-nextcloudpi-config)

Also I suggest that you activate `nc-autoupdate-ncp`, so you don't run into weird situations like this in the future. If you run the wizard, this will be activated automatically.

nachoparker, 2018-02-16:
Now, going into point 2, try to find where your data directory is (you can recognize it because there is a `.ocdata` hidden file inside), and your database directory if you also moved that.

Then, update `/var/www/nextcloud/config/config.php` to use the correct path for `datadir`, and `/etc/mysql/mariadb.conf.d/50-server.cnf` also to use the correct path for `datadir` (in this case the location of your database, if you moved it).

Reboot after that and Nextcloud should be running.

RyanTG, 2018-02-16:
Thanks. Oh, I guess I was semi-regularly updating ncp (I updated that every couple months, including yeterday) but not the raspbian image?

`nc-backup` failed because of the data directory issue. 

Given that USBdrive8 doesn't have the .ocdata file, then I don't know where the real data folder disappeared to. I'm just going to reinstall. There's no tool in ncp to reinstall, right? I'll need to flash it on the SD card?

nachoparker, 2018-02-18:
that sounds like a good option. You should update NCP not less than once a month.

Start fresh with the latest image, and run the wizard which will enable autoupdates.