nextcloud/nextcloudpi #420: Nextcloud borked after upgrade to 12.0.5

leavenotracks, 2018-02-16:
Hello,

First of all thank you for putting together this project - this and the whole of the Nextcloud project has been very inspirational. I am now well and truly on the road to using only freedom-respecting software. I hope to be able to contribute in some way in the future to this or other projects.

<b>Background:</b>
I flash the image of NCP onto an SD card wen through the WebUI install, followed instructions to automount 2 hard disks - one to hold data/database and the other for back-ups. This was not wholly successful (see below). All the other essential features such as letsencrypt, moving datadir/databasedir, etc I was able to set up without too much problem. Before long I had a working Nextcloud instance - woop woop! I set up one admin account, and one other, which we were able to use without any noticeable problems.

After restarting the Pi at some point I noticed that the the second USB drive was no longer mounted, and trying to do a backup would throw up errors. I was hoping to resolve this issue just by adding some lines in fstab and whatever else is necessary.

<b>Issue:</b>
My main problem, however, arose after upgrading to Nextcloud 12.0.5 a few weeks back. I upgraded via the the Nextcloud WebUI and it seemed to work fine thereafter. Not long afterwards I found that I was not able to log into Nextcloud via the WebUI, instead I would get a <i>'Create an admin account'</i> page, similar to the Nextcloud login page.

My ncdata and ncdatabase folders are still on the primary USB drive along with data that I had previously uploaded. It seems like the Nextcloud configuration has been lost in the upgrade and I'm unsure on how to proceed to reinstate it. Another strange observation is that the SD card holing the NCP image is now completely full.

Would basically like to reinstate the nextcloud instance as it was if possible, with the main drive/backup working. Eventually I would also like to upgrade to NC13, but less drama would be nice!

Thank you in advance for any help on this.

Charles

Paste the output of `sudo ncp-report`
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.46.4
NextCloudPi image    NextCloudPi_12-04-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda sdb
datadir               (doesn't exist)
rootfs usage         15G/15G
swapfile             /var/swap
Nextcloud check      ok
Nextcloud version    12.0.5.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.82
gateway              192.168.1.254
interface            eth0
certificates         none
certs due            none
NAT loopback         no
uptime               2days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud is not installed - only a limited number of commands are available

  [Symfony\Component\Console\Exception\CommandNotFoundException]
  There are no commands defined in the "config" namespace.

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Fri Feb 16 06:25:02.000255 2018] [core:notice] [pid 962:tid 1992519680] AH00094: Command line: '/usr/sbin/apache2'
[Fri Feb 16 08:17:59.814478 2018] [log_config:warn] [pid 23954:tid 1681912880] (28)No space left on device: [client 107.6.183.226:48159] AH00646: Error writing to /var/log/apache2/other_vhosts_access.log
[Fri Feb 16 08:18:02.561030 2018] [log_config:warn] [pid 23954:tid 1673524272] (28)No space left on device: [client 107.6.183.228:38647] AH00646: Error writing to /var/log/apache2/other_vhosts_access.log
[Fri Feb 16 08:18:02.701659 2018] [log_config:warn] [pid 23954:tid 1665135664] (28)No space left on device: [client 107.6.183.229:46157] AH00646: Error writing to /var/log/apache2/other_vhosts_access.log
[Fri Feb 16 09:15:49.362037 2018] [log_config:warn] [pid 23954:tid 1606415408] (28)No space left on device: [client 71.6.167.142:45137] AH00646: Error writing to /var/log/apache2/other_vhosts_access.log
[Fri Feb 16 10:16:01.530619 2018] [log_config:warn] [pid 23954:tid 1765798960] (28)No space left on device: [client 169.57.152.101:55860] AH00646: Error writing to /var/log/apache2/other_vhosts_access.log
[Fri Feb 16 10:57:01.515855 2018] [log_config:warn] [pid 23954:tid 1757410352] (28)No space left on device: [client 2.241.143.231:41372] AH00646: Error writing to /var/log/apache2/other_vhosts_access.log
[Fri Feb 16 13:00:12.666814 2018] [log_config:warn] [pid 23953:tid 1749021744] (28)No space left on device: [client 60.191.38.77:63914] AH00646: Error writing to /var/log/apache2/other_vhosts_access.log
[Fri Feb 16 17:44:45.695387 2018] [log_config:warn] [pid 23953:tid 1665135664] (28)No space left on device: [client 169.57.152.101:40042] AH00646: Error writing to /var/log/apache2/other_vhosts_access.log
[Fri Feb 16 18:30:27.535706 2018] [authz_host:error] [pid 23954:tid 1910150192] [client 192.168.1.72:56612] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Fri Feb 16 18:30:27.536868 2018] [log_config:warn] [pid 23954:tid 1631581232] (28)No space left on device: [client 192.168.1.72:56612] AH00646: Error writing to /var/log/apache2/other_vhosts_access.log
[Fri Feb 16 18:30:46.161096 2018] [authz_host:error] [pid 23954:tid 1901761584] [client 192.168.1.72:56614] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Fri Feb 16 18:30:46.186013 2018] [authnz_external:error] [pid 23954:tid 1901761584] [client 192.168.1.72:56614] AuthExtern pwauth [/usr/sbin/pwauth]: Failed (1) for user admin
[Fri Feb 16 18:30:46.186141 2018] [auth_basic:error] [pid 23954:tid 1901761584] [client 192.168.1.72:56614] AH01617: user admin: authentication failure for "/": Password Mismatch
[Fri Feb 16 18:30:46.187286 2018] [log_config:warn] [pid 23954:tid 1623192624] (28)No space left on device: [client 192.168.1.72:56614] AH00646: Error writing to /var/log/apache2/other_vhosts_access.log
[Fri Feb 16 18:31:21.479239 2018] [authz_host:error] [pid 23954:tid 1893372976] [client 192.168.1.72:56617] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Fri Feb 16 18:31:21.659994 2018] [authz_host:error] [pid 23954:tid 1893372976] [client 192.168.1.72:56617] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Fri Feb 16 18:31:21.668050 2018] [authz_host:error] [pid 23954:tid 1893372976] [client 192.168.1.72:56617] AH01753: access check of 'localhost' to /index.php failed, reason: unable to get the remote host name
[Fri Feb 16 18:31:21.780368 2018] [authz_host:error] [pid 23954:tid 1893372976] [client 192.168.1.72:56617] AH01753: access check of 'localhost' to /index.php failed, reason: unable to get the remote host name
[Fri Feb 16 18:31:22.19768
```
</details>

<details>
<summary>Database logs</summary>

```

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/nextcloud.log' for reading: No such file or directory

```
</details>

nachoparker, 2018-02-19:
```
USB devices          sda sdb
datadir               (doesn't exist)
rootfs usage         15G/15G
```

Well, your rootfs definitely is full. Free up some space.

Then it would be interesting to see if really there is a problem with automount or what happened.

```
ls -l /media
```

```
systemctl status automount
```

```
systemctl status automount-links
```

leavenotracks, 2018-02-19:
Hi Nacho,

I have been taking a look at it -

brickCloud and bckp_brickCloud are my main and backup USB HDs respectively. If I run
`sudo mount -l`

I get the following:
[image]

I see that brickCloud is there while bckp_brickCloud is not. However when I do:
`ls -l /media`
I get:
[image]

Now, prior to updating to 12.05, I tried to backup the nextcloud instance to bckp_brickCloud. My hunch is that instead of the backup being created in the USB HD, it has been created it in the SD card:

[image]

Since the size of the tar file is more or less what it would take to fill up the SD card I guess that must explain the SD card filling up? Since this is not going to be a complete backup I guess I can just delete?

nachoparker, 2018-02-19:
Seems like your hunch is correct. I only see the `brickCloud` mounted, not the other one. You didn't do any modifications to `/etc/fstab`, right?

You can probably delete that backup to reclaim that space from your SD card. This command will verify that the assumptions are correct ( if it outputs `/` as the mountpoint, which is your SD ).

```
sudo stat -c "%m" /media/bckp_brickCloud/nextcloud-bkp_20180210.tar
```

Sorry, I misstyped

```
systemctl status nc-automount
```

```
systemctl status nc-automount-links
```

leavenotracks, 2018-02-19:
I moved the file to my brickCloud HD in case. In any case running:
`sudo stat -c "%m" /media/bckp_brickCloud/'
indeed yields
`/`

Now, output of `systemctl` commands:

[image]

nachoparker, 2018-02-19:
well, automount obviously does work, because one of your drives is there, why would the other one not be there?

can you mount it manually through command line? ( I guess it's `/dev/sda1`, but you can check with `lsblk -f` )

leavenotracks, 2018-02-19:
Output of `lsblk -f`:
[image]

sda1 confirmed... but alas cannot seem to mount:

[image]

nachoparker, 2018-02-19:
As suspected... I have seen this before. Your partition table/drive is corrupt and that is why it wasn't mounting.

Try with 

```
sudo btrfs check --repair /dev/sda
```

or similar... or just run `nc-format-USB` again if you don't care about the contents.

closed 2018-02-19 by nachoparker

leavenotracks, 2018-02-19:
Ok - formatted second USB, restarted, and now `mount -l` gives:

[image]

So now I can see both brickCloud and bckp_brickCloud.

However, the output of `ls -l /media` gives the following:

[image]

Does this look right??

leavenotracks, 2018-02-19:
Hi Nacho,

I see you've closed the issue  - apologies, perhaps I should have created two issues (or one question and one issue), but the main thrust of the problem was that the nextcloud instance was not working following upgrade to 12.05:

> My main problem, however, arose after upgrading to Nextcloud 12.0.5 a few weeks back. I upgraded via the the Nextcloud WebUI and it seemed to work fine thereafter. Not long afterwards I found that I was not able to log into Nextcloud via the WebUI, instead I would get a 'Create an admin account' page, similar to the Nextcloud login page.

> My ncdata and ncdatabase folders are still on the primary USB drive along with data that I had previously uploaded. It seems like the Nextcloud configuration has been lost in the upgrade and I'm [...]

> Would basically like to reinstate the nextcloud instance as it was if possible, with the main drive/backup working. Eventually I would also like to upgrade to NC13, but less drama would be nice!

Should I post an issue separately?

Thank you for all your help so far!

nachoparker, 2018-02-20:
Well, I close them when I think it is not a NCP problem ( like this case, where you had a corrupted drive ).

This doesn't mean that we can't still talk here, but it is easier for me to be organized this way so I can spot important problems more easily.

Your other problem came probably from the fact that there was no free space left on the SD card. If you can restore that should solve your issue. Probably it would suffice to restore the `config.php` file alone, but otherwise just do the `nc-restore`, but that is going to take longer because you included a lot of data in the backup.

leavenotracks, 2018-02-20:
Thanks Nacho. I would be interested to know why the drive became corrupted - all I did was format it using `nc-format-USB` and I may have had one or two old backups on there, but nothing more. This highlights to me how essential it is to have redundancy here.

On restoring the settings etc - I'm reluctant to do `nc-restore` since I'm not convinced that the backup is a full backup. It was only as big as the free space on the SD card allowed it to be.

What I will try is to extract the `config.php` from the backup and restore that. Once I have it up and running again, I would like to upgrade to NC13. If the `config.php` is not there, is there somewhere I can refer to to populate config.php correctly?

If all goes well, will then do a full backup and attempt upgrade to NC13 stable.

nachoparker, 2018-02-20:
yeah, that's a good point. Probably that backup is not good, idk if you have older ones.

those dirs are not a problem. What I don't know is what you did when you were trying to fix, but it can work like that.

leavenotracks, 2018-02-20:
There's a couple of `config.php` files in the (incomplete) backup:

[image]

Suppose it's a case of looking at them and seeing, which looks more right? If I am not able to retrieve the `config.php` and restore - is there a reference for what this file should contain so I can populate it myself?

leavenotracks, 2018-02-20:
I realise my last question was a little silly, given that there are system-generated passwords in `config.php` etc.

In any case I managed to restore the file, restarted and looks like nextcloud is back up and running. I am now trying to create a backup to `/media/bckp_brickCloud` so that I can attempt to upgrade to NC13 but running into problems...

[image]

Looking at the below however....

[image]

...it looks as if the sda/sdb have swapped and mountpoints have changed or is this more or less what I should have expected?

So have gone back to `sudo nextcloudpi-config`, `nc-backup` and modified `DESTDIR' to '/media/bckp_brickCloud1` and seemed to start smoothly and then....

[image]

Looks like the destination drive is read-only. This is after having re-formatted it etc. Tried to remount...
[image]

...without success, any ideas?

leavenotracks, 2018-02-21:
So after resetting the pi, filesystem on backup USB seemed to be read-write again. Tried to back-up again using nc-backup and got as far as approx. 8GB:
[image]
There were a lot of of these along the way... `tar: Skipping to next header`.

And now the filesystem has reverted to read-only again:

[image]

Any ideas on what could be causing this behaviour?

nachoparker, 2018-02-21:
looks to me that you are having input/output errors and linux is reacting by changing the mountpoint to read only

maybe you can try ext4 if you are out of ideas, but I think that your drive might be faulty at this point

leavenotracks, 2018-02-21:
argh ok - is there any way to tell conclusively if the drive is shot?

Presumably reformatting to ext4 would not eliminate the fault necessarily?

nachoparker, 2018-02-21:
> Presumably reformatting to ext4 would not eliminate the fault necessarily?

No, it is just to rule out BTRFS. If you have I/O problems there is no good solution other than replacing the thing.

leavenotracks, 2018-02-21:
Re-formatted to ext4 - still getting issues.

Nacho just a thought  I have the latest RPi 3 and I am trying to power both USB HDs from it - could this potentially be the issue? having trawled through the forums, this does seem to be a problem for people, albeit on older RPi models.

nachoparker, 2018-02-21:
> Nacho just a thought I have the latest RPi 3 and I am trying to power both USB HDs from it - could this potentially be the issue? having trawled through the forums, this does seem to be a problem for people, albeit on older RPi models.

Of course this can be the issue. Probably it is!!

https://github.com/nextcloud/nextcloudpi/wiki/FAQ#why-is-nextcloud-so-slow--failing-randomly--getting-server-errors-even-with-a-fresh-image

Most issues come from power and SD cards. It is very much recommended to use USB drives with their own power supply.

leavenotracks, 2018-02-23:
Y USB cable arrived today alongside powered USB hub - I can tell you straight away that there is a noticeable difference with how the hard drives are spinning now. Before, there was an irregular clicking noise that I had assumed to be normal! They have not clicked once since being powered independently. - whisper silent :)

OK - so reformatted to btrfs and once again attemped to do a back up. I still got an error though:

[image]

Just one error this time though, and the tar file generated is now much larger than before. Previously the tar generated was only 5-6GB. Now the size is approximating what is should be:

[image]

In the backup drive I have 193GB vs 206GB in the main drive. Still a big difference though not so big as before. Wonder if there is a way I can determine how complete the backup is?

On another note - having other issues connecting to the nextcloud instance, whether it be through my iphone or Web UI:

[image]

will update with output of `sudo ncp-report` in due course...

leavenotracks, 2018-02-25:
I upgraded to NC 13 via command line - now seems to be working smoothly :)

nachoparker, 2018-02-26:
ok, so the process still fails but seems more robust. 

No doubt there is a hardware issue. Double check that you are providing enough power to both the pi and the HDDs. I use a separated plug for each

leavenotracks, 2018-02-27:
scratch that - I wouldn't say smoothly...more sporadically. Getting a lot of 'internal server error' messages on iphone app. There appear to be a bunch of images in the phone app that can't be downloaded from the Nextcloud instance.

I appreciate that this is probably more of a NC issue so will post there.

nachoparker, 2018-02-27:
ok, let us know how it goes.

[this thread](https://github.com/nextcloud/nextcloudpi/issues/406) might be related

nero120, 2018-02-27:
@leavenotracks just for comparison, I run a 1TB drive as my main drive plus a 256GB SSD as my backup drive, both formatted to BTRFS, with an RP3. I use nc-backup weekly to backup my NC instance plus data with in total is ~50GB. Not had any issues yet so I don't see why running two drives from the RP3 with BTRFS shouldn't work fine for you as well.

nachoparker, 2018-02-27:
@nero120 thanks for the feedback

leavenotracks, 2018-03-02:
@nero120 thanks for sharing. I can say conclusively that supplying independent power to the two portable HDs has made an audible difference if nothing else! They were both making disconcerting clicking noises fairly frequently until I powered them from the powered USB hub.

I would be interested to know if you ever have issues with speed or errors such as "unable to download file/photo", or the dreaded "Internal server error"? * I have set up only 2 users on there, I connect either via the webUI or iOS app, the other uses Android/webUI. Android user can't even see any files/folders. On iOS I see photos etc though some are missing thumbnails and clicking on them results in a "unable to download" error. Sometimes just refreshing produces an error also.

(*) I see from the issue Nacho mentions above that you too have had issues :)

There is also strange behavior with the iOS app - for example Auto-upload -- > Upload the whole camera roll ... creates empty directory tree based on the 'date taken' info for all images in the device, but does not upload any images. Only seems to upload new photos or manually uploaded photos.

I wish I could make more sense of the logs, I'll get there eventually! Thanks Nacho for all your support so far.

nero120, 2018-03-03:
@leavenotracks perhaps I need to look into getting a powered hub for the drives then! I don't like the idea of the RP3 struggling to supply enough power to them leading to any kind of data corruption.

As detailed in #406, my issues typically occur when I'm looking at photos (I have thousands on my NC instance) or browsing folders with photos in them, using either the webUI or mobile app. I use the Preview Generator app and have preview generation scheduled via `occ preview:pre-generate` which as far as I can tell, works fine.

When I'm browsing photos however, performance always degrades until I get timeouts and server 500 errors. I believe this happened much sooner before I had the Preview Generator app set up, as the thumbnails would only appear very slowly. Now they appear quickly, but the NC site still dies fairly soon after I start browsing photos.

If I run `top` I can see 1-3 `php-fpm7.0' processes running, the first uses >35% CPU, and the others appear and disappear continuously and use significant amounts of CPU as well:

[image]

This is just from me browsing some photos! No uploading happening, or anyone else accessing the server.

leavenotracks, 2018-03-19:
Sorry for delayed reply - hectic couple of weeks.

@nero120  I can confirm that when running `top`  I also see 2/3 (sometimes 4!) php processes, all taking up approx. same amount of CPU.

Not had any joy from my NC instance in the last few weeks. Have posted some issues on the nextcloud/server forum. Will see how it goes, in the meantime, trying to educate myself a little more on mysql/php/apache :smile:

nero120, 2018-03-20:
Yeah that's normal. Multiple PHP FPM processes are spawned to deal with requests, limited to a maximum of `pm.max_children` in `/etc/php/7.0/fpm/pool.d/www.conf`. I've been [playing with these settings](https://github.com/nextcloud/nextcloudpi/issues/406) in order to try and get a more stable service but so far no joy.