nextcloud/nextcloudpi #424: SMB/CIFS for data directory

RyanTG, 2018-02-19:
Is there a workaround to `Only ext/btrfs filesystems can hold the data directory`? I was hoping to use a mounted remote drive for storage. Thanks.

Paste the output of `sudo ncp-report`

nachoparker, 2018-02-20:
hi,

please use the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) for technical questions and conversation

closed 2018-02-20 by nachoparker

RyanTG, 2018-02-20:
Hmm OK. I think nextcloud generally supports smb/cifs. But it's your https://github.com/nextcloud/nextcloudpi/etc/nextcloudpi-config.d/nc-datadir.sh script that is saying nope. I'll just set it up manually.

theCalcaholic, 2018-02-21:
I think you can just use nc-datadir with different file formats and ignore the warning. You should just be aware that some file systems don't support the Linux permission system.