nextcloud/nextcloudpi #426: Procedure to migrate data to a fresh install

burgerkingGH, 2018-02-22:
A huge thanks for this brilliant piece of work. 

I was running on Raspbian 8, and needed to reimage my SD card to get the latest version of ncp. I attempted to restore from a database backup, but it didn't work. But it doesn't seem like the database is so important to migrate anyway. 

Therefore, my plan of attack was to setup the new server, recreate all the users, and then point the datadir to the existing location. This failed, since nc-datadir only moves to empty directories. The procedure that eventually worked was this:

First rename the existing folder with the data dir in it.

`mv nextcloud nextcloud12`

Then create a new one.

`mkdir nextcloud`

Fix the permissions as follows (otherwise nc-datadir will fail):

`sudo chown -R www-data:www-data /media/USBDrive/nextcloud`
`sudo chmod -R 750 /media/USBDrive/nextcloud`

Then run 

`sudo nextcloudpi-config ## nc-datadir, and use /media/USBDrive/nextcloud/data`

Recreate each user from an admin account on the server. Now copy all the data back into /media/USBDrive/nextcloud/data for each individual user. Finally, 

`nc-scan`

So first of all, is there a better way to do this? I couldn't find any info in my searches. 

Now for the feature request. I think there is an improvement that could facilitate this process. Namely, if nc-fix-permissions allowed us to chose any directory, to prepare it with the appropriate permissions for nc-datadir to work. This would allow us to fix the permissions of the folder, and then relocate with nc-datadir without needing the chown/mod command line step. Ultimately, it would be nice to be able to point to a non-empty data dir. I know there are potential complications, but it seems tractable.

Any thoughts?

Thanks!

Paste the output of `sudo ncp-report`

nachoparker, 2018-02-27:
Why don't you use `nc-backup` and `nc-restore`? it's the best approach. Take a look at [this guide](https://github.com/nextcloud/nextcloudpi/wiki/How-to-backup-and-restore-a-NextCloudPi-instance-using-nextcloudpi-config)

In the future, please use the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) for questions and technical conversation.

Thanks for your feedback!

nachoparker, 2018-02-27:
(closed, but can still talk here)

closed 2018-02-27 by nachoparker