nextcloud/nextcloudpi #429: Fail2ban start error

pehebo, 2018-02-26:


nachoparker, 2018-02-27:
output of `sudo ncp-output` please?

pehebo, 2018-02-27:
ncp-report (i removed some sensitive information about my system):
<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.46.34
NextCloudPi image    NextCloudPi_12-04-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      3,1G/1,9T
rootfs usage         2,0G/15G
swapfile             /var/swap
Nextcloud check      ok
Nextcloud version    13.0.0.14
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   XXXXXXXXXX
gateway              XXXXXXXXXX
interface            eth0
certificates         XXXXXXXXXXX 
certs due            70 days 
NAT loopback         no
uptime               0min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "4": "XXXXXXXX",
            "5": "XXXXXXl",
            "1": "XXXXXXX",
            "3": "XXXXXXX"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "https:\/\/ncstormrunner.strangled.net",
        "dbtype": "mysql",
        "version": "13.0.0.14",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
[... 70 lines cut ...]
```

{"reqId":"WpRBOH8AAQEAAANnv7IAAABG","level":2,"time":"2018-02-26T17:17:45+00:00","remoteAddr":"99.153.197.208","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"99.153.197.208\" tried to access using \"217.85.16.24\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"13.0.0.14"}
{"reqId":"WpRBsX8AAQEAAANnv9AAAABV","level":2,"time":"2018-02-26T17:19:45+00:00","remoteAddr":"99.153.197.208","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"99.153.197.208\" tried to access using \"217.85.16.24\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"13.0.0.14"}
{"reqId":"WpRCKn8AAQEAAANnv9EAAABW","level":2,"time":"2018-02-26T17:21:46+00:00","remoteAddr":"99.153.197.208","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"99.153.197.208\" tried to access using \"217.85.16.24\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"13.0.0.14"}
{"reqId":"WpRCo38AAQEAAANnv9IAAABX","level":2,"time":"2018-02-26T17:23:47+00:00","remoteAddr":"99.153.197.208","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"99.153.197.208\" tried to access using \"217.85.16.24\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"13.0.0.14"}
{"reqId":"WpRDG38AAQEAAANmQ6gAAAAQ","level":2,"time":"2018-02-26T17:25:47+00:00","remoteAddr":"99.153.197.208","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"99.153.197.208\" tried to access using \"217.85.16.24\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"13.0.0.14"}
{"reqId":"WpRqxn8AAQEAAANtDLYAAABJ","level":2,"time":"2018-02-26T20:15:03+00:00","remoteAddr":"120.234.128.5","user":"--","app":"core","method":"GET","url":"\/admin\/config.php","message":"Trusted domain error. \"120.234.128.5\" tried to access using \"217.85.16.24\" as host.","userAgent":"curl\/7.15.5 (i686-redhat-linux-gnu) libcurl\/7.15.5 OpenSSL\/0.9.8b zlib\/1.2.3 libidn\/0.6.5","version":"13.0.0.14"}
{"reqId":"WpRsDX8AAQEAAANtDMoAAABN","level":2,"time":"2018-02-26T20:20:30+00:00","remoteAddr":"71.6.165.200","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"71.6.165.200\" tried to access using \"217.85.16.24\" as host.","userAgent":"Mozilla\/5.0 (X11; Linux x86_64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/34.0.1847.137 Safari\/537.36","version":"13.0.0.14"}
{"reqId":"WpRsHH8AAQEAAANtDMsAAABT","level":2,"time":"2018-02-26T20:20:44+00:00","remoteAddr":"71.6.165.200","user":"--","app":"core","method":"GET","url":"\/sitemap.xml","message":"Trusted domain error. \"71.6.165.200\" tried to access using \"217.85.16.24\" as host.","userAgent":"--","version":"13.0.0.14"}
{"reqId":"WpRsHn8AAQEAAANtDMwAAABU","level":2,"time":"2018-02-26T20:20:46+00:00","remoteAddr":"71.6.165.200","user":"--","app":"core","method":"GET","url":"\/.well-known\/security.txt","message":"Trusted domain error. \"71.6.165.200\" tried to access using \"217.85.16.24\" as host.","userAgent":"--","version":"13.0.0.14"}
[... cut ...] la\/5.0 (Macintosh; Intel Mac OS X 10_11_5) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/50.0.2661.102 Safari\/537.36","version":"13.0.0.14"}
{"reqId":"WpTcXn8AAQEAAANtDwAAAABA","level":2,"time":"2018-02-27T04:19:42+00:00","remoteAddr":"62.212.68.129","user":"--","app":"core","method":"GET","url":"\/recordings\/\/theme\/main.css","message":"Trusted domain error. \"62.212.68.129\" tried to access using \"217.85.16.24\" as host.","userAgent":"curl\/7.29.0","version":"13.0.0.14"}

```
</details>

nachoparker, 2018-02-27:
what does `/var/log/fail2ban.log` say?

pehebo, 2018-02-27:
fail2ban.log (after starting service manually):
2018-02-27 12:13:03,280 fail2ban.server         [1575]: INFO    Changed logging target to /var/log/fail2ban.log for Fail2ban v0.9.6
2018-02-27 12:13:03,303 fail2ban.database       [1575]: INFO    Connected to fail2ban persistent database '/var/lib/fail2ban/fail2ban.sqlite3'
2018-02-27 12:13:03,323 fail2ban.jail           [1575]: INFO    Creating new jail 'ssh'
2018-02-27 12:13:03,340 fail2ban.jail           [1575]: INFO    Jail 'ssh' uses poller {}
2018-02-27 12:13:03,428 fail2ban.jail           [1575]: INFO    Initiated 'polling' backend
2018-02-27 12:13:03,431 fail2ban.filter         [1575]: INFO    Set maxRetry = 6
2018-02-27 12:13:03,433 fail2ban.filter         [1575]: INFO    Added logfile = /var/log/auth.log
2018-02-27 12:13:03,435 fail2ban.filter         [1575]: INFO    Set findtime = 600
2018-02-27 12:13:03,436 fail2ban.actions        [1575]: INFO    Set banTime = 600
2018-02-27 12:13:03,438 fail2ban.filter         [1575]: INFO    Set maxlines = 10
2018-02-27 12:13:03,849 fail2ban.server         [1575]: INFO    Jail ssh is not a JournalFilter instance
2018-02-27 12:13:03,878 fail2ban.jail           [1575]: INFO    Creating new jail 'nextcloud'
2018-02-27 12:13:03,883 fail2ban.jail           [1575]: INFO    Jail 'nextcloud' uses poller {}
2018-02-27 12:13:03,888 fail2ban.jail           [1575]: INFO    Initiated 'polling' backend
2018-02-27 12:13:03,890 fail2ban.filter         [1575]: INFO    Set maxRetry = 6
2018-02-27 12:13:04,439 fail2ban.filter         [1575]: INFO    Added logfile = /media/USBdrive/ncdata/nextcloud.log
2018-02-27 12:13:04,440 fail2ban.filter         [1575]: INFO    Set findtime = 600
2018-02-27 12:13:04,441 fail2ban.actions        [1575]: INFO    Set banTime = 600
2018-02-27 12:13:04,497 fail2ban.jail           [1575]: INFO    Jail 'ssh' started
2018-02-27 12:13:04,535 fail2ban.jail           [1575]: INFO    Jail 'nextcloud' started

Best regards,
Peter

nachoparker, 2018-02-27:
So it starts manually... do you mean that it doesn't start at boot? 

can you get the log in this situation?

are you using ramlogs?

pehebo, 2018-02-27:
Yes, I am using ramlogs. It failes to start at boot, saying that it was started too early (my words..:).
Startin it manually afterwards succeeds... I think it is a simple question of the moment the service is started??

pehebo, 2018-02-27:
As I mentioned at starting this, the service failes with the message
fail2ban.service: Start request repeated too quickly.
...

nachoparker, 2018-02-27:
what about the log? `/var/log/fail2ban.log`

pehebo, 2018-02-27:
Look above.. i already sent it...

nachoparker, 2018-02-28:
I mean, what's the contents of the log _when it is on a failed state_

pehebo, 2018-02-28:
Sorry for misunderstanding you :)
There is no /var/log/fail2ban.log after reboot, when the service is failing..
:(

nachoparker, 2018-02-28:
can you try disabling ramlogs?

pehebo, 2018-02-28:
No luck! Disabled ramlogs through nextcloudpi, rebootet. Still fail2ban service failes with error:
fail2ban.service: Start request repeated too quickly.

pehebo, 2018-02-28:
Maybe I found a hint in journalctl:
fail2ban-client[505]: WARNING 'backend' not defined in 'ssh'. Using default one: 'auto'
fail2ban-client[505]: WARNING 'backend' not defined in 'nextcloud'. Using default one: 'auto'
fail2ban-client[505]: **ERROR  No file(s) found for glob /media/USBdrive/ncdata/nextcloud.log**
fail2ban-client[505]: ERROR  Failed during configuration: Have not found any log file for nextcloud jail
systemd[1]: fail2ban.service: Control process exited, code=exited status=255
systemd[1]: Failed to start Fail2Ban Service.

Fail2ban starts too early, before nextcloud.log is available?

nachoparker, 2018-02-28:
ok, so it must be because of automount...

nachoparker, 2018-03-03:
I reviewed the dependencies for automount. I think it should be fixed in `v0.46.37`, but I cannot verify because this issue doesn't happen to me.

Please, run

```
sudo ncp-update devel
```

, and verify. 

We will reopen if the issue persists.

Thanks for the feedback!

closed 2018-03-03 by nachoparker

pehebo, 2018-03-03:
Hello,
thank you very much for caring about my problem so intensively!
Sorry to say that the new version doesnt fix the problem :(
Installed 0.46.37, rebooted, but error still persists (journalctl):
fail2ban-client[513]: WARNING 'backend' not defined in 'ssh'. Using default one: 'auto'
fail2ban-client[513]: WARNING 'backend' not defined in 'nextcloud'. Using default one: 'auto'
fail2ban-client[513]: ERROR  No file(s) found for glob /media/USBdrive/ncdata/nextcloud.log
fail2ban-client[513]: ERROR  Failed during configuration: Have not found any log file for nextcloud jail
systemd[1]: fail2ban.service: Control process exited, code=exited status=255
systemd[1]: Failed to start Fail2Ban Service.
:( Best regards Peter

pehebo, 2018-03-04:
Now it works! After installing 0.46.39 and rebooting the fail2ban service ist started, as it should be...
Could you please tell me how you solved this problem...Removed ramlogs?
Thany you very much for your great enthusiasm!!!
Best regards Peter

pehebo, 2018-03-04:
Sorry to say that I found another sideeffect:
Now the Nextcloud App (MacOS) shows a gray sign meaning "no connection", although this is not true.

pehebo, 2018-03-04:
After switching off modsecuity, apche2 still failes:
Cannot access directory '/var/log/apache2/' for main error

pehebo, 2018-03-04:
is it possible to switch back to non-developer version?

nachoparker, 2018-03-04:
> Now it works! After installing 0.46.39 and rebooting the fail2ban service ist started, as it should be...
Could you please tell me how you solved this problem...Removed ramlogs?

great! I adjusted the init dependencies of fail2ban and automount.

> is it possible to switch back to non-developer version?

yes, just update normally ( `sudo ncp-update` )

> After switching off modsecuity, apche2 still failes:
Cannot access directory '/var/log/apache2/' for main error

are you in `v0.46.39`? if not, update normally and reboot. 

If it still doesn't work, give us the output of

```
cat /etc/fstab
```
and

```
mount
```

pehebo, 2018-03-05:
As you told me, I updated through sudo ncp-update and rebooted. Version stays at 046.39. Now everything works (fail2ban, apache2). Thank you very much for caring with my problems so much!!!
Best regards,
Peter

nachoparker, 2018-03-05:
great!

I think you helped us more than we helped you ;)

Thanks for reporting and testing and your patience