nextcloud/nextcloudpi #431: Fail2Ban not banning Trusted Domain Errors

pehebo, 2018-03-01:
Hello again,
I have another question: my nextcloud gets lots of access attempts (most likely from bots) that are blocked as "Trusted domain errors". Nevertheless the server is heavily loaded by these attempts. None of these IPs are banned by fail2ban. My question: doesn't fail2ban ban them, as it only looks for "Login errors"? Or is the configuration of fail2ban insufficient?
Best regards,
Peter

nachoparker, 2018-03-03:
Hi,

You are right, only logging attempts are blocked. Actually it could be a good idea to also block those, but taking into account that fail2ban only bans for some minutes, I do not think it would be very useful.

The situation you are seeing will be improved if #408 gets implemented

In the future, please use the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) for questions and technical discussion.

Thanks for reporting!

closed 2018-03-03 by nachoparker

Haraade, 2018-08-18:
Hello. 
Seeing some connection attempts against server.
Has therefore modified fail2ban.

/etc/fail2ban/jail.local

[nextcloud2] 
enabled = true 
port = http,https 
filter = nextcloud2 
logpath = /data/nextcloud/nextcloud.log
maxretry = 1 
findtime = 1 
bantime = 604800 ; 1 week 

[ufwban]
enabled = true
port = ssh, http, https
filter = ufwban
logpath = /var/log/ufw.log
maxretry = 1
findtime = 1
bantime = 604800
action = ufw

and

/etc/fail2ban/filter.d/nextcloud2.conf

[INCLUDES] 
before = common.conf 

[Definition] 
failregex = ^.*\"remoteAddr\":\"<HOST>\".*Trusted domain error.*$
ignoreregex = 

and

/etc/fail2ban/filter.d/ufwban.conf

[Definition] 
failregex = UFW BLOCK.* SRC=<HOST> ignoreregex =