nextcloud/nextcloudpi #433: Initial installation for raspberry PI3 failed with  etc/nextcloudpi-config.d/nc-nextcloud.sh: line 89: /usr/lib/systemd/system/nc-provisioning.service: No such file or directory

asv285041, 2018-03-01:
Hey,

i have rasPI3 and latest image Raspi lite 
when i start script i faced with error below

Synchronizing state of redis-server.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable redis-server
./etc/nextcloudpi-config.d/nc-nextcloud.sh: line 89: /usr/lib/systemd/system/nc-provisioning.service: No such file or directory

Checked folders and there is no such folder /usr/lib/systemd/system

So error appears and cannot continue installation. 
if you create this folder - installation can be started again and continue

wouterbr, 2018-03-02:
Hi theresame here;
Executing: /lib/systemd/systemd-sysv-install enable redis-server
./etc/nextcloudpi-config.d/nc-nextcloud.sh: line 89: /usr/lib/systemd/system/nc-provisioning.service: No such file or directory

fspin, 2018-03-02:
this is a quickfix: 

in my case i had to create the folder `system`
`mkdir /usr/lib/systemd/system`

and touch the file:
`touch /usr/lib/systemd/system/nc-provisioning.service`.

after that, you'll be able to run the script again.

nachoparker, 2018-03-03:
Fixed in `v0.46.36`.

Thanks for reporting

closed 2018-03-03 by nachoparker