nextcloud/nextcloudpi #434: to much privileges for user ncp

yo-k, 2018-03-03:
As mentioned in #430 from nachoparker:

> If you enable SSH through ncp-web or ncp-config, then it forces you to change this password. Also, this user has no privileges whatsoever.

 * I was not forced to change the password. Logged in with default pw ownyourbits.
 * ncp has got a full shell /bin/sh
 * ncp is a normal user (UID>=1000)
 * wrong entry in /etc/passwd (/home/ncp does not exist)
 * may be missing entry /bin/false or /usr/sbin/nologin

```
$ whoami
ncp
$ grep ncp /etc/passwd
ncp:x:1001:1001::/home/ncp:
$ echo $SHELL
/bin/sh
$ 
```

I do not know the deeper meaning of user ncp. Would it be better to change ncp from normal user to system user? Should user ncp really got a login shell? For what is user ncp good for?

nachoparker, 2018-03-04:
did you enable SSH through ncp-web or ncp-config? because it is not supposed to allow you to do so until you change the _ncp_ password.

The _ncp_ user is only used for accessing ncp-web, so it makes sense to disable it's login shell.

Thanks for the feedback.

nachoparker, 2018-03-04:
Implemented `v0.46.38`

closed 2018-03-04 by nachoparker

yo-k, 2018-03-04:
I have disabled ncp-web and doing all the stuff via ncp-config. I have changed "the" ncp password in nextcloud itself but login password is still default password ownyourbits. Sure it is.

Should I change a ncp password anywhere else, too? Give me a hint. 

IMHO this is a bug, a security hole, not an enhancement. A normal user will not log in as ncp to change the password. A normal user do not know anything about a user ncp with a full login shell. So,  activating ssh could result in full featured login shell for free for anybody on a nextcloudpi machine.

nachoparker, 2018-03-04:
> I have disabled ncp-web and doing all the stuff via ncp-config. I have changed "the" ncp password in nextcloud itself but login password is still default password ownyourbits. Sure it is.

Weird... I test this step everytime I test an image, so very often and it works. Are you saying that doing `nc-passwd` doesn't change the _ncp_ password? You can look at the [code](https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-passwd.sh#L23), it's really simple.

I just tested that today.

> IMHO this is a bug, a security hole, not an enhancement. A normal user will not log in as ncp to change the password. A normal user do not know anything about a user ncp with a full login shell. 

I think you are not getting this right. _ncp_ user is not meant for logging in via shell, it is only used to provide the password mechanism for ncp-web.

It is not intended that it has shell access, and therefore it no longer has ( it is set to `nologin` according to your suggestion ).

The intended way to change the password for _ncp_ user ( this is, for ncp-web ) is to use the ncp-app `nc-passwd`

> So, activating ssh could result in full featured login shell for free for anybody on a nextcloudpi machine.

Again, this has been already prevented by setting `nologin` for _ncp_. Also, activating `SSH` from ncp-config or ncp-web used to refuse to work if you didn't first change the _ncp_ password. Because there is no longer the possibility of getting a loging shell for this user I have now removed this check.

Hope it makes sense now

nachoparker, 2018-03-04:
> I have changed "the" ncp password in nextcloud itself

Maybe you were thinking that by changing the admin password of Nextcloud you would also change the password for ncp-web and the _ncp_ user?

yo-k, 2018-03-04:
Maybe, I can't remember each step in detail. Maybe I missed some steps or did it wrong. I was not sure about the necessary steps and tried to login as ncp. That worked :-(  Anyhow, I ended up with a functional ncp login shell with default password. That is a security hole - or better that WAS a security hole. Thanks for changing the code and disable login. I think it's better to disable a login if not needed.

yo-k, 2018-03-04:
One more question. Could I remove the user ncp? Will the update scripts still work or will they fail?

nachoparker, 2018-03-05:
> One more question. Could I remove the user ncp? Will the update scripts still work or will they fail?

you can. The only thing is that you won't be able to use ncp-web