nextcloud/nextcloudpi #437: Docker ownyourbits/nextcloudpi-armhf with custom data directory

ivanmarban, 2018-03-08:
I'm running some problems while trying to customize the ncdata volume.

I've created a custom ncdata volume before running nextcloudpi container:

``docker volume create --opt type=none --opt device=/media/usbstorage/nextcloudpi --opt o=bind ncdata``

I'm getting these errors:

``Making /usr/local/etc/nextcloudpi-config.d persistent ...
Making /etc/services-enabled.d persistent ...
/run-parts.sh: line 43: /etc/services-enabled.d/010lamp: Permission denied
/run-parts.sh: line 43: /etc/services-enabled.d/020nextcloud: Permission denied
Init done
``

nemo19, 2018-03-13:
I tried the same thing using pretty much the same command to create the volume:
```
docker volume create -d local -o type=none -o
 o=bind -o device=/path/to/dir ncdata
```

For me it's working with the nextcloudpi-armhf image.

ivanmarban, 2018-03-13:
I'm running ```Docker version 18.02.0-ce, build fc4de447b5``` on Arch Linux ARM ```Linux raspberrypi2 4.14.24-1-ARCH #1 SMP Mon Mar 5 20:00:46 UTC 2018 armv7l GNU/Linux```

Which versions are yours ?

nemo19, 2018-03-13:
`Docker version 17.12.1-ce` on Armbian `Linux helios4 4.14.18-mvebu #22 SMP Fri Feb 9 10:41:38 CET 2018 armv7l GNU/Linux`

nachoparker, 2018-03-14:
The way I did it is moving all the docker containers to the external HDD. Like this

https://ownyourbits.com/2017/06/08/nextcloudpi-docker-for-raspberry-pi/

ivanmarban, 2018-03-14:
Seems to be a valid workaround. Even so I would like to understand if those errors are docker or nextcloudpi-armhf related.

ivanmarban, 2018-03-14:
I've made some research and I found that those errors were generated by "noexec" mount option of the external HDD. After changing it, everything runs normal.

nachoparker, 2018-03-18:
good, I'll write a note about reviewing better ways in #386

closed 2018-03-18 by nachoparker

crouthamela, 2018-03-31:
@ivanmarban Thank you for the write-up, it helped me get my data dir moved to my NFS-mounted NAS.

crouthamela, 2018-04-10:
Just a follow-up, I have issues with my database when running over NFS to my NAS this way. I instead mounted just the user data directory, leaving the database and associated files on a local volume on the SD card. This appears to be the ideal setup, and is very fast. Here is my run command for the next person reading this:

`docker run -d --network=host --restart unless-stopped -v nextcloudpidata:/data --mount type=bind,source=/mnt/users/data,target=/data/app/data --name nextcloudpi ownyourbits/nextcloudpi-armhf:march mydomain.com`

Note, /mnt/users is my NFS mount, mounted via /etc/fstab to the host.

nachoparker, 2018-04-11:
cool, thanks for sharing

I also have good news. I spent a few days working on the docker container and I just pushed the latest version.

- We can now mount any external directory ( not necessarily a named volume )
- We now have internal updates, just like the SD card version
- Added some missing apps from the SD card version
- Other smaller improvements

crouthamela, 2018-04-11:
Could you elaborate on #1? I have some other NAS shares I mounted on the host, then added via the External Storage app as a local storage mount. This seems to work ok for now. How does this other mount method work?

nachoparker, 2018-04-11:
I'll detail it in the release announcement at ownyourbits.com, but basically if you run `docker run` pass `-v /myfolder:/data`, or in the docker-composefile do `- /myfolder:/data`

forl NC, it won't be external data, but the regular data at the "main location"