nextcloud/nextcloudpi #438: letsencrypt-auto giving error: Had a problem installing Python packages

FlexMcMurphy, 2018-03-08:
Hello,

I have had NextCloudPi running on my Raspberry Pi 3 for about six months now. I can access the web interface remotely with port forwarding of 80 and 443 configured from my router.

Recently I can no longer get to the web interface remotely via HTTPS.... Chrome started giving the Certificate out of date error page and not letting me go any further. I added a rule to Chrome browser to force it to use HTTP.

Basically the problem I think is because I need to run letsencrypt to refresh my SSL certificate? However when I go into nextcloudpi-config in putty and run the letsencrypt tool I get this message:

```
Launching letsencrypt
Bootstrapping dependencies for Debian-based OSes... (you can skip this with --no-bootstrap)
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
augeas-lenses is already the newest version (1.8.0-1+deb9u1).
ca-certificates is already the newest version (20161130+nmu1).
gcc is already the newest version (4:6.3.0-4).
libaugeas0 is already the newest version (1.8.0-1+deb9u1).
libffi-dev is already the newest version (3.2.1-6).
libssl-dev is already the newest version (1.1.0f-3+deb9u1).
openssl is already the newest version (1.1.0f-3+deb9u1).
python is already the newest version (2.7.13-2).
python-dev is already the newest version (2.7.13-2).
python-virtualenv is already the newest version (15.1.0+ds-1).
virtualenv is already the newest version (15.1.0+ds-1).
0 upgraded, 0 newly installed, 0 to remove and 41 not upgraded.
Creating virtual environment...
Installing Python packages...
Had a problem while installing Python packages.

pip prints the following errors:
=====================================================
Collecting argparse==1.4.0 (from -r /tmp/tmp.mizglPW5vi/letsencrypt-auto-requirements.txt (line 11))
Exception:
Traceback (most recent call last):
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/commands/install.py", line 342, in run
    requirement_set.prepare_files(finder)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py", line 795, in get_page
    resp.raise_for_status()
  File "/opt/eff.org/certbot/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/models.py", line 893, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
HTTPError: 404 Client Error: Not Found for url: https://www.piwheels.hostedpi.com/simple/zope.components/argparse/
=====================================================

Certbot has problem setting up the virtual environment.

We were not be able to guess the right solution from your pip
output.

Consult https://certbot.eff.org/docs/install.html#problems-with-python-virtual-environment
for possible solutions.
[... 149 lines cut ...]
{"reqId":"Wp-rgn8AAQEAABFcOlIAAAAW","level":2,"time":"2018-03-07T13:39:14+00:00","remoteAddr":"46.17.47.102","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"46.17.47.102\" tried to access using \"86.45.47.16\" as host.","userAgent":"libwww-perl\/6.31","version":"12.0.3.3"}
{"reqId":"Wp-3vn8AAQEAABFcOlMAAAAD","level":2,"time":"2018-03-07T14:31:26+00:00","remoteAddr":"74.82.47.2","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"74.82.47.2\" tried to access using \"86.45.47.16\" as host.","userAgent":"--","version":"12.0.3.3"}
{"reqId":"Wp-7zH8AAQEAABFdbe4AAABR","level":3,"time":"2018-03-07T14:48:44+00:00","remoteAddr":"46.161.55.108","user":"--","app":"PHP","method":"GET","url":"\/a2billing\/common\/javascript\/misc.js","message":"Exception: The requested uri(\/a2billing\/common\/javascript\/misc.js) cannot be processed by the script '\/core\/templates\/404.php') at \/var\/www\/nextcloud\/lib\/private\/AppFramework\/Http\/Request.php#729","userAgent":"python-requests\/2.18.4","version":"12.0.3.3"}
{"reqId":"Wp-9P38AAQEAABFdbe8AAABW","level":2,"time":"2018-03-07T14:54:55+00:00","remoteAddr":"164.132.91.13","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"164.132.91.13\" tried to access using \"86.45.47.16\" as host.","userAgent":"(KHTML, like Gecko) Chrome\/41.0.2227.1 Safari\/537.36","version":"12.0.3.3"}
{"reqId":"WqAePH8AAQEAABFcOlUAAAAX","level":2,"time":"2018-03-07T17:15:40+00:00","remoteAddr":"107.170.231.128","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"107.170.231.128\" tried to access using \"86.45.47.16\" as host.","userAgent":"Mozilla\/5.0 zgrab\/0.x","version":"12.0.3.3"}
{"reqId":"WqAjb38AAQEAABFcOlYAAAAE","level":3,"time":"2018-03-07T17:37:51+00:00","remoteAddr":"46.161.55.107","user":"--","app":"PHP","method":"GET","url":"\/webfig\/master-078380d0e221.css","message":"Exception: The requested uri(\/webfig\/master-078380d0e221.css) cannot be processed by the script '\/core\/templates\/404.php') at \/var\/www\/nextcloud\/lib\/private\/AppFramework\/Http\/Request.php#729","userAgent":"python-requests\/2.18.4","version":"12.0.3.3"}
{"reqId":"WqB@oH8AAQEAABFcOlkAAAAK","level":2,"time":"2018-03-08T00:06:56+00:00","remoteAddr":"62.173.139.46","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"62.173.139.46\" tried to access using \"86.45.47.16\" as host.","userAgent":"--","version":"12.0.3.3"}
{"reqId":"WqDDUH8AAQEAABFcOlwAAAAD","level":3,"time":"2018-03-08T05:00:00+00:00","remoteAddr":"46.161.55.106","user":"--","app":"PHP","method":"GET","url":"\/a2billing\/common\/javascript\/misc.js","message":"Exception: The requested uri(\/a2billing\/common\/javascript\/misc.js) cannot be processed by the script '\/core\/templates\/404.php') at \/var\/www\/nextcloud\/lib\/private\/AppFramework\/Http\/Request.php#729","userAgent":"python-requests\/2.18.4","version":"12.0.3.3"}
{"reqId":"WqDKkn8AAQEAABFdbfEAAABG","level":3,"time":"2018-03-08T05:30:58+00:00","remoteAddr":"201.6.107.126","user":"--","app":"PHP","method":"GET","url":"\/admin\/config.php","message":"Exception: The requested uri(\/admin\/config.php) cannot be processed by the script '\/core\/templates\/404.php') at \/var\/www\/nextcloud\/lib\/private\/AppFramework\/Http\/Request.php#729","userAgent":"curl\/7.15.5 (x86_64-redhat-linux-gnu) libcurl\/7.15.5 OpenSSL\/0.9.8b zlib\/1.2.3 libidn\/0.6.5","version":"12.0.3.3"}
{"reqId":"WqEJU38AAQEAACj00JAAAABT","level":3,"time":"2018-03-08T09:58:43+00:00","remoteAddr":"221.239.99.65","user":"--","app":"PHP","method":"GET","url":"\/currentsetting.htm","message":"Exception: The requested uri(\/currentsetting.htm) cannot be processed by the script '\/core\/templates\/404.php') at \/var\/www\/nextcloud\/lib\/private\/AppFramework\/Http\/Request.php#729","userAgent":"--","version":"12.0.3.3"}
{"reqId":"WqEVaX8AAQEAACjz8VQAAAAT","level":3,"time":"2018-03-08T10:50:17+00:00","remoteAddr":"46.161.55.106","user":"--","app":"PHP","method":"GET","url":"\/a2billing\/common\/javascript\/misc.js","message":"Exception: The requested uri(\/a2billing\/common\/javascript\/misc.js) cannot be processed by the script '\/core\/templates\/404.php') at \/var\/www [... cut ...] 4-redhat-linux-gnu) libcurl\/7.15.5 OpenSSL\/0.9.8b zlib\/1.2.3 libidn\/0.6.5","version":"12.0.3.3"}

```
</details>

I tested the ports 80 and 443 and they are accessible from outside my network. I didn't add or change anything on the Pi... except today I allowed NextCloudPi to update to the latest version.

For now I can connect to the nextcloudpi web interface via http but I would like to get https working again. How can I refresh my ssl cert using letsencrypt?

Thank you for your help.

Flex

nachoparker, 2018-03-10:
Hi, normally letsencrypt automatically renews, but in your case it couldn't succeed because of that error and this is the first time you notice this.

You should only have this in `/etc/pip.conf`

```
[global]
extra-index-url=https://www.piwheels.hostedpi.com/simple
```

If you have something different and correcting it still doesn't work for you, reinstall the whole thing.

```
sudo su
rm -r /etc/letsencrypt
cd /etc
git clone https://github.com/letsencrypt/letsencrypt
```
and then try again

nachoparker, 2018-03-13:
I'll close but let us know if it works. We can reopen if needed be

closed 2018-03-13 by nachoparker

FlexMcMurphy, 2018-03-14:
Hello,

Sorry for the delay, I just got to try out your solution now. It's still not working even though I tried what you suggested.

My /etc/pip.conf file was like this:

```
[global]
extra-index-url=https://www.piwheels.hostedpi.com/simple
extra-index-url=https://www.piwheels.hostedpi.com/simple/zope.components
```

I removed the second line, went back in to `nextcloudpi-config` and ran letsencrypt but got the same error...

```
Launching letsencrypt
Bootstrapping dependencies for Debian-based OSes... (you can skip this with --no-bootstrap)
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
augeas-lenses is already the newest version (1.8.0-1+deb9u1).
ca-certificates is already the newest version (20161130+nmu1).
gcc is already the newest version (4:6.3.0-4).
libaugeas0 is already the newest version (1.8.0-1+deb9u1).
libffi-dev is already the newest version (3.2.1-6).
libssl-dev is already the newest version (1.1.0f-3+deb9u1).
openssl is already the newest version (1.1.0f-3+deb9u1).
python is already the newest version (2.7.13-2).
python-dev is already the newest version (2.7.13-2).
python-virtualenv is already the newest version (15.1.0+ds-1).
virtualenv is already the newest version (15.1.0+ds-1).
0 upgraded, 0 newly installed, 0 to remove and 20 not upgraded.
Creating virtual environment...
Installing Python packages...
Had a problem while installing Python packages.

pip prints the following errors:
=====================================================
Collecting argparse==1.4.0 (from -r /tmp/tmp.791LEYxpik/letsencrypt-auto-requirements.txt (line 11))
Exception:
Traceback (most recent call last):
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/commands/install.py", line 342, in run
    requirement_set.prepare_files(finder)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py", line 795, in get_page
    resp.raise_for_status()
  File "/opt/eff.org/certbot/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/models.py", line 893, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
HTTPError: 404 Client Error: Not Found for url: https://www.piwheels.hostedpi.com/simple/zope.components/argparse/
=====================================================

Certbot has problem setting up the virtual environment.

We were not be able to guess the right solution from your pip
output.

Consult https://certbot.eff.org/docs/install.html#problems-with-python-virtual-environment
for possible solutions.
You may also find some support resources at https://certbot.eff.org/support/ .
Done. Press any key...
```
Then I did this:

```
sudo su
rm -r /etc/letsencrypt
cd /etc
git clone https://github.com/letsencrypt/letsencrypt
```
... and went back in to `nextcloudpi-config` and ran letsencrypt but got the same above error again.

Unless all the stuff I'm saving to my NCP is filling up the SD card on the PI?

```
root@nextcloudpi:/# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       2.7G  1.9G  592M  77% /
devtmpfs        460M     0  460M   0% /dev
tmpfs           464M     0  464M   0% /dev/shm
tmpfs           464M   24M  440M   6% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1   42M   21M   21M  51% /boot
/dev/sda1        29G  7.7G   22G  27% /media/4632-A3A7
tmpfs            93M     0   93M   0% /run/user/1000
root@nextcloudpi:/#
```

I have a 30GB USB attached to the Pi 3 at SDA1 but I don't think files I am putting on the NCP are saving to that USB key. Nonetheless /dev/root is only 77% used up.

I have read elsewhere this might be to do with low memory? NCP is the only thing I have on this SD card. Unless you can think of something else for me to try I might just do a fresh re-install of NCP and see if that fixes it.

Thanks, I'll let you know,

Flex

nachoparker, 2018-03-14:
try again after deleting `/etc/pip.conf` completely

your error is not related to low memory, it just can't find this address

```
https://www.piwheels.hostedpi.com/simple/zope.components/argparse/
```

FlexMcMurphy, 2018-03-15:
Hello again,

Thank you for your continued help :)

It's not working still. I tried deleting /etc/pip.conf and then ran letsencrypt again from nextcloudpi-config. this time I got a different error... sorry but I have spent several hours this evening so I can't remember the error I just know that it wasn't that Python packages error above anymore.

Now instead I get:

````
Launching letsencrypt
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Plugins selected: Authenticator webroot, Installer None
Obtaining a new certificate
Performing the following challenges:
http-01 challenge for mydomain.ddns.net
Using the webroot path /var/www/nextcloud for all unmatched domains.
Waiting for verification...
Cleaning up challenges
Failed authorization procedure. mydomain.ddns.net (http-01): urn:acme:error:connection :: The server could not connect to the client to verify the domain :: Fetching http://mydomain.ddns.net/.well-known/acme-challenge/XeI-_gQZ-RnfKzNIuw6t4JgOS4ZA0Llxc11fFZz0I4A: Connection refused
IMPORTANT NOTES:
 - The following errors were reported by the server:

   Domain: mydomain.ddns.net
   Type:   connection
   Detail: Fetching
   http://mydomain.ddns.net/.well-known/acme-challenge/XeI-_gQZ-RnfKzNIuw6t4JgOS4ZA0Llxc11fFZz0I4A:
   Connection refused

   To fix these errors, please make sure that your domain name was
   entered correctly and the DNS A/AAAA record(s) for that domain
   contain(s) the right IP address. Additionally, please check that
   your computer has a publicly routable IP address and that no
   firewalls are preventing the server from communicating with the
   client. If you're using the webroot plugin, you should also verify
   that you are serving files from the webroot path you provided.
Done. Press any key...
````
I checked to see if my port 80 and 443 were still accessible from external (ports open) but now they seem to be closed. I spent several hours trying to re-configure my router but I think the issue is not in the router because I didn't change anything there to cause port 80 and 443 to be closed all of a sudden. I think the error might be when I ran the command you gave me yesterday ....
````
sudo su
rm -r /etc/letsencrypt
cd /etc
git clone https://github.com/letsencrypt/letsencrypt
````
I think maybe the NCP webserver is not running now and that's why the ports seem to be closed... they are open but the server process is not listening.

Here is the output of sudo ncp-report which I think shows that the webserver is not running...
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.47.1
NextCloudPi image    NextCloudPi_09-28-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      2.0G/2.7G
rootfs usage         2.0G/2.7G
swapfile             /var/swap
Nextcloud check      ok
Nextcloud version    12.0.3.3
HTTPD service        down
PHP service          up
MariaDB service      up
[... 132 lines cut ...]

Mar 15 03:18:27 nextcloudpi systemd[1]: Starting The Apache HTTP Server...
Mar 15 03:18:27 nextcloudpi apachectl[12319]: AH00526: Syntax error on line 5 of /etc/apache2/sites-enabled/ncp.conf:
Mar 15 03:18:27 nextcloudpi apachectl[12319]: SSLCertificateFile: file '/etc/letsencrypt/live/mydomain.ddns.net/fullchain.pem' does not exist or is empty
Mar 15 03:18:27 nextcloudpi apachectl[12319]: Action 'start' failed.
Mar 15 03:18:27 nextcloudpi apachectl[12319]: The Apache error log may have more information.
Mar 15 03:18:27 nextcloudpi systemd[1]: apache2.service: Control process exited, code=exited status=1
Mar 15 03:18:27 nextcloudpi systemd[1]: Failed to start The Apache HTTP Server.
Mar 15 03:18:27 nextcloudpi systemd[1]: apache2.service: Unit entered failed state.
Mar 15 03:18:27 nextcloudpi systemd[1]: apache2.service: Failed with result 'exit-code'.
root@nextcloudpi:/home/pi#
````
This is line 5 of /etc/apache2/sites-enabled/ncp.conf
````
SSLCertificateFile /etc/letsencrypt/live/mydomain.ddns.net/fullchain.pem
````
... like the rest of the above error message states... the /etc/letsencrypt/live folder does not even exist. Maybe this is because letsencrypt never runs to give me a certificate? Without a certificate I can't run the webserver and without the webserver I can't run letsencrypt it seems??

Local web access to NCP is not available from my laptop either... NCP is installed on my Pi with fixed ip address 192.168.1.124... if I type https://192.168.1.124 or nextcloudpi.local I get "refused to connect" error in the browser. I get the same message if I try my dynamic domain address too.

Do you think there is a fix for this or should I just re-install all of NCP? I'm wondering will I have the same problem again in a few months time because letsencrypt won't get set up correctly after a re-install so it won't renew the certificate then it will all stop working again requiring a total re-install in a few months time again?

Thanks for your help Nacho,

Flex

macauwaterball, 2018-03-15:
I have the same problem. can't resolve it. and I have many files uploaded to my nextcloud pi. if can't fix it, it is very trouble.

nachoparker, 2018-03-15:
@FlexMcMurphy good, now letsencrypt is correctly installed!!

the problem now is that apache doesn't start. My bad, forgot to tell you to remove the lines in `/etc/apache2/sites-available.d/nextcloud.conf` that start with `SSLCertificateKeyFile` and `SSLCertificateFile`

FlexMcMurphy, 2018-03-15:
I deleted those lines from /etc/apache2/sites-available.d/nextcloud.conf as you suggested. Well the dir is called [sites-available] on my system. I then got errors about the same lines in ncp.conf so I deleted those from there as well. I rebooted the Pi and checked to see if apache was running but got this...

````
pi@nextcloudpi:~ $ systemctl status apache2.service
â apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-03-15 22:55:22 UTC; 6min ago
  Process: 634 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)

Mar 15 22:55:21 nextcloudpi systemd[1]: Starting The Apache HTTP Server...
Mar 15 22:55:22 nextcloudpi apachectl[634]: Action 'start' failed.
Mar 15 22:55:22 nextcloudpi apachectl[634]: The Apache error log may have more information.
Mar 15 22:55:22 nextcloudpi systemd[1]: apache2.service: Control process exited, code=exited status=1
Mar 15 22:55:22 nextcloudpi systemd[1]: Failed to start The Apache HTTP Server.
Mar 15 22:55:22 nextcloudpi systemd[1]: apache2.service: Unit entered failed state.
Mar 15 22:55:22 nextcloudpi systemd[1]: apache2.service: Failed with result 'exit-code'.
````
The error log in \\NEXTCLOUDPI\root\var\log\apache2 looks like this...
````
[Thu Mar 15 22:55:22.793243 2018] [ssl:emerg] [pid 650:tid 1992757248] AH02311: Fatal error initialising mod_ssl, exiting. See 

/var/www/nextcloud/data/error.log for more information
AH00016: Configuration Failed
````

The last two lines of the error log in \\NEXTCLOUDPI\root\var\www\nextcloud\data looks like this...

````
[Thu Mar 15 22:55:22.785104 2018] [ssl:emerg] [pid 650:tid 1992757248] AH02572: Failed to configure at least one certificate and key for mydomain.net:443
[Thu Mar 15 22:55:22.793182 2018] [ssl:emerg] [pid 650:tid 1992757248] SSL Library Error: error:140A80B1:SSL routines:SSL_CTX_check_private_key:no certificate assigned
````
So then I tried running Letsencrypt again from the terminal config menu and got...
````
Launching letsencrypt
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Plugins selected: Authenticator webroot, Installer None
Obtaining a new certificate
Performing the following challenges:
http-01 challenge for mydomain.net
Using the webroot path /var/www/nextcloud for all unmatched domains.
Waiting for verification...
Cleaning up challenges
Failed authorization procedure. tosho.ddns.net (http-01): urn:acme:error:connection :: The server could not connect to the client to verify the domain 

:: Fetching http://mydomain.net/.well-known/acme-challenge/4uCHWaxT36_fIx8acGK0i6koSI4bLzigVJNvKIuGQsM: Connection refused
IMPORTANT NOTES:
 - The following errors were reported by the server:

   Domain: mydomain.net
   Type:   connection
   Detail: Fetching
   http://mydomain.net/.well-known/acme-challenge/4uCHWaxT36_fIx8acGK0i6koSI4bLzigVJNvKIuGQsM:
   Connection refused

   To fix these errors, please make sure that your domain name was
   entered correctly and the DNS A/AAAA record(s) for that domain
   contain(s) the right IP address. Additionally, please check that
   your computer has a publicly routable IP address and that no
   firewalls are preventing the server from communicating with the
   client. If you're using the webroot plugin, you should also verify
   that you are serving files from the webroot path you provided.
Done. Press any key...
````
There are a few problems here I think. I happen to know that my port forwarding more than likely isn't working. My main router has port 80 and 443 locked down by my ISP. I can bridge it again with another router and open the ports on the other router... BUT anyway even if I did that it wouldn't make a difference because Apache won't start and it looks like it won't because it needs to find an SSL Cert?

I'll try to get port forwarding going again and then run LetsEncrypt once more.. maybe if it creates the SSL cert then Apache will start up.. is that the correct order?

Thank you Nacho,

Flex

FlexMcMurphy, 2018-03-16:
Just confirming that I got my router bridged to another router which does allow me to port forward 80 and 443 and I was able to confirm the ports are open, via yougetsignal.com, when I temporarily run a little simpleHTTPServer on the NCP image using ports 80 and 443.

Nevertheless I get the same errors as above with LetsEncrypt: (http-01): urn:acme:error:connection :: The server could not connect to the client to verify the domain... connection refused and with Apache: Failed to start The Apache HTTP Server. On this latest attempt I didn't get any new error messages in those two error.log files also mentioned in my last post.

I don't think I have any firewall running anywhere. upNp is enabled on both routers. I also ran nc-forward-ports by the way and it did report that it opened ports 80 and 443.

Maybe my problem is to do with Apache not running but I don't know how to start it up?

I might have to bite the bullet and just do a re-install of NCP. I've also been reading about the other ways to get SSL certificates via DNS. That looks very cool and I wouldn't have to bridge one router to another anymore as well !

Flex

nachoparker, 2018-03-16:
Yes, you need apache running ( and port forwarding ) in order to make letsencrypt work.

You are right, you need certificates, deleting is not enough. Paste this 

```
<IfModule mod_ssl.c>
  <VirtualHost _default_:443>
    DocumentRoot /var/www/nextcloud
    CustomLog /var/www/nextcloud/data/access.log combined
    ErrorLog /var/www/nextcloud/data/error.log
    SSLEngine on
    SSLCertificateFile      /etc/ssl/certs/ssl-cert-snakeoil.pem
    SSLCertificateKeyFile /etc/ssl/private/ssl-cert-snakeoil.key
  </VirtualHost>
  <Directory /var/www/nextcloud/>
    Options +FollowSymlinks
    AllowOverride All
    <IfModule mod_dav.c>
      Dav off
    </IfModule>
    LimitRequestBody 0
    SSLRenegBufferSize 10486000
  </Directory>
</IfModule>
```

for NC, and this 

```
Listen 4443
<VirtualHost _default_:4443>
  DocumentRoot /var/www/ncp-web
  SSLEngine on
  SSLCertificateFile      /etc/ssl/certs/ssl-cert-snakeoil.pem
  SSLCertificateKeyFile /etc/ssl/private/ssl-cert-snakeoil.key

  # 2 days to avoid very big backups requests to timeout
  TimeOut 172800

  <IfModule mod_authnz_external.c>
    DefineExternalAuth pwauth pipe /usr/sbin/pwauth
  </IfModule>

</VirtualHost>
<Directory /var/www/ncp-web/>

  AuthType Basic
  AuthName "ncp-web login"
  AuthBasicProvider external
  AuthExternal pwauth

  SetEnvIf Request_URI "^" noauth
  SetEnvIf Request_URI "^index\.php$" !noauth
  SetEnvIf Request_URI "^/$" !noauth
  SetEnvIf Request_URI "^/wizard/index.php$" !noauth
  SetEnvIf Request_URI "^/wizard/$" !noauth

  <RequireAll>

   <RequireAny>
      Require host localhost
      Require local
      Require ip 192.168
      Require ip 172
      Require ip 10
   </RequireAny>

   <RequireAny>
      Require env noauth
      Require user ncp
   </RequireAny>

  </RequireAll>

</Directory>
```

for ncp-web, then you should be able to `systemctl start apache2`

FlexMcMurphy, 2018-03-16:
Nacho,

Updating those VirtualHost files got apache running again and now I have managed to renew my SSL certs. Thank you for all your help and patience!

My original problem was that /etc/pip.conf file and deleting it would have been enough to solve it. However removing the /etc/letsencrypt folder stopped apache from running because the VirtualHost files: 
````
/etc/apache2/sites-available/ncp.conf
/etc/apache2/sites-available/nextcloud.conf
````
... were looking for my old SSL files in /etc/letsencrypt that were already deleted.

Maybe LetsEncrypt made a change so pip.conf is no longer required and then maybe you rolled out an update that removed it? but I never ran that update.

Well whatever the reason I don't care any more.. I'm now interested in how to renew SSL certs without using port 80 and 443... preferably by DNS without using ports at all. Also, it would be great being able to run the NCP web and management interfaces on any port I like... my VDSL router locks out 80, 443 etc so I have to bridge it to an old Billion 2.4G only router that doesn't have any port forwarding restrictions.

I'm reading these links next:
Letsencrypt without Port 80 Setup
https://github.com/nextcloud/nextcloudpi/issues/293

How to get certificate with Letsencrypt using DNS to verify domain
https://github.com/nextcloud/nextcloudpi/wiki/How-to-get-certificate-with-Letsencrypt-using-DNS-to-verify-domain

How to configure Let's Encrypt with closed ports 80 and 443
https://github.com/nextcloud/nextcloudpi/wiki/How-to-configure-Let's-Encrypt-with-closed-ports-80-and-443

Thanks for all your great work,

Flex

nachoparker, 2018-03-17:
Your assumptions are correct, basically we had to work around the following bug https://github.com/nextcloud/nextcloudpi/issues/319. You might have missed the update.

About #293 it will happen, but if nobody helps it will happen slow because 99% of the people use standard ports and there are more important things to address first.