nextcloud/nextcloudpi #439: nc-restore is not working correctly

pmjklemm, 2018-03-10:
Hey, 
 
i tried to recover a recent backup with nc-restore. The progress just stops at "backing up existing ..." this seems kinda wrong, since i want to do a restore not a backup ?  

here is the output of nc-restore:
> [ nc-restore ]
> INFO: overwriting old instance
> extracting backup file /media/DATA/nextcloud-bkp_20180221.tar...
> restore files...
> restore database...
> restore datadir to /var/www/nextcloud/data...
> backing up existing /var/www/nextcloud/data

then after ~12 hours of waiting, nothing happens :-O 

maybe you have a clue, why this is happening ?

i use a 128 gb sd card with all the nextcloudpi stuff and the data. I have a 1 TB hard drive with the backup connected via usb. The whole nextcloudpi instance was already running fine on this setup but now i wanted to test the restore function. I already freshly reinstalled nextcloudpi on the sd card (full clear) and still the same error.

thanks again for this amazing project, 
pmjk

nachoparker, 2018-03-10:
Hello,

How much data are we talking about? are you using BTRFS? you can run `top` or `ps aux` to see what step are we stuck at.

My first guess is that it is moving a huge amount of data around ( backing up your old data )

pmjklemm, 2018-03-10:
>How much data are we talking about?

~90gb

>are you using BTRFS?

no i am not using btrfs. I used the normal nc-backup function.

>you can run top or ps aux to see what step are we stuck at.

`27191 root       20   0  3084  1644  1484 D  0.0  0.2  1:31.13 mv /media/DATA/nextcloud-bkp_20180221.tar-tmp/data/admin /media/DATA/nextcloud-bkp_20180221.tar-`

but the sd card is empty, so this

`mv "$DATADIR" "$DATADIR-$( date "+%m-%d-%y" )" || return 1`

should be done instantly or not ?

pmjklemm, 2018-03-11:
Ahhh ok my fault, it just took ~15 hours to complete :D 

so everything is fine

closed 2018-03-11 by pmjklemm