nextcloud/nextcloudpi #440: Davdroid is not syncing with NC13

Ricardosgeral, 2018-03-10:
Hi

I've tried to look other places with no luck so I'm trying to inquire here if anyone using nextcloud pi is having trouble syncing calendar and contacts with android using davdroid 

Is there any configuration in recent nc-update that can mess with webdav? 

Every thing was working fine until a few days. I don't know if it was because an update 
Thanks

Paste the output of `sudo ncp-report`

nachoparker, 2018-03-10:
not that I can think about. Does the NC web interface work? output of `sudo ncp-report`?

hurrikan77, 2018-03-10:
I am using both nc13 and davdroid and both works without any problem.

Ricardosgeral, 2018-03-11:
OK. Must be something else

nachoparker, 2018-03-12:
still not working?

I recommend that you ask general questions in the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) because it is a more appropriate place and also you will reach more people there

closed 2018-03-12 by nachoparker

Ricardosgeral, 2018-03-13:
I hard reset my phone and now it works
Should have been something with the phone settings. Not now what.
I don't think it has anything to do with nextcloudpi updates.
you can close this issue
thanks for the concern