nextcloud/nextcloudpi #441: "Access though untrusted domain" configuration trouble

Amolith, 2018-03-11:
# PROBLEM

I have spent the entire day setting up NCPi and I finally get most things working except for my domain name. I am using the domain name ********.ddns.net and when I got to that URL, NC says that the URL isn't trusted. Clicking the button takes me to an invalid address. I have looked around for the past few hours for `config/config.php` that the error mentions: `edit the "trusted_domains" setting in config/config.php like the example in config.sample.php.` and I cannot find it anywhere.

# Output of `sudo ncp-report`
<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.47.1
NextCloudPi image    NextCloudPi_02-06-18
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      1.9G/15G
rootfs usage         1.9G/15G
swapfile             /var/swap
Nextcloud check      ok
Nextcloud version    13.0.0.14
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.7
gateway              192.168.1.1
interface            eth0
certificates         none
certs due            none
NAT loopback         no
uptime               1:05

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "192.168.1.7"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbtype": "mysql",
        "version": "13.0.0.14",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
[... 79 lines cut ...]
```

{"reqId":"xqd7cOWmjufe4qHr6orB","level":4,"time":"2018-02-06T15:15:11+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"ocs2vLU1T0fJHiKfyzDd","level":4,"time":"2018-02-06T15:30:15+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"YHLyROIq6wZypYxjIp5B","level":4,"time":"2018-02-06T15:45:22+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"DWZHLutXQ63TJ5CcLTzk","level":4,"time":"2018-02-06T16:00:19+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"rV8HXYLvJWl4tVSOXsYe","level":4,"time":"2018-02-06T16:15:28+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"8MvQecXiEefmCKqxLtnX","level":3,"time":"2018-02-06T16:15:45+00:00","remoteAddr":"","user":"--","app":"PHP","method":"--","url":"--","message":"fileowner(): stat failed for \/var\/www\/nextcloud\/config\/config.php at \/var\/www\/nextcloud\/console.php#65","userAgent":"--","version":""}
{"reqId":"WqRDAH8AAQEAAAhOKu0AAAEY","level":2,"time":"2018-03-10T20:41:37+00:00","remoteAddr":"192.168.1.2","user":"--","app":"core","method":"POST","url":"\/index.php\/login","message":"Login failed: 'amolithseregion@protonmail.com' (Remote IP: '192.168.1.2')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:58.0) Gecko\/20100101 Firefox\/58.0","version":"13.0.0.14"}
{"reqId":"WqRDEH8AAQEAAAhOKu8AAAIB","level":2,"time":"2018-03-10T20:41:52+00:00","remoteAddr":"192.168.1.2","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=amolithseregion%40protonmail.com","message":"Login failed: 'pi' (Remote IP: '192.168.1.2')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:58.0) Gecko\/20100101 Firefox\/58.0","version":"13.0.0.14"}
{"reqId":"WqRDGn8AAQEAAAhOKvEAAAMD","level":2,"time":"2018-03-10T20:42:02+00:00","remoteAddr":"192.168.1.2","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=pi","message":"Login failed: 'pi' (Remote IP: '192.168.1.2')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:58.0) Gecko\/20100101 Firefox\/58.0","version":"13.0.0.14"}
{"reqId":"WqRDhn8AAQEAAAhPLdcAAEMN","level":2,"time":"2018-03-10T20:43:51+00:00","remoteAddr":"192.168.1.2","user":"--","app":"core","method":"POST","url":"\/index.php\/login","message":"Login failed: 'admin' (Remote IP: '192.168.1.2')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:58.0) Gecko\/20100101 Firefox\/58.0","version":"13.0.0.14"}
{"reqId":"WqRDj38AAQEAAAhPLdkAAEQD","level":2,"time":"2018-03-10T20:44:01+00:00","remoteAddr":"192.168.1.2","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=admin","message":"Login failed: 'admin' (Remote IP: '192.168.1.2')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:58.0) Gecko\/20100101 Firefox\/58.0","version":"13.0.0.14"}
{"reqId":"WqRHyH8AAQEAAAhPLk8AAFQK","level":2,"time":"2018-03-10T21:02:01+00:00","remoteAddr":"192.168.1.2","user":"ncp","app":"core","method":"GET","url":"\/index.php\/settings\/admin","message":"Login failed: 'ncp' (Remote IP: '192.168.1.2')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:58.0) Gecko\/20100101 Firefox\/58.0","version":"13.0.0.14"}
{"reqId":"WqRfHH8AAQEAAAhPLtAAAE8K","level":2,"time":"2018-03-10T22:41:32+00:00","remoteAddr":"24.158.39.215","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"24.158.39.215\" tried to access using \"amolithnextcloudpi.ddns.net\" as host.","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:58.0) Gecko\/20100101 Firefox\/58.0","version":"13.0.0.14"}
{"reqId":"WqRfHn8AAQEAAAhPLtYAAFAP","level":2,"time":"2018-03-10T22:41:34+00:00","remoteAddr":"24.158.39.215","user":"--","app":"core","method":"GET","url":"\/index.php\/js\/core\/merged-template-prepend.js?v=83945674-0","message":"Trusted domain error. \"24.158.39.215\" tried to access using \"amolithnextcloudpi.ddns.net\" as host.","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:58.0) Gecko\/20100101 Firefox\/58.0","version":"13.0.0.14"}
{"reqId":"WqRgJn8AAQEAAAhPLv0AAEAS","level":2,"time":"2018-03-10T22:45:59+00:00","remoteAddr":"24.158.39.215","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"24.158.39.215\" tried to access using \"amolithnextcloudpi.ddns.net\" as host.","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:58.0) Gecko\/20100101 Firefox\/58.0","version":"13.0.0.14"}
{"reqId":"WqRgKH8AAQEAAAhPLv4AAEEP","level":2,"time":"2018-03-10T22:46:00+00:00","remoteAddr":"24.158.39.215","user":"--","app":"core","method":"GET","url":"\/index.php\/js\/core\/merged-template-prepend.js?v=83945674-0","message":"Trusted domain error. \"24.158.39.215\" tried to access using \"amolithnextcloudpi.ddns.net\" as host.","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:58.0) Gecko\/20100101 Firefox\/58.0","version":"13.0.0.14"}
{"reqId":"WqSt3H8AAQEAAANQO-QAAAAD","level":2,"time":"2018-03-11T04:17:32+00:00","remoteAddr":"24.158.39.215","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"24.158.39.215\" tried to access using \"amolithnextcloudpi.ddns.net\" as host.","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:58.0) Gecko\/20100101 Firefox\/58.0","version":"13.0.0.14"}
{"reqId":"WqSyEX8AAQEAAANQO-UAAAEB","level":2,"time":"2018-03-11T04:35:29+00:00","remoteAddr":"24.158.39.215","user":"--","app":"core","method":"GET","url":"\/index.php\/js\/core\/merged-template-prepend.js?v=83945674-0","message":"Trusted domain error. \"24.158.39.215\" tried to access using \"amolithnextcloudpi.ddns.net\" as host.","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:58.0) Gecko\/20100101 Firefox\/58.0","version":"13.0.0.14"}
[... cut ...] 1 Firefox\/58.0","version":"13.0.0.14"}
{"reqId":"WqTAq38AAQEAAANQPCYAABcK","level":2,"time":"2018-03-11T05:37:47+00:00","remoteAddr":"24.158.39.215","user":"--","app":"core","method":"GET","url":"\/index.php\/js\/core\/merged-template-prepend.js?v=83945674-0","message":"Trusted domain error. \"24.158.39.215\" tried to access using \"amolithnextcloudpi.ddns.net\" as host.","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:58.0) Gecko\/20100101 Firefox\/58.0","version":"13.0.0.14"}

```
</details>

ovpc, 2018-03-11:
Hi there,

Edit your config file were it is set to

 "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "192.168.1.7"
]

To look like 

 "trusted_domains": {
            "0": "yoursubdomain.ddns.net",
            "5": "nextcloudpi.local",
            "1": "192.168.1.7"
 }

To edit I use the command
sudo nano  /var/www/nextcloud/config/config.php

closed 2018-03-11 by nachoparker

ro-76, 2018-03-26:
Apologies for hijacking this thread, but I'm having the same issue, but with a small difference.

I'm trying to access my home NCP from a work LAN. If I use the DDNS option, it's blocked by a webfilter. If I use the actual IP of my home network, I get blocked by the untrusted domain error.  I can ping the IP no problem. 

If I add my home public IP to the config, can I access it by typing it into a browser? Can I add another line ? what number should I start with?

Obviously, I understand that the IP may change over time. 

Thx!