nextcloud/nextcloudpi #444: using x86 docker image, got "Error MySQL/MariaDB enter the database username and name."

carlhung, 2018-03-15:
I followed the link:
https://ownyourbits.com/2017/06/08/nextcloudpi-docker-for-raspberry-pi/
to install step by step.

I also move the root dic of the docker that also following the link to move the root dic to external HDD.

I can successfully see the site, I typed the username and password, then, I got this error on top of username field and below the Nextcloud's logo:
username field: admin or ncp
password: ownyourbits
Error MySQL/MariaDB enter the database username and name.

nachoparker, 2018-03-18:
you need to wait until you see 'nc init done' in the logs the first time

closed 2018-03-18 by nachoparker

carlhung, 2018-03-18:
I can't even login. where to find "nc init done"?
thanks