nextcloud/nextcloudpi #448: Which passwords need changing?

kevquirk, 2018-03-21:
Hey,

I'm going through the process of setting NCP up on my Pi 3+, however I'm not clear on what I need to do in order to harden my instance. This will be a pubicly available, production instance, so I don't want **any** default passwords on there.

I can see from the FAQ that the following exists on the system:

- For ssh or terminal pi/raspberry (root/1234 on armbian)
- For ssh or terminal ncp/ownyourbits (you have to change the password manually! See #434 #430)
- For nextcloudpi.local:4443 ncp/ownyourbits,
- For nextcloudpi.local ncp/ownyourbits
- For Database user is ncadmin/password is generated dynamically and stored in /root/.my.cnf

What's not clear to me, is what impact changing passwords will have? SSH account, no problem, I got that and will change those. But what about the others?

If I delete the ncp user, and create a new admin user with my own password and 2FA, will I still be able to access the NCP WebUI with that account?

Also, the FAQ says that the ncadmin password is generated dynamically. I assume this is a secure, long password that is generated? If not, can this also be changed to further harden the server?

Thanks for clarifying these point. I'd be happy to update the wiki for you once I have clarity around this.

theCalcaholic, 2018-03-21:
Don't delete any users (that would mess up ownership and programs acting as these users). Instead, just change the passwords.

The ncp user is only used for accessing the admin webui. You can disable the latter if your don't feel comfortable with it, but you can't use 2fa for it (that's currently not supported by the login mechanism).

Passwords for all nextcloud users (including ncadmin) can be changed in the nextcloud profile settings. You can also enable 2fa in nextcloud.

For ssh, you can disable passwords authentication completely and use ssh keys instead - however you would have to do that from the terminal as there is no setting for it in ncp web.

On an additional note: ncp web is only accessible from the local network, so you're relatively safe against remote attackers. You should also enable fail2ban (from within ncp web or ncp-config) to improve security by a huge shot.

nachoparker, 2018-03-21:
Hello,

Also it can be pointed out that SSH is disabled by default. If you enable it through ncp-web it will force you to change the password from the default.

Also, the `ncp` user is only used for login to ncp-web which, as @theCalcaholic said is local network only. This user has no shell access through SSH even if the attacker knew the password.

You can get rid of the `ncp` user, but it is probably better to just disable ncp-web if you feel better by doing so. You can do that with the `nc-webui` option.

You can find more details in ownyourbits, posts like [this one](https://ownyourbits.com/2018/02/06/nextcloudpi-updated-to-nc13-brings-ufw-security-improvements-auto-snapshots-and-more/) or [this one](https://ownyourbits.com/2018/02/06/nextcloudpi-updated-to-nc13-brings-ufw-security-improvements-auto-snapshots-and-more/). 

The passwords that you mention are randomized for every user, as per [this code](https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp-provisioning.sh#L31), so they are randomized 32 byte long passwords.

All this means that the default passwords that need to be changed are NC admin user, and ncp-web user. The `pi` user password will be forced to be changed upon activation of SSH. In the future those two passwords will also be randomized, that is the plan, but as of today they have to be changed manually.

The system has sensible defaults for security, and if you go through the wizard, fail2ban and such will also be activated for you. If you have further suggestions to improve security, those are always welcome.

I hope it is clear for you now. If you find anything that is not well explained or that can be expanded in the wiki feel free to improve it! this is a colaborative effort ;)

Finally, it is more appropriate to ask questions in the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm)

closed 2018-03-30 by nachoparker