nextcloud/nextcloudpi #449: Locked Files

Robekay, 2018-03-21:
Hey,

I am quiet new here so I hope it is alright that I publish my issue here.

I have installed NextcloudPi a couple of days ago on my Raspberry Pi 3B.

At one stage I was uploading a folder with many small files.
Everything has worked fine so far, until I my Nextcloud Desktop Sync Client reportet an Error saying:

"Error transferring https ..... - server replied: Locked"

I tried in every way to delete these files, but without any success ;/

I already looked that issue up but I am not sure how to deal with it.
I have read that I could disable filelocking in the config.php of redis, but I don´t really know how to get there on my raspi :/

It would be really nice if someone could help out ;)

Robekay, 2018-03-22:
Alright, so there is apperently a cronjob, which deletes all the locks after a certain amount of time.
I was able to delete the file the next day.

Thanks anyway  ;)

nachoparker, 2018-03-22:
Good that you solved it

Feel free to write on the wiki FAQ, this is a collaborative effort ;)

Also, the best place to ask this kind of questions/help is the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm)

closed 2018-03-22 by nachoparker