nextcloud/nextcloudpi #450: NCP user not existing ?

Dolordo, 2018-03-24:
1) Seeing update request
NextCloudPi v0.52.1 is outdated
update to v0.53.5 through 'nextcloudpi-config' or type 'sudo ncp-update'
Last login: Fri Mar 16 19:37:38 2018 from 192.168.1.12

2) upgrade doesn't work
pi@yolo:~ $ sudo ncp-update
Downloading updates
Performing updates
chsh: user 'ncp' does not exist
Removed /etc/systemd/system/sysinit.target.wants/log2ram.service.
NextCloudPi updated to version v0.53.5

3) first trial for ncp-report doesn't worked
pi@yolo:~ $ sudo ncp-report
[forever]
^C
4) reboot
5) v0.53.5 is up to date...
6) Retried some ncp-report, still no output, etc.
7) NextCloud admin interface shows 12.0.2 version
8) Tried through web interface to upgrade, same error ncp not existing.
9) Tried to see if any connection between web interface and ncp user existence, web interface was activated, removed (without validation) and added again, trying to apply change, error was '[ nc-webui ]
ERROR: Site ncp not properly enabled: /etc/apache2/sites-enabled/ncp.conf is a real file, not touching it
ncp-web enabled'
10) Stuck

So my question
=> What's the best option to fix & upgrade my system ? New fresh install or any guidance to fix these issues ?
Additionally you may probably check that user ncp is existing before trying to chsh, or something like that?

Thanks in advance !

Dolordo, 2018-03-24:
From point 7, I've realized that I need to update manually, which is done now, Nextcloud 13.0.1 is working. Still same issue with user 'ncp'

nachoparker, 2018-03-27:
`/etc/apache2/sites-enabled/ncp.conf` should be a link, not a file. Did you play with this file?

You probably have a very old image and have not updated frequently if you don't have the ncp user. I suggest that you backup your instance and restore in the latest image.

closed 2018-03-27 by nachoparker