nextcloud/nextcloudpi #451: Getting wildcard SSL Certs in NextCloudPi

FlexMcMurphy, 2018-03-24:
Hello,

Are you going to update the Certbot client in NCP to the latest version?

I'm running NextCloudPi on my Pi3 and I'd like to get wildcard SSL certs for my web domain. The version of Certbot I have now is v0.10.2 but apparently you need certbot version 0.22.0 for wildcard certs to work with LetsEncrypts ACMEv2.

I only installed certbot in the last week so I guess the latest version isn't in apt-get under Raspbian..

Alternatively could you give the steps how to upgrade to the latest Certbot on a Pi? Do you have to clone it from GitHub?

Cheers,

Flex

nachoparker, 2018-03-26:
Hi,

If I am not mistaken, letsencrypt-auto automagically updates itself every now and then. If not you can always do it manually.

NCP does not install letsencrypt through _apt_, but by cloning it, so if you go to `/etc/letsencrypt` that is already a git repository that you can update with `git pull`.

Let us know if that worked for you.

Finally, it is more appropriate to ask this kind of questions in the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) and leave github for bugs and feature requests ;)

nachoparker, 2018-04-02:
(close for cleanup, but we can still talk)

closed 2018-04-02 by nachoparker