nextcloud/nextcloudpi #453: can't access setting panel using nextCloudPi x86 docker

carlhung, 2018-03-25:
My internal IP is 192.168.0.111
docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v ncdata:/data --name nextcloudpi ownyourbits/nextcloudpi-x86 192.168.0.111
after seeing ncp init done
I can access http://192.168.0.111:80
but I can't access, https://192.168.0.111:4443 , properly.
It showed up a box that I had to fill in login and password which are "ncp" and "ownyourbits"
then, I pressed "OK". the box popped up again and asked for the login and password.

nachoparker, 2018-03-26:
I tested that before pushing the docker image and it worked, but I ran the container with docker-compose.

Tomorrow I'll check again using `docker run`

Thanks for your feedback

nachoparker, 2018-03-27:
I just tested this using your command with the latest docker image, and it logs in without problem with ncp/ownyourbits ( using Firefox at least )

closed 2018-03-27 by nachoparker