nextcloud/nextcloudpi #454: Permissions error

Qballl, 2018-03-25:
Im having issues installing nextcloud on raspberry pi zero I orignially had the permissions for my data directoryu as 777 but told me to change it to 0770 so I did and now I get this error from the web page 
`Your data directory is invalid

Ensure there is a file called ".ocdata" in the root of the data directory.

Cannot create "data" directory

This can usually be fixed by giving the webserver write access to the root directory. `
 I cannot paste the output of sudo ncp-report as thats not a command for me

nachoparker, 2018-03-26:
If you don't have the `ncp-report` you both have a very old image and didn't upgrade it often enough.

I suggest you install the latest image, and go through the web wizard, or at least enable automatic updates in ncp-web or ncp-config.

About your current problem, did you move the datadir to USB? you can check the location in the file `/var/www/nextcloud/config/config.php`. Does that directory exist? what permissions does it have? is it empty?

Qballl, 2018-03-26:
Im using version 13 point something I used this tutorial http://www.instructables.com/id/NextCloud-on-the-Raspberry-Pi-DIY-Dropbox/

Qballl, 2018-03-27:
I realized that I installed the server on my pi and not this version so I'll close this

closed 2018-03-27 by Qballl