nextcloud/nextcloudpi #456: Not able to format Drives in NCP

deljones, 2018-03-26:
I have come across an issue regarding formatting drives.

So I have i USB drive called myCloudDrive which as my NCP, database and ncdata.. All good.
I switch off the pi unplug the drive and plug in a new blank USB drive (1TB) which will be my back up. I will call it ncpbackup. When I start the format I fill in the name of the drive and hit return but get back the following message:

ERROR: Counted 5 devices. Please, only plug in the USB drive you want to format.

It is the only USB drive plugged in. When I unplug it the Counted devices drops to 4.

The only storage drives plugged in to the pi is a fresh SD card with ncpi and a blank USB drive. Have I missed a step somewhere? 

Thanks
DJ

nachoparker, 2018-03-29:
what is the output of

```
lsblk -l -n
```

nachoparker, 2018-04-03:
I'll close because of cleanup, you can continue when you have the time

closed 2018-04-03 by nachoparker

nachoparker, 2018-04-09:
There was a fix for a similar situation when the power is cut without shutting down the pi. Probably your issue was the same

TomTurnschuh, 2018-05-19:
This issue also occurs, if the drive should be formated while ZRAM is enabled.