nextcloud/nextcloudpi #459: Stuck at update in process...

JunkyardCat, 2018-03-27:
My nextcloud pi is stuck at update in process,  no other things are shown except that text. Is there a way to check the update process like how many percentage has already happened?

Here is how i updated my pi
1.) go to web ui as admin and click settings
2.) click update
3.) wait steps to finish
4.) encountered error during download step (parsing error)
5.) refreshed in hopes that it will fix things but found out i couldnt see the progress again
5.) terminated maintenance mode through ssh
7.) relogged in
8.) proceed to run update again
9.) amazed that it already done in download and proceeded in next part
10.) click finish button(forgot the exact name but its the one after choosing to end maintenance mode)
11.) update in process... is displayed in my browser. kept refreshing after an 1hour still nothing

JunkyardCat, 2018-03-27:
i decided to reimage my pi

closed 2018-03-27 by JunkyardCat

nachoparker, 2018-03-27:
Hi, it is more appropriate to ask questions in the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) and leave this for bug reports and feature requests.

You will get more help there ;)