nextcloud/nextcloudpi #461: ncp-web authentication prompt keeps popping up

FlexMcMurphy, 2018-03-28:
Hello,

The fixed ip of my NextCloudPi is 192.168.1.124. I go to the ncp-web homepage using 192.168.1.124:4443 but the ncp-web authentication prompt keeps popping up and I have to enter the Pi username and pwd several times before it finally stops popping up but if I click on links on the left panel the anuthentication box appears again.

I actually completely removed and re-installed Chrome thinking it was an extension issue but the problem is still there.

I wonder is it because of the SSL which I have set up on the NextCloud site.. When I enter 192.168.1.124:4443 into the browser and his enter I get:

[image]

Then I enter the Pi username and password and hit enter...

[image]

... but then I have to enter the details again and again...

[image]

Here is my ncp.conf file:

```
Listen 4443
<VirtualHost _default_:4443>
  DocumentRoot /var/www/ncp-web
  SSLEngine on
  SSLCertificateFile /etc/ssl/certs/ssl-cert-snakeoil.pem
  SSLCertificateKeyFile /etc/ssl/private/ssl-cert-snakeoil.key
  
  # 2 days to avoid very big backups requests to timeout
  TimeOut 172800

  <IfModule mod_authnz_external.c>
    DefineExternalAuth pwauth pipe /usr/sbin/pwauth
  </IfModule>

</VirtualHost>
<Directory /var/www/ncp-web/>

  AuthType Basic
  AuthName "ncp-web login"
  AuthBasicProvider external
  AuthExternal pwauth

  <RequireAll>

   <RequireAny>
      Require host localhost
      Require local
      Require ip 192.168
      Require ip 10
   </RequireAny>

   Require user pi

  </RequireAll>

</Directory>
```

Here are the details of ncp-report:

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.53.5
NextCloudPi image    NextCloudPi_09-28-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
[... 133 lines cut ...]
{"reqId":"Wrqd-n8AAQEAAA2WK60AAAAB","level":3,"time":"2018-03-27T19:39:43+00:00","remoteAddr":"95.211.226.185","user":"--","app":"PHP","method":"GET","url":"\/recordings\/","message":"Exception: The requested uri(\/recordings\/) cannot be processed by the script '\/core\/templates\/404.php') at \/var\/www\/nextcloud\/lib\/private\/AppFramework\/Http\/Request.php#729","userAgent":"curl\/7.29.0","version":"12.0.3.3"}
{"reqId":"Wrq8jX8AAQEAACsqXaYAAFgQ","level":3,"time":"2018-03-27T21:50:05+00:00","remoteAddr":"213.233.150.113","user":"--","app":"PHP","method":"GET","url":"\/html\/index.html","message":"Exception: The requested uri(\/html\/index.html) cannot be processed by the script '\/core\/templates\/404.php') at \/var\/www\/nextcloud\/lib\/private\/AppFramework\/Http\/Request.php#729","userAgent":"Mozilla\/5.0 (Windows NT 6.1; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/65.0.3325.181 Safari\/537.36","version":"12.0.3.3"}
{"reqId":"Wrq8mn8AAQEAACspCCMAAAwL","level":3,"time":"2018-03-27T21:50:19+00:00","remoteAddr":"213.233.150.113","user":"--","app":"PHP","method":"GET","url":"\/html\/index.html","message":"Exception: The requested uri(\/html\/index.html) cannot be processed by the script '\/core\/templates\/404.php') at \/var\/www\/nextcloud\/lib\/private\/AppFramework\/Http\/Request.php#729","userAgent":"Mozilla\/5.0 (Windows NT 6.1; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/65.0.3325.181 Safari\/537.36","version":"12.0.3.3"}

```
</details>

The ncp-report shows "Password Mismatch" but I am typing in the correct password for the Pi user on my Raspberry Pi.

000-default.conf
```
<VirtualHost _default_:80>
  DocumentRoot /var/www/nextcloud
  <IfModule mod_rewrite.c>
    RewriteEngine On
    RewriteCond %{HTTPS} !=on
    RewriteRule ^/?(.*) https://%{SERVER_NAME}/$1 [R,L]
  </IfModule>
</VirtualHost>
```
Does 000-default.conf force the 192.168.1.124:4443 url to an https: request and is that throwing the authentication off maybe? The HTTPd log shows the error: "unable to get the remote host name" quite a few times. Maybe my SSL certs in ncp.conf are wrong too somehow.

Any help greatly appreciated,

Flex

nachoparker, 2018-03-28:
The username/password for ncp-web is _ncp_/_ownyourbits_ as explained [here](https://github.com/nextcloud/nextcloudpi/wiki/FAQ#what-are-pre-set-userspasswords-on-nextcloudpi). In any case, it is very recommended to change it.

It is normal to get the certificate security warning if you are accessing through IP. If you don't want to see that warning, you can access through nextcloudpi.local, or the DDNS/dnsmasq domain name.

closed 2018-03-28 by nachoparker

FlexMcMurphy, 2018-03-28:
Hello,

I am still having trouble logging in to the ncp-web admin panel.. the anthentication log in box keeps popping up.

Here is my /etc/apache2/sites-available/ncp.conf...

```
Listen 4443
<VirtualHost _default_:4443>
  DocumentRoot /var/www/ncp-web
  SSLEngine on
  SSLCertificateFile /etc/ssl/certs/ssl-cert-snakeoil.pem
  SSLCertificateKeyFile /etc/ssl/private/ssl-cert-snakeoil.key
  
  # 2 days to avoid very big backups requests to timeout
  TimeOut 172800

  <IfModule mod_authnz_external.c>
    DefineExternalAuth pwauth pipe /usr/sbin/pwauth
  </IfModule>

</VirtualHost>
<Directory /var/www/ncp-web/>

  AuthType Basic
  AuthName "ncp-web login"
  AuthBasicProvider external
  AuthExternal pwauth

  <RequireAll>

   <RequireAny>
      Require host localhost
      Require local
      Require ip 192.168
      Require ip 10
   </RequireAny>

   Require user pi

  </RequireAll>

</Directory>
```

Please bear with me while I check my understanding of ncp.conf above..

As ncp.conf is configured, in the RequireAll block, doesn't it require a username of Pi for authentication? That's why I use my Pi username and pwd... the same log-in I use to get into my NextCoudPi image on the Pi3 via SSH. Except when I enter it into the authentication box and hit enter the box just comes back again... if Chrome asks me do I want to save the password that doesn't make any difference.

When I try userame: ncp and pwd: ownyourbits it doesn't authenticate at all.

I can't access ncp-web from nextcloudpi.local, or the DDNS/dnsmasq domain name either. In the case of the DDNS/dnsmasq domain name this is probably failing and because of ncp.conf RequireAll block above? I do have a NameCheap domain name that I used with the LetsEncrypt dns-01 challenge to get an SSL cert. But if I use that domain it will fail because of the RequireAll block and it's SSL cert won't match the snakeoil one in ncp.conf either.

I guess the design of ncp.conf is that you shouldn't be accessing this ncp-web panel from across the external web but only on your local network... is that correct? That's why I use my Pis local ip address:4443 to access that ncp-web admin panel. And when I do it that way here is what I get:

[image]

When I try https://nextcloudpi:4443 here is what I get:

[image]

Here is my /etc/hosts file: 

```
127.0.0.1       localhost
::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters

127.0.1.1 nextcloudpi
192.168.1.124 [my NameCheap domain name]
```

Should I have nextcloudpi.local in there. 192.168.1.124 is my Pi3s local ip address. Maybe that should not be mapped to the NameCheap domain name?

Sorry for all these questions, thank you for your time Nacho,

Flex

nachoparker, 2018-03-29:
Your `ncp.conf` is very outdated. You probably didn't upgrade frequently in the past.

It should look like

```
Listen 4443
<VirtualHost _default_:4443>
  DocumentRoot /var/www/ncp-web
  SSLEngine on
  SSLCertificateFile /etc/ssl/certs/ssl-cert-snakeoil.pem
  SSLCertificateKeyFile /etc/ssl/private/ssl-cert-snakeoil.key

  # 2 days to avoid very big backups requests to timeout
  TimeOut 172800

  <IfModule mod_authnz_external.c>
    DefineExternalAuth pwauth pipe /usr/sbin/pwauth
  </IfModule>

</VirtualHost>
<Directory /var/www/ncp-web/>

  AuthType Basic
  AuthName "ncp-web login"
  AuthBasicProvider external
  AuthExternal pwauth

  SetEnvIf Request_URI "^" noauth
  SetEnvIf Request_URI "^index\.php$" !noauth
  SetEnvIf Request_URI "^/$" !noauth
  SetEnvIf Request_URI "^/wizard/index.php$" !noauth
  SetEnvIf Request_URI "^/wizard/$" !noauth

  <RequireAll>

   <RequireAny>
      Require host localhost
      Require local
      Require ip 192.168
      Require ip 172
      Require ip 10
   </RequireAny>

   <RequireAny>
      Require env noauth
      Require user ncp
   </RequireAny>

  </RequireAll>

</Directory>
```

You might be missing other important updates though. I suggest you backup and then restore to a new instance.

Then, follow the wizard or at least activate ncp-autoupdate to avoid this situation in the future

FlexMcMurphy, 2018-03-29:
Thanks for the response Nacho.. I'll do a fresh re-install from scratch. Nice project for me this weekend!

Flex